mainsail-crew/crowsnest #109: Issues with 2 camera's on M8P with CM4 & CB1

smwoodward, 2023-05-06:
### What happened

Here's the logs on debug and the errors that happens.  the last few lines keep repeating.

[05/06/23 13:00:15] crowsnest: Shutdown or Killed by User!
[05/06/23 13:00:15] crowsnest: Please come again :)
[05/06/23 13:00:15] crowsnest: Goodbye...
[05/06/23 13:00:17] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[05/06/23 13:00:17] crowsnest: Version: v3.0.7-3-g20ed6a8
[05/06/23 13:00:17] crowsnest: Prepare Startup ...
[05/06/23 13:00:17] crowsnest: INFO: Host information:
[05/06/23 13:00:17] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[05/06/23 13:00:17] crowsnest: Host Info: Kernel: Linux 6.1.21-v8+ aarch64
[05/06/23 13:00:17] crowsnest: Host Info: Model: Raspberry Pi Compute Module 4 Rev 1.0
[05/06/23 13:00:17] crowsnest: Host Info: Available CPU Cores: 4
[05/06/23 13:00:17] crowsnest: Host Info: Available Memory: 7819936 kB
[05/06/23 13:00:17] crowsnest: Host Info: Diskspace (avail. / total): 7.6G / 15G
[05/06/23 13:00:17] crowsnest: INFO: Checking Dependencys
[05/06/23 13:00:17] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/06/23 13:00:17] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/06/23 13:00:17] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[05/06/23 13:00:17] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/06/23 13:00:17] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[05/06/23 13:00:17] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/06/23 13:00:17] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[05/06/23 13:00:17] crowsnest: Version Control: ustreamer is up to date. (v5.38)
[05/06/23 13:00:17] crowsnest: Version Control: rtsp-simple-server is up to date. (v0.20.2)
[05/06/23 13:00:18] crowsnest: Version Control: ffmpeg is up to date. (4.3.6-0+deb11u1+rpt1)
[05/06/23 13:00:18] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[05/06/23 13:00:18] crowsnest: 		[crowsnest]
[05/06/23 13:00:18] crowsnest: 		log_path: ~/printer_data/logs/crowsnest.log
[05/06/23 13:00:18] crowsnest: 		log_level: debug
[05/06/23 13:00:18] crowsnest: 		delete_log: false
[05/06/23 13:00:18] crowsnest: 		
[05/06/23 13:00:18] crowsnest: 		[cam 1]
[05/06/23 13:00:18] crowsnest: 		mode: mjpg
[05/06/23 13:00:18] crowsnest: 		port: 8080
[05/06/23 13:00:18] crowsnest: 		device: /dev/video0
[05/06/23 13:00:18] crowsnest: 		resolution: 2592x1944
[05/06/23 13:00:18] crowsnest: 		max_fps: 15
[05/06/23 13:00:18] crowsnest: 		
[05/06/23 13:00:18] crowsnest: 		[cam 2]
[05/06/23 13:00:18] crowsnest: 		mode: mjpg
[05/06/23 13:00:18] crowsnest: 		port: 8081
[05/06/23 13:00:18] crowsnest: 		device: /dev/video2
[05/06/23 13:00:18] crowsnest: 		resolution: 800x600
[05/06/23 13:00:18] crowsnest: 		max_fps: 30
[05/06/23 13:00:18] crowsnest: INFO: Detect available Devices
[05/06/23 13:00:18] crowsnest: INFO: Found 2 total available Device(s)
[05/06/23 13:00:18] crowsnest: INFO: Found 2 available camera(s)
[05/06/23 13:00:18] crowsnest: /dev/v4l/by-id/usb-VNV_USB_Camera-video-index0 -> /dev/video0
[05/06/23 13:00:18] crowsnest: Supported Formats:
[05/06/23 13:00:18] crowsnest: 		[0]: 'MJPG' (Motion-JPEG, compressed)
[05/06/23 13:00:18] crowsnest: 		Size: Discrete 2592x1944
[05/06/23 13:00:18] crowsnest: 		Interval: Discrete 0.067s (15.000 fps)
[05/06/23 13:00:18] crowsnest: 		Interval: Discrete 0.100s (10.000 fps)
[05/06/23 13:00:18] crowsnest: 		Interval: Discrete 0.133s (7.500 fps)
[05/06/23 13:00:18] crowsnest: 		Interval: Discrete 0.200s (5.000 fps)
[05/06/23 13:00:18] crowsnest: 		Interval: Discrete 0.267s (3.750 fps)
[05/06/23 13:00:18] crowsnest: 		Interval: Discrete 0.400s (2.500 fps)
[05/06/23 13:00:18] crowsnest: 		Size: Discrete 2048x1536
[05/06/23 13:00:18] crowsnest: 		Interval: Discrete 0.067s (15.000 fps)
[05/06/23 13:00:19] crowsnest: 		Interval: Discrete 0.100s (10.000 fps)
[05/06/23 13:00:19] crowsnest: 		Interval: Discrete 0.133s (7.500 fps)
[05/06/23 13:00:19] crowsnest: 		Interval: Discrete 0.200s (5.000 fps)
[05/06/23 13:00:19] crowsnest: 		Interval: Discrete 0.267s (3.750 fps)
[05/06/23 13:00:19] crowsnest: 		Interval: Discrete 0.400s (2.500 fps)
[05/06/23 13:00:19] crowsnest: 		Size: Discrete 1600x1200
[05/06/23 13:00:19] crowsnest: 		Interval: Discrete 0.067s (15.000 fps)
[05/06/23 13:00:19] crowsnest: 		Interval: Discrete 0.100s (10.000 fps)
[... 285 lines cut ...]
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1426.121 stream] -- Using input channel: 0
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1426.132 stream] -- Using resolution: 2592x1944
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1426.132 stream] -- Using format: MJPEG
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1426.143 stream] -- Using HW FPS: 15
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- ERROR [1426.144 stream] -- Device doesn't support setting of HW encoding quality parameters
[05/06/23 13:00:29] crowsnest: DEBUG: Value is now: focus_absolute=5
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1426.144 stream] -- Using IO method: MMAP
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1426.188 stream] -- Requested 5 device buffers, got 5
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- ERROR [1426.196 stream] -- Can't start capturing: No space left on device
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1426.214 stream] -- Device fd=8 closed
[05/06/23 13:00:29] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1426.214 stream] -- Sleeping 1 seconds before new stream init ...
[05/06/23 13:00:30] crowsnest: DEBUG: Value is now: focus_absolute=0
[05/06/23 13:00:30] crowsnest: DEBUG: ustreamer [cam 1]: ================================================================================

### What did you expect to happen

Both camera's working

### How to reproduce

n/a

### Additional information

I'm not 100% sure if this might be the M8P USB controller isn't able to handle 2 camera's, or if this might be another issue.

KwadFan, 2023-05-09:
Hey there :wave: 

So, first of all, this isnt a bug related to crowsnest, therefor I will drop the bug label.

But, we have seen this a lot of times the last days on our discord and I am the one, who have to tell the bad news...

Its primarily attached to BTT Manta MCU Boards and it seems the used USB chip isnt capable of delivering the needed bandwith for two cams.

There is litterally nothing I can do about besides letting you know the facts.
I will let this issue open and pinned for others, that face the same issue.

Regards Kwad

smwoodward, 2023-05-09:
Yeah I was further testing it out and found out that it does appear to be a bandwidth issue on the USB controller.  I've emailed BTT about it to see if they can give any more information on it.

truelecter, 2023-06-19:
Hi. Just to put my 5 cents here. I'm using 2 usb cameras - one c920 and another is 4k 3do nozzle camera. 
However, I'm not using their maximum resolution - both a set to 640x480 and are provideing 60fps mjpeg stream without stuttering using crowsnest on this settings. Setting nozzle cam to 4k resolution will drop FPS to 15, but still the image will be shown

```[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: verbose
delete_log: false

[cam 1]
mode: mjpg
port: 8080
device: /dev/v4l/by-id/usb-046d_HD_Pro_Webcam_C920_89E7787F-video-index0
resolution: 640x480
max_fps: 60

[cam 2]
mode: mjpg
port: 8081
device: /dev/v4l/by-id/usb-XCG-221208-J_3DO_NOZZLE_CAMERA_4K_01.00.00-video-index0
resolution: 640x480
max_fps: 60```

The reason is not in controller itself, rather the fact that all USB ports are connected to a single USB 2.0 lane, which is available on CM4 by design

dragz, 2023-09-26:
Hi. I had the same issue with the manta board and two usb cams. Lowering the frame rate to 10 and resolution to 640x480 seems to make both cams work. Not ideal though...

treaves, 2024-01-11:
Does anyone know if this is 'fixed' on the Manta 2.0 boards?

meteyou, 2024-01-11:
@treaves this is a hardware limitation. so we cannot 'fix' this issue.

treaves, 2024-01-11:
To be clear, ‘on the Manta 2.0 boards’ recognizes that this is a hardware issue. I was asking if anyone who’s read this thread has any information as to if it too is using out-dated hardware, or might it now have a USB 3 bus.

RoisinDubh79, 2024-01-25:
@treaves can't comment on the Manta 2.0, but I can confirm a similar hardware constraint exists on the fairly new Bigtreetech Pad 7 with a CB1.
Was thinking of swapping out the CB1 for a CM4 but it's likely the underlying board that has the issue.

looxonline, 2024-03-25:
Just a heads up that this is not a limitation of the M8P. Rather, it is a known problem with some USB cameras and linux. The camera reports a required bandwidth that is far in excess of what it actually needs and then the driver says that all available bandwidth on the USB 2.0 bus is consumed. There are various hacks that can be used to bypass this limit but boy do I mean "hacks". https://stackoverflow.com/questions/9781770/capturing-multiple-webcams-uvcvideo-with-opencv-on-linux/23881125#23881125

mryel00, 2024-03-25:
The reason why we think it's the M8P is that this issue occurs only with that board and with CB1 and CM4 and it seems to appear always with that combination. So it's possible that this is also a problem with the kernel, but it's still most likely due to the M8P.

Furthermore we won't give any help with recompiling the uvc driver, or any problems resulting from it.

mryel00, 2024-03-25:
If that's the case I'm wondering why that issue isn't there on a Pi3B+. If I'm not mistaken it should have only one USB 2.0 controller as well.
Like I already said, we have a lot of users running two cams but only that specific combination shows this specific error.
So it would be a pretty bad coincidence that those users having such cams resulting in this issue, all have that specific combination.

mryel00, 2024-03-25:
Also as a sidenote: I'm not directly experienced in how this whole stuff plays together. It's just from my experiences supporting this stuff daily, it looks like a limitation of the M8P or maybe the CB1 and CM4 directly, but not because of the one USB 2.0 controller, as the Pi3B+ should then have the same problem, if I'm not mistaken.

looxonline, 2024-03-25:
> Also as a sidenote: I'm not directly experienced in how this whole stuff plays together. It's just from my experiences supporting this stuff daily, it looks like a limitation of the M8P or maybe the [...]

It does only have a single USB controller but is it possible that users are using 1 x MIPI CSI camera and 1 x USB camera in such instances? We are going to run a bunch of comprehensive tests tomorrow morning to get empirical results when using the same cameras with different setups. Will let you know what we find.

smwoodward, 2024-03-25:
I can say with the testing I did, exact same cameras, I decided to add a pi3 that I had laying around as a camera server, I haven't had any issues with that. It was when I was running it on a manta board is when I had this issue, regardless of it being CB1 or CM4. Moved the exact same cameras to a pi3, no issues.

treaves, 2024-03-25:
My understanding is that compute modules do not provide USB; that’s up to the host board. That’s why both the CB1 & CM4 experience this. The Manta has the USB controller, and according to the docs, it’s a single USB 2 controller.

looxonline, 2024-03-25:
> My understanding is that compute modules do not provide USB; that’s up to the host board. That’s why both the CB1 & CM4 experience this. The Manta has the USB controller, and according to the docs, it’s a single USB 2 controller.

No. The controller is very much within the compute module. The compute module breaks out the USB peripheral which is just a differential pair. Thereafter that pair must be either routed directly out or sent to a hub for expansion. The M8P simply routes it to a high-speed hub for expansion. It can be set to a direct routing but that is for OTG mode to program the compute module if there is onboard mem.

looxonline, 2024-03-25:
> I can say with the testing I did, exact same cameras, I decided to add a pi3 that I had laying around as a camera server, I haven't had any issues with that. It was when I was running it on a manta [...]

Thanks for sharing. Will use this info during our testing.

mryel00, 2024-03-25:
I opened a [discussion](https://github.com/orgs/mainsail-crew/discussions/1833) about this topic that we continue there. I will lock this issue, as it's not an issue about crowsnest directly. Please post everything regarding this topic in the [discussion](https://github.com/orgs/mainsail-crew/discussions/1833).

Me and the mainsail-crew won't give help directly about fixes to this issue, like the one brought up by @looxonline. We would be grateful if @looxonline or other users here could help non tech-savvy users in that [discussion](https://github.com/orgs/mainsail-crew/discussions/1833), to get their stuff running.

closed 2024-03-25 by mryel00

mryel00, 2024-03-30:
We got the test results in the discussion and I again wrote a small summary of the solution that should resolve this issue:
https://github.com/orgs/mainsail-crew/discussions/1833#discussioncomment-8958956

Big thanks for the in depth testing 😄 

If someone needs help with it, please ask in the discussion.
We might provide a pre-applied solution in future. I will post an update here again, after that happened.