mainsail-crew/crowsnest #111: No picture in camera preview after several seconds or minutes

UchihaYuki, 2023-05-18:
### What happened

If I set the fps to 30 there will be no picture in camera preview in almost 1 second. And if I set the fps to 5, it will be around several minutes. (The delay is ramdomly)

### What did you expect to happen

A constant camera preview

### How to reproduce

I guess you need the same camera to reproduce it, because I have no problem when use Logitech C920.

### Additional information

```
biqu@BTT-CB1:~$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
```

```
biqu@BTT-CB1:~/crowsnest/bin/ustreamer$ ./ustreamer -p 8080 -d /dev/video0 --device-timeout=2 -m MJPEG --encoder=HW -r 1920x1080 -f 5 --allow-origin=* --static /home/biqu/crowsnest/ustreamer
-www --host=0.0.0.0
-- INFO  [2638.462      main] -- Starting PiKVM uStreamer 5.38 ...
-- INFO  [2638.462      main] -- Using internal blank placeholder
-- INFO  [2638.462      main] -- Enabling HTTP file server: /home/biqu/crowsnest/ustreamer-www
-- INFO  [2638.463      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [2638.463    stream] -- Using V4L2 device: /dev/video0
-- INFO  [2638.463    stream] -- Using desired FPS: 5
-- INFO  [2638.463      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [2638.533    stream] -- Device fd=8 opened
-- INFO  [2638.533    stream] -- Using input channel: 0
-- INFO  [2638.544    stream] -- Using resolution: 1920x1080
-- INFO  [2638.544    stream] -- Using format: MJPEG
-- INFO  [2638.557    stream] -- Using HW FPS: 5
-- ERROR [2638.557    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [2638.557    stream] -- Using IO method: MMAP
-- INFO  [2638.563    stream] -- Requested 5 device buffers, got 5
-- INFO  [2638.848    stream] -- Capturing started
-- INFO  [2638.848    stream] -- Using JPEG quality: encoder default
-- INFO  [2638.848    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [2638.848    stream] -- Capturing ...
-- INFO  [2651.512      http] -- HTTP: NEW client (now=1): [192.168.31.123]:59415, id=8942cd2901b27185
-- INFO  [2895.342      http] -- HTTP: DEL client (now=0): [192.168.31.123]:59415, id=8942cd2901b27185, Resource temporarily unavailable (reading,eof)
-- INFO  [2895.374      http] -- HTTP: NEW client (now=1): [192.168.31.123]:59619, id=ce4432f7401f0b57
-- INFO  [2912.262      http] -- HTTP: DEL client (now=0): [192.168.31.123]:59619, id=ce4432f7401f0b57, Resource temporarily unavailable (reading,eof)
-- INFO  [2912.274      http] -- HTTP: NEW client (now=1): [192.168.31.123]:59630, id=27192d2db408509c
-- INFO  [2932.435      http] -- HTTP: DEL client (now=0): [192.168.31.123]:59630, id=27192d2db408509c, Resource temporarily unavailable (reading,eof)
-- INFO  [2932.448      http] -- HTTP: NEW client (now=1): [192.168.31.123]:59646, id=37525a37e440fab1
-- INFO  [3173.183      http] -- HTTP: DEL client (now=0): [192.168.31.123]:59646, id=37525a37e440fab1, Resource temporarily unavailable (reading,eof)
-- INFO  [3173.221      http] -- HTTP: NEW client (now=1): [192.168.31.123]:59745, id=9cf1783ad5d570fa
-- INFO  [3297.912      http] -- HTTP: DEL client (now=0): [192.168.31.123]:59745, id=9cf1783ad5d570fa, Bad file descriptor (reading,eof)
```
Sometimes there will be no info in the log (just Capturing ...), when the camera preview has no picture.

UchihaYuki, 2023-05-18:
https://github.com/pikvm/ustreamer/issues/214

I can still see the cam preview when I use the ustreamer web page https://private-ip:8080, but no picture in fluidd or mainsail after several seconds or minutes.

[image]

UchihaYuki, 2023-05-18:
When I start to print, I get another error https://github.com/pikvm/ustreamer/issues/215.

KwadFan, 2023-05-20:
> ### What happened
> If I set the fps to 30 there will be no picture in camera preview in almost 1 second. And if I set the fps to 5, it will be around several minutes. (The delay is ramdomly)
>
> ### [...]

This is related to a network issue. The client looses connection to its host, in this case seems your browser looses connection to the host.
So, I am sorry but that isnt a bug in that regard. Software is working as intended. Cant do much about WiFi or network issues.

Therefor I will close this for now.
Regards Kwad

closed 2023-05-20 by github-actions[bot]