mainsail-crew/crowsnest #112: Update v4.0.0-0-g4f69c94  Cams stopped working

ken226, 2023-05-22:
### What happened

After updating to v4.0.0-0-g4f69c94,  both of my webcams stopped working.

After using KIAUH to uninstall, then reinstall Crowsnest, I was able to get my first camera working again.   The second camera still won't work.   I've reconfigured the crowsnest.conf file to add the second cam and updated the path, but after this update the second cam won't work.

### What did you expect to happen

I expected both cams to work, as they were before the update.

### How to reproduce

have a working system using two lotigech C270 webcams, then update to Crowsnest v4.0.0-0-g4f69c94

sammcj, 2023-05-23:
Just came here to log a bug for this as well.

Logs don't show any errors but the camera shows no signal despite working fine with other tools.

- Armbian (latest stable with all updates).
- Logitech G920 / G922
- RockPi 4 SE

[image]

```
rockpi:~:# cat /home/octo/printer_data/logs/crowsnest.log
[05/23/23 09:58:32] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[05/23/23 09:58:32] crowsnest: Version: v4.0.0
[05/23/23 09:58:32] crowsnest: Prepare Startup ...
[05/23/23 09:58:32] crowsnest: INFO: Checking Dependencys
[05/23/23 09:58:32] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/23/23 09:58:32] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/23/23 09:58:32] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/23/23 09:58:32] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/23/23 09:58:32] crowsnest: INFO: Detect available Devices
[05/23/23 09:58:32] crowsnest: Try to start configured Cams / Services...
[05/23/23 09:58:34] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[05/23/23 09:58:34] crowsnest: V4L2 Control: Device: [cam 1]
[05/23/23 09:58:34] crowsnest: V4L2 Control: Options: focus_auto=0,focus_absolute=30,power_line_frequency=1
[05/23/23 09:58:36] crowsnest: Starting ustreamer with Device /dev/cam_octoprint ...
[05/23/23 09:58:37] crowsnest:  ... Done!
```

ken226, 2023-05-23:
> Just came here to log a bug for this as well.
>
> Logs don't show any errors but the camera shows no signal despite working fine with other tools.
>
> * Armbian (latest stable with all updates).
> * [...]

Exactly the same here.  If even shows the correct frame rate,  just no video.  My log shows the same as yours as well.

Raspberry Pi CM4, MainsailOS,  two Logitech C270s

sammcj, 2023-05-23:
I've checked through the release notes - no mention of any breaking changes.

I tried downgrading back to the previous 3.x version, and doing a fresh build/install etc... but the same problem exists. On the 3.x version I did notice errors about rtsp-simple-server which I didn't see on 4.0.0 - that might have something to do with simple rstp server being rebranded with a new git url etc.... https://github.com/bluenviron/mediamtx#note-about-rtsp-simple-server - could be completely unrelated unless 4.0.0 is suppressing error output.

jalanjarosz, 2023-05-23:
Same issue, my log is:

`[05/22/23 21:35:04] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[05/22/23 21:35:05] crowsnest: Version: v4.0.0
[05/22/23 21:35:05] crowsnest: Prepare Startup ...
[05/22/23 21:35:05] crowsnest: INFO: Host information:
[05/22/23 21:35:05] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[05/22/23 21:35:05] crowsnest: Host Info: Release: MainsailOS release 1.1.1 (bullseye)
[05/22/23 21:35:05] crowsnest: Host Info: Kernel: Linux 6.1.21-v7+ armv7l
[05/22/23 21:35:05] crowsnest: Host Info: Model:  Raspberry Pi Zero 2 W Rev 1.0
[05/22/23 21:35:05] crowsnest: Host Info: Available CPU Cores: 4
[05/22/23 21:35:05] crowsnest: Host Info: Available Memory: 372020 kB
[05/22/23 21:35:05] crowsnest: Host Info: Diskspace (avail. / total): 108G / 118G
[05/22/23 21:35:05] crowsnest: INFO: Checking Dependencys
[05/22/23 21:35:05] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/22/23 21:35:05] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/22/23 21:35:05] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/22/23 21:35:05] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/22/23 21:35:05] crowsnest: Dependency: 'camera-streamer' not found. Exiting!
`

ken226, 2023-05-23:
> Same issue, my log is:
> 
> [05/22/23 21:35:04] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services. [05/22/23 21:35:05] crowsnest: Version: v4.0.0 [05/22/23 21:35:05] crowsnest: Prepare Startup ... [05/22/23 21:35:05] crowsnest: INFO: Host information: [05/22/23 21:35:05] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye) [05/22/23 21:35:05] crowsnest: Host Info: Release: MainsailOS release 1.1.1 (bullseye) [05/22/23 21:35:05] crowsnest: Host Info: Kernel: Linux 6.1.21-v7+ armv7l [05/22/23 21:35:05] crowsnest: Host Info: Model: Raspberry Pi Zero 2 W Rev 1.0 [05/22/23 21:35:05] crowsnest: Host Info: Available CPU Cores: 4 [05/22/23 21:35:05] crowsnest: Host Info: Available Memory: 372020 kB [05/22/23 21:35:05] crowsnest: Host Info: Diskspace (avail. / total): 108G / 118G [05/22/23 21:35:05] crowsnest: INFO: Checking Dependencys [05/22/23 21:35:05] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini. [05/22/23 21:35:05] crowsnest: Dependency: 'find' found in /usr/bin/find. [05/22/23 21:35:05] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs. [05/22/23 21:35:05] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer. [05/22/23 21:35:05] crowsnest: Dependency: 'camera-streamer' not found. Exiting!
> 
> **FIXED!**
> 
> SSH into SBC
> 
> ```
> cd ~/crowsnest
> sudo make install
> ```

I tried that already.  After I ran sudo make install,  my first webcam started working by my second cam still doesn't work.

KwadFan, 2023-05-23:
Seems I made a mistake in ustreamer.sh library. Already working on it.
I am sorry for inconvinience

KwadFan, 2023-05-23:
Should be fixed with https://github.com/mainsail-crew/crowsnest/commit/308109323a76b9dbf8b7150e886e10b0e9cfd948

ken226, 2023-05-23:
> Should be fixed with [3081093](https://github.com/mainsail-crew/crowsnest/commit/308109323a76b9dbf8b7150e886e10b0e9cfd948)

Dude,  you ROCK!!!   Updated, and both cams working again.  Thank you so much!

mostlyflawles, 2023-05-23:
I've updated to '4.0.1-0-g3081093' but still having problem with my raspberry camera.

```
05/23/23 08:35:43] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/23/23 08:35:43] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/23/23 08:35:43] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/23/23 08:35:43] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/23/23 08:35:43] crowsnest: Dependency: 'camera-streamer' not found. Exiting!

```

Runing make install...
```
pi@mainsailos:~/crowsnest $ sudo make install
Debian 10 (buster) has reached end of life.
Therefor crowsnest has also end support for this.
We are sorry for inconvinience...
make: *** [Makefile:17: install] Error 1

```

Not sure if it's a crowsnest or MainsailOS issue, but it occurred after crowsnest update 4 and all are up to date in 'Update-manager'.

meteyou, 2023-05-23:
`Debian 10 (buster) has reached end of life.` -> camera-streamer can only run on bullseye with a updated kernel

KwadFan, 2023-05-23:
> I've updated to '4.0.1-0-g3081093' but still having problem with my raspberry camera.
>
> ```
> 05/23/23 08:35:43] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
> [05/23/23 08:35:43] [...]

To stick with the old version, please run

```
cd crowsnest
git fetch && git pull
git switch legacy/v3
sudo systemctl restart crowsnest.service
```
Also modify your `moonraker.conf` to
```
# Crowsnest update_manager entry
[update_manager crowsnest]
type: git_repo
path: ~/crowsnest
origin: https://github.com/mainsail-crew/crowsnest.git
primary_branch: legacy/v3
managed_services: crowsnest
install_script: tools/pkglist.sh
```
and reboot afterwards

Regard Kwad

lanman1, 2023-05-23:
I have updated to v4.0.2-0-gb0c011b and still no streaming:

`[05/23/23 11:56:38] crowsnest: INFO: Checking Dependencys
[05/23/23 11:56:39] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/23/23 11:56:39] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/23/23 11:56:39] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/23/23 11:56:39] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/23/23 11:56:39] crowsnest: Dependency: 'camera-streamer' not found. Exiting!`

SpicyLimes, 2023-05-23:
I am having the same issue - I am running two of the same USB Webcams and only one is showing, however the FPS readout is active. I just did an update as well, but it didn't help.

I tried changing the mode to `camera-streamer`, and also tried the different methods of listing the device (as shown below).

Config and Log File below...

```
[crowsnest]
log_path: /home/pi/printer_data/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: ustreamer
enable_rtsp: false
rtsp_port: 8554
port: 8080
# device: /dev/video1
device: /dev/v4l/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-video-index1
resolution: 640x480
max_fps: 15
#custom_flags:
#v4l2ctl:

[cam 2]
mode: ustreamer
enable_rtsp: false
rtsp_port: 8554
port: 8081
# device: /dev/video0
device: /dev/v4l/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-video-index0
resolution: 640x480
max_fps: 15
#custom_flags:
#v4l2ctl:

```

```
[05/23/23 14:01:12] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[05/23/23 14:01:12] crowsnest: Version: v4.0.2
[05/23/23 14:01:12] crowsnest: Prepare Startup ...
[05/23/23 14:01:12] crowsnest: INFO: Host information:
[05/23/23 14:01:12] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[05/23/23 14:01:12] crowsnest: Host Info: Kernel: Linux 6.1.21-v8+ aarch64
[05/23/23 14:01:12] crowsnest: Host Info: Model:  Raspberry Pi 4 Model B Rev 1.4
[05/23/23 14:01:12] crowsnest: Host Info: Available CPU Cores: 4
[05/23/23 14:01:12] crowsnest: Host Info: Available Memory: 1713732 kB
[05/23/23 14:01:12] crowsnest: Host Info: Diskspace (avail. / total): 51G / 59G
[05/23/23 14:01:12] crowsnest: INFO: Checking Dependencys
[05/23/23 14:01:12] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/23/23 14:01:12] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/23/23 14:01:12] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/23/23 14:01:12] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/23/23 14:01:12] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[05/23/23 14:01:12] crowsnest: Version Control: ustreamer is up to date. (v5.38)
[05/23/23 14:01:12] crowsnest: Version Control: camera-streamer is up to date. ((9106d2c))
[05/23/23 14:01:12] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/crowsnest.conf'
[05/23/23 14:01:12] crowsnest:          [crowsnest]
[05/23/23 14:01:12] crowsnest:          log_path: /home/pi/printer_data/crowsnest.log
[05/23/23 14:01:12] crowsnest:          log_level: verbose
[05/23/23 14:01:12] crowsnest:          delete_log: false
[05/23/23 14:01:12] crowsnest:          no_proxy: false
[05/23/23 14:01:12] crowsnest:
[05/23/23 14:01:12] crowsnest:          [cam 1]
[05/23/23 14:01:12] crowsnest:          mode: ustreamer
[05/23/23 14:01:12] crowsnest:          enable_rtsp: false
[05/23/23 14:01:12] crowsnest:          rtsp_port: 8554
[05/23/23 14:01:12] crowsnest:          port: 8080
[... 81 lines cut ...]
[05/23/23 14:01:13] crowsnest:          0: Disabled
[05/23/23 14:01:13] crowsnest:          1: 50 Hz
[05/23/23 14:01:13] crowsnest:          2: 60 Hz
[05/23/23 14:01:13] crowsnest:          white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
[05/23/23 14:01:13] crowsnest:          sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
[05/23/23 14:01:13] crowsnest:          backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=1 value=1
[05/23/23 14:01:13] crowsnest:
[05/23/23 14:01:13] crowsnest:          Camera Controls
[05/23/23 14:01:13] crowsnest:
[05/23/23 14:01:13] crowsnest:          auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
[05/23/23 14:01:13] crowsnest:          1: Manual Mode
[05/23/23 14:01:13] crowsnest:          3: Aperture Priority Mode
[05/23/23 14:01:13] crowsnest:          exposure_time_absolute 0x009a0902 (int)    : min=1 max=80000 step=1 default=312 value=2496 flags=inactive
[05/23/23 14:01:13] crowsnest:          error 5 getting ext_ctrl Exposure, Dynamic Framerate
[05/23/23 14:01:13] crowsnest:          focus_absolute 0x009a090a (int)    : min=0 max=150 step=1 default=0 value=22 flags=inactive
[05/23/23 14:01:13] crowsnest:          focus_automatic_continuous 0x009a090c (bool)   : default=1 value=1
[05/23/23 14:01:13] crowsnest: Try to start configured Cams / Services...
[05/23/23 14:01:14] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[05/23/23 14:01:15] crowsnest: INFO: Configuration of Section [cam 2] looks good. Continue ...
[05/23/23 14:01:16] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[05/23/23 14:01:16] crowsnest: V4L2 Control: No parameters set for [cam 2]. Skipped.
[05/23/23 14:01:18] crowsnest: Starting ustreamer with Device /dev/v4l/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-video-index0 ...
[05/23/23 14:01:18] crowsnest: Starting ustreamer with Device /dev/v4l/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-video-index1 ...
[05/23/23 14:01:20] crowsnest:  ... Done!
```

meteyou, 2023-05-23:
@SpicyLimes you use ustreamer in your config and not camera-streamer...

SpicyLimes, 2023-05-23:
> @SpicyLimes you use ustreamer in your config and not camera-streamer...

I am not sure if that's a question, or a statement of fact.

I am currently using `ustreamer` but I also attempted to use `camera-streamer`, to which both did not work.

SpicyLimes, 2023-05-23:
> first, you have both paths to the same webcam. this cannot work and you dont use camera-streamer in your config, but you write above, that you changed it to...

If you check the very last word in the device line, you will notice that they are different. 

`index0 vs. index1`

If you're suggesting that this is the same device, then why am I getting two different FPS on the web UI? 

As far as the `camera-streamer` comment: I was simply listing that I had tried using both options but neither worked.

micag001, 2023-05-23:
I am having the same issue as @mostlyflawles, my raspi cam stopped working after update, when I tried to remove and reinstall through KIAUH I got the same message "Debian 10 (buster) has reached end of life" but when I tried the fix suggested by @KwadFan it is reporting "git: 'switch' is not a git command. See 'git --help'.

ken226, 2023-05-23:
> I am having the same issue - I am running two of the same USB Webcams and only one is showing, however the FPS readout is active. I just did an update as well, but it didn't help.
> 
> I tried changing the following settings:
> 
> * Mode to `camera-streamer`
> * Different methods of listing the device (as shown below)
> * Changing the resolution to 320x240 rather than 640x480
> 
> Config and Log File below...
> 
> [image]
> 
> ```
> [crowsnest]
> log_path: /home/pi/printer_data/crowsnest.log
> log_level: verbose                      # Valid Options are quiet/verbose/debug
> delete_log: false                       # Deletes log on every restart, if set to true
> no_proxy: false
> 
> [cam 1]
> mode: ustreamer
> enable_rtsp: false
> rtsp_port: 8554
> port: 8080
> # device: /dev/video1
> device: /dev/v4l/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-video-index1
> resolution: 640x480
> max_fps: 15
> #custom_flags:
> #v4l2ctl:
> 
> [cam 2]
> mode: ustreamer
> enable_rtsp: false
> rtsp_port: 8554
> port: 8081
> # device: /dev/video0
> device: /dev/v4l/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-video-index0
> resolution: 640x480
> max_fps: 15
> #custom_flags:
> #v4l2ctl:
> ```
> 
> ```

> ```

If it helps.  I got mine working after the last update.  This is what I have in my crowsnest.conf file:

[crowsnest]
log_path: /home/ken226/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam Crowsnest]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video0                     # See Log for available ...
resolution: 1280x720                     # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

[cam Gantry]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8081                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video2                     # See Log for available ...
resolution: 1280x720                    # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

Sambo726, 2023-05-23:
I am on v4.0.2 and cams still not working. Reverted to legacy and still a no go.
[crowsnest.log](https://github.com/mainsail-crew/crowsnest/files/11548234/crowsnest.log)

SpicyLimes, 2023-05-23:
> If it helps. I got mine working after the last update. This is what I have in my crowsnest.conf file:
> 
> [image]

Appreciate it, but it seems that your config is one that I have already tried. 

Also, if it is any help in understanding my specific issue, the output of `ls -l /dev/v4l/by-path` is:

```
lrwxrwxrwx 1 root root 13 May 23 13:42 platform-bcm2835-codec-video-index0 -> ../../video18
lrwxrwxrwx 1 root root 13 May 23 13:42 platform-bcm2835-isp-video-index0 -> ../../video20
lrwxrwxrwx 1 root root 13 May 23 13:42 platform-bcm2835-isp-video-index1 -> ../../video21
lrwxrwxrwx 1 root root 13 May 23 13:42 platform-bcm2835-isp-video-index2 -> ../../video22
lrwxrwxrwx 1 root root 13 May 23 13:42 platform-bcm2835-isp-video-index3 -> ../../video16
lrwxrwxrwx 1 root root 12 May 23 13:42 platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 May 23 13:42 platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 13 May 23 13:42 platform-feb10000.codec-video-index0 -> ../../video19
```

meteyou, 2023-05-23:
@Sambo726 you have also to run make uninstall > sudo make install!

```
[05/23/23 16:27:49] crowsnest: Dependency: 'ustreamer' not found. Exiting!
```

when you only change the repo, nothing backend with the backend services...

KwadFan, 2023-05-23:
> I am having the same issue as @mostlyflawles, my raspi cam stopped working after update, when I tried to remove and reinstall through KIAUH I got the same message "Debian 10 (buster) has reached end [...]

Then the git command ist too old.Use `git checkout legacy/v3` instead. `switch` is the modern variant

Sambo726, 2023-05-23:
I was able to finally get it going. New version works. First unintsall crowsnest then do a fresh install.

KwadFan, 2023-05-23:
Also to repeat that, a last time.
If you are running a Debian Buster based OS you can not update! This was alrwady mentioned in the pinned `Breaking Change...` issue. Camera-streamer relies on libcamera and the newest Kernel (5.15.x or 6.1.x). Both are not implementes in buster! Buster based debians need ro use legacy/v3. This branch is to provide a fallback. This branch will not recieve any update in the future because this version reached end of life equally to buster.

ytugarev, 2023-05-24:
I have very old Rpi cam, libcamera-still -o img.jpg -v seems to work fine.

[05/24/23 03:54:23] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[05/24/23 03:54:23] crowsnest:		[crowsnest]
[05/24/23 03:54:23] crowsnest:		log_path: /home/pi/printer_data/logs/crowsnest.log
[05/24/23 03:54:23] crowsnest:		log_level: debug
[05/24/23 03:54:23] crowsnest:		delete_log: true
[05/24/23 03:54:23] crowsnest:		no_proxy: false
[05/24/23 03:54:23] crowsnest:		
[05/24/23 03:54:23] crowsnest:		[cam 1]
[05/24/23 03:54:23] crowsnest:		mode: ustreamer
[05/24/23 03:54:23] crowsnest:		
[05/24/23 03:54:23] crowsnest:		enable_rtsp: false
[05/24/23 03:54:23] crowsnest:		rtsp_port: 8554
[05/24/23 03:54:23] crowsnest:		port: 8080
[05/24/23 03:54:23] crowsnest:		device: /dev/video0
[05/24/23 03:54:23] crowsnest:		resolution: 640x480
[05/24/23 03:54:23] crowsnest:		max_fps: 15
[05/24/23 03:54:23] crowsnest: INFO: Detect available Devices
[05/24/23 03:54:23] crowsnest: INFO: Found 1 total available Device(s)
[05/24/23 03:54:23] crowsnest: Detected 'libcamera' device -> /base/soc/i2c0mux/i2c@1/ov5647@36
[05/24/23 03:54:23] crowsnest: Try to start configured Cams / Services...
[05/24/23 03:54:28] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[05/24/23 03:54:30] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[05/24/23 03:54:34] crowsnest:  ... Done!
[05/24/23 03:54:34] crowsnest: Starting ustreamer with Device /dev/video0 ...
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: Parameters: --host 127.0.0.1 -p 8080 -d /dev/video0 --device-timeout=2 -r 640x480 -f 15 --allow-origin=* --static /home/pi/crowsnest/ustreamer-www
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.543      main] -- Starting PiKVM uStreamer 5.38 ...
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.544      main] -- Using internal blank placeholder
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.544      main] -- Enabling HTTP file server: /home/pi/crowsnest/ustreamer-www
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.545      main] -- Listening HTTP on [127.0.0.1]:8080
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.545    stream] -- Using V4L2 device: /dev/video0
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.545    stream] -- Using desired FPS: 15
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.546      http] -- Starting HTTP eventloop ...
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: ================================================================================
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.546    stream] -- Device fd=8 opened
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.546    stream] -- Using input channel: 0
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.546    stream] -- Using resolution: 640x480
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.546    stream] -- Using format: YUYV
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.547    stream] -- Querying HW FPS changing is not supported
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.547    stream] -- Using IO method: MMAP
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.553    stream] -- Requested 5 device buffers, got 5
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- ERROR [1062.555    stream] -- Can't start capturing: Invalid argument
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.556    stream] -- Device fd=8 closed
[05/24/23 03:54:34] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [1062.556    stream] -- Sleeping 1 seconds before new stream init ...

KwadFan, 2023-05-24:
Please stop using this issue for solving configuration issues.
As stated in the README please use our discord and open a post in support-forum.

I need to keep track of real bugs according to my software, not solving configuration issues.

But in terms of raspicams:

If you read through the log you will see a path beginning with `/base/soc` thats the device path you have to use. Also raspicams are not longer working with libcamera and ustreamer. Ustreamer is only a fallback option for those who cant use camera-streamer for some reason. So, if you want to use a raspicam you are forced to use either camera-streamer or switching to legacy/v3 branch

Regards Kwad

KwadFan, 2023-05-24:
PS: If the documentation is unclear at some point, please open an issue with the documentation label and describe as good as you can what is the issue with it.

SpicyLimes, 2023-05-24:
No offense, but some of us don't use Discord. Just felt like mentioning that in the kindest way possible. 

And sometimes discussions like this can lead to solutions - for example, my situation. Because of the multiple issues discussed in this particle issue, I actually resolved my problem based on someone else's general response.

Again, not trying to be aggressive or anything of the sort - but topics likes this can really help other. I apologize for complicating things on your end.

meteyou, 2023-05-24:
@SpicyLimes This is an open-source project that we all run in our spare time. You must also understand that our spare time is also important to us, and our support is as voluntary as developing this further. Please look at this issue. More and more people are coming, posting partly the same error without reading anything before.

We want to run the issue tracker only for issues and FR and handle the support in our discord server because there are some users who also help other users. Of course, it's not the best way for some people, but we have to pay attention to our mental state, and we also need time to develop. If we also support here, the development would come to a complete standstill.

I will close this issue now because discussions are also a waste of time for us.

closed 2023-05-24 by meteyou