mainsail-crew/crowsnest #113: Raspberry Pi Cam 3 "No signal"

godmanpicoli, 2023-05-24:
### What happened

My camera only displays "no signal"
What can it be ?

[05/24/23 12:39:13] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[05/24/23 12:39:14] crowsnest: Version: v4.0.2
[05/24/23 12:39:14] crowsnest: Prepare Startup ...
[05/24/23 12:39:14] crowsnest: INFO: Host information:
[05/24/23 12:39:14] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[05/24/23 12:39:14] crowsnest: Host Info: Kernel: Linux 6.1.21-v7+ armv7l
[05/24/23 12:39:14] crowsnest: Host Info: Model:  Raspberry Pi 3 Model B Rev 1.2
[05/24/23 12:39:14] crowsnest: Host Info: Available CPU Cores: 4
[05/24/23 12:39:14] crowsnest: Host Info: Available Memory: 858832 kB
[05/24/23 12:39:14] crowsnest: Host Info: Diskspace (avail. / total): 7.7G / 15G
[05/24/23 12:39:14] crowsnest: INFO: Checking Dependencys
[05/24/23 12:39:15] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/24/23 12:39:15] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/24/23 12:39:15] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/24/23 12:39:15] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/24/23 12:39:15] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[05/24/23 12:39:15] crowsnest: Version Control: ustreamer is up to date. (v5.38)
[05/24/23 12:39:17] crowsnest: Version Control: camera-streamer is up to date. ((9106d2c))
[05/24/23 12:39:17] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[05/24/23 12:39:17] crowsnest:		[crowsnest]
[05/24/23 12:39:17] crowsnest:		log_path: /home/pi/printer_data/logs/crowsnest.log
[05/24/23 12:39:17] crowsnest:		log_level: verbose
[05/24/23 12:39:17] crowsnest:		delete_log: false
[05/24/23 12:39:17] crowsnest:		no_proxy: false
[05/24/23 12:39:17] crowsnest:		
[05/24/23 12:39:17] crowsnest:		[cam 1]
[05/24/23 12:39:17] crowsnest:		mode: ustreamer
[05/24/23 12:39:17] crowsnest:		
[05/24/23 12:39:17] crowsnest:		enable_rtsp: false
[05/24/23 12:39:17] crowsnest:		rtsp_port: 8554
[05/24/23 12:39:17] crowsnest:		port: 8080
[05/24/23 12:39:17] crowsnest:		device: /dev/video0
[05/24/23 12:39:17] crowsnest:		resolution: 640x480
[05/24/23 12:39:17] crowsnest:		max_fps: 15
[05/24/23 12:39:17] crowsnest: INFO: Detect available Devices
[05/24/23 12:39:18] crowsnest: INFO: Found 1 total available Device(s)
[05/24/23 12:39:19] crowsnest: Detected 'libcamera' device -> /base/soc/i2c0mux/i2c@1/ov5647@36
[05/24/23 12:39:19] crowsnest: Try to start configured Cams / Services...
[05/24/23 12:39:22] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[05/24/23 12:39:23] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[05/24/23 12:39:27] crowsnest: Starting ustreamer with Device /dev/video0 ...
[05/24/23 12:39:27] crowsnest:  ... Done!

### What did you expect to happen

display the camera image

### How to reproduce

raspi cam

jmwhelan, 2023-05-24:
@godmanpicoli I had to do the following to get my pi cam 3 working:

SSH into pi.

pi@vzbot:~ $ libcamera-vid --list-cameras
Available cameras
-----------------
0 : imx708 [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SBGGR10_CSI2P' : 1536x864 [30.00 fps - (0, 0)/0x0 crop]
                             2304x1296 [30.00 fps - (0, 0)/0x0 crop]
                             4608x2592 [30.00 fps - (0, 0)/0x0 crop]

Then I updated crowsnest.conf with the following:

[cam 1]
mode: camera-streamer
enable_rtsp: false
port: 8080
device: /base/soc/i2c0mux/i2c@1/imx708@1a
resolution: 2304x1296
max_fps: 15

Then I updated the webcam section of mainsail:

[image]

You also have to make sure that camera_auto_detect=1 in your /boot/config.txt

closed 2023-05-24 by github-actions[bot]

Nox-jpg, 2024-12-12:
THANK YOU SO MUCH, after much torture this was the only solution to my problem.