mainsail-crew/crowsnest #116: V4 broke camera support

bistory, 2023-05-25:
### What happened

I just upgraded crowsnest using Kiauh and now I have no stream anymore.

### What did you expect to happen

I expected the upgrade to go well and to get my camera stream.

### How to reproduce

Upgrade from latest v3 to latest v4.

### Additional information

Log file may help, it doesn't find camera-streamer : 

```
[05/25/23 13:46:07] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[05/25/23 13:46:07] crowsnest: Version: v4.0.2
[05/25/23 13:46:07] crowsnest: Prepare Startup ...
[05/25/23 13:46:07] crowsnest: INFO: Host information:
[05/25/23 13:46:07] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[05/25/23 13:46:07] crowsnest: Host Info: Kernel: Linux 6.1.21-v8+ aarch64
[05/25/23 13:46:07] crowsnest: Host Info: Model:  Raspberry Pi 4 Model B Rev 1.4
[05/25/23 13:46:07] crowsnest: Host Info: Available CPU Cores: 4
[05/25/23 13:46:07] crowsnest: Host Info: Available Memory: 1713728 kB
[05/25/23 13:46:07] crowsnest: Host Info: Diskspace (avail. / total): 20G / 29G
[05/25/23 13:46:07] crowsnest: INFO: Checking Dependencys
[05/25/23 13:46:07] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/25/23 13:46:07] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/25/23 13:46:07] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/25/23 13:46:07] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/25/23 13:46:07] crowsnest: Dependency: 'camera-streamer' not found. Exiting!
```

closed 2023-05-25 by github-actions[bot]

meteyou, 2023-05-25:
@bistory because you don't follow the upgrade guide ;-)

bistory, 2023-05-25:
> @bistory because you don't follow the upgrade guide ;-)

Ho sorry, I didn't find any upgrade guide in crowsnest doc, I should have missed it ;-)

thetic, 2023-05-26:
Where's the upgrade guide?

meteyou, 2023-05-26:
@bistory @thetic we post this kind of news/announcements on Discord and on Patreon. These are our "news" platforms.