mainsail-crew/crowsnest #118: Latest update kills Rpi4 CSI camera feed, even on fresh build

3DPrintDemon, 2023-05-27:
### What happened

Previously working CSI camera no longer streams after installing latest version, just left with black screen & "No Signal" message & 15fps.

### What did you expect to happen

the camera to continue working

### How to reproduce

install latest update

### Additional information

[05/27/23 12:02:54] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[05/27/23 12:02:54] crowsnest: Version: v4.0.2
[05/27/23 12:02:54] crowsnest: Prepare Startup ...
[05/27/23 12:02:54] crowsnest: INFO: Host information:
[05/27/23 12:02:54] crowsnest: Host Info: Distribution: Debian GNU/Linux 11 (bullseye)
[05/27/23 12:02:54] crowsnest: Host Info: Kernel: Linux 6.1.21-v8+ aarch64
[05/27/23 12:02:54] crowsnest: Host Info: Model:  Raspberry Pi 4 Model B Rev 1.5
[05/27/23 12:02:54] crowsnest: Host Info: Available CPU Cores: 4
[05/27/23 12:02:54] crowsnest: Host Info: Available Memory: 3705524 kB
[05/27/23 12:02:54] crowsnest: Host Info: Diskspace (avail. / total): 51G / 59G
[05/27/23 12:02:54] crowsnest: INFO: Checking Dependencys
[05/27/23 12:02:54] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/27/23 12:02:54] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/27/23 12:02:54] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/27/23 12:02:54] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/27/23 12:02:54] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[05/27/23 12:02:54] crowsnest: Version Control: ustreamer is up to date. (v5.38)
[05/27/23 12:02:54] crowsnest: Version Control: camera-streamer is up to date. ((f0634f9))
[05/27/23 12:02:54] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[05/27/23 12:02:54] crowsnest:		[crowsnest]
[05/27/23 12:02:54] crowsnest:		log_path: /home/pi/printer_data/logs/crowsnest.log
[05/27/23 12:02:54] crowsnest:		log_level: verbose
[05/27/23 12:02:54] crowsnest:		delete_log: false
[05/27/23 12:02:54] crowsnest:		no_proxy: false
[05/27/23 12:02:54] crowsnest:		
[05/27/23 12:02:54] crowsnest:		[cam 1]
[05/27/23 12:02:54] crowsnest:		mode: ustreamer
[05/27/23 12:02:54] crowsnest:		
[05/27/23 12:02:54] crowsnest:		enable_rtsp: false
[05/27/23 12:02:54] crowsnest:		rtsp_port: 8554
[05/27/23 12:02:54] crowsnest:		port: 8080
[05/27/23 12:02:54] crowsnest:		device: /dev/video0
[05/27/23 12:02:54] crowsnest:		resolution: 640x480
[05/27/23 12:02:54] crowsnest:		max_fps: 15
[05/27/23 12:02:54] crowsnest: INFO: Detect available Devices
[05/27/23 12:02:54] crowsnest: INFO: Found 1 total available Device(s)
[05/27/23 12:02:54] crowsnest: Detected 'libcamera' device -> /base/soc/i2c0mux/i2c@1/ov5647@36
[05/27/23 12:02:54] crowsnest: Try to start configured Cams / Services...
[05/27/23 12:02:56] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[05/27/23 12:02:56] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[05/27/23 12:02:57] crowsnest: Starting ustreamer with Device /dev/video0 ...
[05/27/23 12:02:59] crowsnest:  ... Done!

meteyou, 2023-05-27:
Just check your log or read the docs: https://crowsnest.mainsail.xyz/faq-trouble-shoot/how-to-setup-a-raspicam

closed 2023-05-27 by github-actions[bot]

3DPrintDemon, 2023-05-27:
> Just check your log or read the docs: https://crowsnest.mainsail.xyz/faq-trouble-shoot/how-to-setup-a-raspicam

Missed the changes there, thanks for the headsup. All fixed