mainsail-crew/crowsnest #119: Cannot find source for 'SNAPSHOT' for one of the formats 'JPEG, MJPG'

xyzaxyz, 2023-05-27:
### What happened

Unsure whether this issue is correct in crowsnest or in https://github.com/ayufan/camera-streamer .

I am using an raspberry pi camera. With an recent update of crowsnest (with `mode: mjpeg`) the camera stream stopped working. I reinstalled crowsnest several times with no success. Tried ustreamer and camera-streamer in combination with legacy camera support and not.

Currently I have switched legacy camera support off and was trying camera-streamer, as it seems to be the more modern solution.

This **was** my crowsnest config before the update which broke my camera stream:
```ini
[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true

[cam 1]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/video0                     # See Log for available ...
resolution: 1280x720                     # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
```

This is my crowsnest config:
```ini
[crowsnest]
log_path: /home/admin/printer_data/logs/crowsnest.log
log_level: debug                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: camera-streamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video0                     # See Log for available ...
resolution: 800x600                     # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
```

This is the output of `tail -f ./printer_data/logs/crowsnest.log`
```sh
[05/27/23 19:29:52] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/device.c: CAMERA: Device path=/dev/video0 fd=14 opened
[05/27/23 19:29:52] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/device_options.c: CAMERA: Configuring option horizontalflip (00980914) = 0
[05/27/23 19:29:52] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/device_options.c: CAMERA: Configuring option verticalflip (00980915) = 0
[05/27/23 19:29:52] crowsnest: DEBUG: camera-streamer [cam 1]: device/buffer_list.c: CAMERA:capture: Using: 800x600/pGAA, buffers=3, bytesperline=1024, sizeimage=0.6MiB
[05/27/23 19:29:52] crowsnest: DEBUG: camera-streamer [cam 1]: device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 1.8 MiB
[05/27/23 19:29:52] crowsnest: DEBUG: camera-streamer [cam 1]: device/camera/camera_output.c: CAMERA: Cannot find source for 'SNAPSHOT' for one of the formats 'JPEG, MJPG'.
[05/27/23 19:29:52] crowsnest: DEBUG: camera-streamer [cam 1]: cmd/camera-streamer/main.c: ?: Automatically reconnecting in 1 seconds...
```

This is the output of `v4l2-ctl --list-devices`
```sh
unicam (platform:3f801000.csi):
        /dev/video0
        /dev/media3

bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media2

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media1
```

What can I do?

### What did you expect to happen

Camera stream works

### How to reproduce

- Use RPi3
- Use Pi Camera
- Use old crowsnest (where no camera-streamer/ustreamer was implemented) and let config auto-update

xyzaxyz, 2023-05-27:
It works again, I had to do the following steps:

https://os.ratrig.com/blog/crowsnest-v4-breaking-changes-manual-upgrade  

1. Ensure legacy-camera support is off
2. `sudo nano /boot/config.txt` -> add `dtoverlay=ov5647` in the `[all]` section
3. `sudo reboot now`
4. `libcamera-hello --list-cameras` listed me the device name `/base/soc/i2c0mux/i2c@1/ov5647@36`
5. Use the device name `/base/soc/i2c0mux/i2c@1/ov5647@36` in `crowsnest.conf`
6. In Mainsail, set WebCam settings: `URL Stream: webcam/webrtc`, `URL Snapshot: webcam/webrtc`, `Service: WebRTC (camera-streamer)`

meteyou, 2023-05-28:
It's also in our docs: https://crowsnest.mainsail.xyz/faq-trouble-shoot/how-to-setup-a-raspicam

closed 2023-05-28 by meteyou