mainsail-crew/crowsnest #120: Upgrade guide from v3 to v4

meteyou, 2023-05-28:
### What happened

a

### What did you expect to happen

a

### How to reproduce

a

DVSVIDEO, 2023-06-25:
Thank you for your reply. 
I am not a Linux expert. I did the whole Klipper/Mainsail installation by learning from a Youtube video, step by step.
Next time, if everything works fine finally, I will not update anymore.
I understand the difficulties of upgrading, but troubleshooting the bugs that arise from upgrading is not the level of ordinary people.
Thanks again for the links to the solution!

dderg, 2023-08-08:
I upgraded following the guide, and then used [this guide](https://os.ratrig.com/blog/crowsnest-v4-breaking-changes-manual-upgrade/) to get my old camera to work again. but it seems to ignore my v4l2ctl parameters now, so I cannot set the compression to a better quality. is there any other way to fix this than downgrading to v3?

meteyou, 2023-08-08:
@dderg pls use our discord server for support...

tcurdt, 2023-08-17:
Is there a more detailed guide on what has changed - and why it's worth upgrading?
I was hoping to find the usual `HISTORY.md` file to give a bit more context - but that does not seem to exist.

mryel00, 2023-08-17:
@tcurdt I'm not sure if something exists in this repo like that but there was an [announcement](https://discord.com/channels/758059413700345988/781284268654854154/1110660341693677650) on the [Mainsail Discord](https://discord.gg/mainsail) describing best what's new and why you want it:
> We have added the streamer camera-streamer from @ayufanpl  in Crowsnest v4 (atm only available for Raspberry Pi's). This streamer can provide snapshots, mjpeg & webrtc streams at the same time. With this streamer, it is now possible to stream live high-resolution streams via webrtc to the client without the bandwidth getting tight. (Attention: the HW encoder of the Raspberry can only encode full HD with 30fps. If multiple streams or higher resolution is required, the FPS will still decrease).

Also crowsnest got support for raspicam module v3 and arducams with this new streamer.