mainsail-crew/crowsnest #123: camera-streamer skipped on Pi running non-Raspbian OS

numanair, 2023-06-01:
### What happened

There are two different functions used for checking if Crowsnest is being installed on a Raspberry Pi. The `is_raspbian()` function returns 0 even when `is_raspberry_pi()` returns 1. This causes the camera-streamer install to get skipped on some Pi OS's such as Dietpi.

### What did you expect to happen

camera-streamer should be installed on non-Rasbian Pi OS.

### How to reproduce

Attempt to install Crowsnest on a Pi without `/etc/rpi-issue`

### Additional information

The issue is specifically in `build_apps()`, but it also needs `get_pkglist()` changed to is_raspberry_pi so that it pulls packages from pkglist-rpi. The pkglist-rpi works fine on my Dietpi Pi 3 system, but I am unsure if that is true for other cases. Is there ever a reason to use `is_raspbian()` specifically?

```bash
build_apps() {
    msg "Build dependend Stream Apps ..."
    msg "Cloning ustreamer repository ..."
    clone_ustreamer
    ## Detect Image build for Raspberry Pi's
    if [[ "$(is_raspberry_pi)" = "1" ]]; then
        msg "Cloning camera-streamer repository ..."
        clone_cstreamer
    fi
    if [[ "$(is_raspberry_pi)" = "0" ]]; then
        msg "Install of camera-streamer skipped, only supported on Raspberry SBC's! ... "
    fi
    sudo -u "${BASE_USER}" "${PWD}"/bin/build.sh --build
}

### Check if device is Raspberry Pi
is_raspberry_pi() {
    if [[ -f /proc/device-tree/model ]] &&
    grep -q "Raspberry" /proc/device-tree/model; then
        echo "1"
    else
        echo "0"
    fi
}
```

KwadFan, 2023-06-05:
Hello @numanair ,

First of all, thank for this issue.

The reason, why I removed the bug label is simple. As you can see in the readme, DietPI is currently not listed as supported. Therefor I will handle this issue more as a feature request as a real bug.

As a bug I describe things that arent working as intended. But if DietPi decides to do things different from 'Raspbian' , I have to test against them also. That would end in 'not a bug', more in 'please add dietpi compatibilty' right?

Therefor I will leave this issue open for now. I dont have a clue how fast I can do that, but I will take a look at by the time.
This would not only include rewriting build test also testing the whole variety of streamer capabilitys which simply take time, that I cant afford right now.

I will keep track of but will not provide any promises when its done... Said in 'mainsail-crew' manner: It's done, when it's done (soon tm) ...

Regards Kwad

mryel00, 2023-11-06:
@numanair I just looked into this in further detail and it seems like it's not as easy to just ditch the `is_raspbian` function. For Ubuntu there are no `libcamera` packages. Those are needed for camera-streamer to work. So we would have to change those detections to something like `is_debian_and_raspi` or something similar.
> Is there ever a reason to use `is_raspbian()` specifically (maybe for setting memory split)?

You are write about your assumption with the memory split. We have to know if it has a `/boot/config.txt` and if it's Raspberry Pi OS, as it seems like e.g. DietPi OS uses the same structure and settings but it uses it's own `gpu_mem` parameters for some reason. So we need a second function specific for DietPi OS to set some missing settings and also edit some files to be able to use the kernel codec modules and camera modules that are by default disabled.

tcurdt, 2023-11-14:
Urgh.

@mryel00 @numanair did you get this work?
I had the webcam running fine.

I upgraded and now also ended up with non-working webcam :-/

mryel00, 2023-11-14:
@tcurdt We will release an official update for this soon.
Currently it should complain about `camera-streamer not found`, if you have Crowsnest installed but didn't install camera-streamer on your own.
#201 and #204 will add support for it. The installer will then correctly install camera-streamer and set all the needed options for camera-streamer to work.
If you have camera-streamer already installed but it doesn't work correctly, you should run:
```bash
sudo /boot/dietpi/func/dietpi-set_hardware rpi-codec enable
sudo /boot/dietpi/func/dietpi-set_hardware rpi-camera enable
```
Those will activate the needed kernel modules that are disabled by default.

numanair, 2023-11-14:
> @tcurdt We will release an official update for this soon. Currently it should complain about `camera-streamer not found`, if you have Crowsnest installed but didn't install camera-streamer on your own. #201 and #204 will add support for it. The installer will then correctly install camera-streamer and set all the needed options for camera-streamer to work. If you have camera-streamer already installed but it doesn't work correctly, you should run:
> 
> ```shell
> sudo /boot/dietpi/func/dietpi-set_hardware rpi-codec enable
> sudo /boot/dietpi/func/dietpi-set_hardware rpi-camera enable
> ```
> 
> Those will activate the needed kernel modules that are disabled by default.

I just ran these commands and then make install from the develop branch. Worked perfectly on a new dietpi install. Thanks for working on this fix!

mryel00, 2023-11-14:
Thank you for testing 😄 

> I just ran these commands and then make install from the develop branch.

`sudo make install` will run those commands automatically on a DietPi system.

mryel00, 2023-11-15:
This is now merged with [v4.1.0](https://github.com/mainsail-crew/crowsnest/releases/tag/v4.1.0)

closed 2023-11-15 by mryel00