mainsail-crew/crowsnest #126: Raspicam

Michel-lev, 2023-06-03:
### What happened

Bonjour 
J'ai mis à jour la dernière version tout c'est bien passé par contre j'ai remarqué que j'ai un câble de 15 cm ça marche et quand je met un câble d 1 mètre ça ne fonctionne pas câble neuf d'où vient le problème 

### What did you expect to happen

J'attends que le câble d 1 mètres fonctionne 

### How to reproduce

Juste changer le câble 15 cm par un 1 mètres

Jacques1476, 2023-06-03:
Bonjour j'ai le même problème je vais voir avec des cable de 30 cm et 60 cm si le problème est toujours la car le 1 mètre et 1m12 ça ne fonctionne pas que 15 cm

meteyou, 2023-06-04:
Pls write English or we have to close this issue...

KwadFan, 2023-06-05:
Hello @Jacques1476 and @Michel-lev,

Just to give you a brief introduction to crowsnest. this is only a bash script started by debian's init system called `systemd`
This bash script only starts one of the available and compatible streamers called ustreamer and camera-streamer, both arent developed by myself and also I didint develop the hardware limitations of a raspberry pi.

So, if you ribbon cable in a certain length isnt working, that is not an crowsnest bug, I cant write better ribbon cables for you.... Please headover to raspberry forum or contact support of the manufacturer.

Also I2C (pronounced I squared C) is a short distance data transmission protocol even in controlled environments and industry quality cables you are limited to an length of 2,25m distance. A printer and a 2 Dollar ribbon cable is neither of that.

Therefor I will close this issue.

closed 2023-06-05 by github-actions[bot]