mainsail-crew/crowsnest #127: impossible to complete installing

scuba1976, 2023-06-07:
### What happened

when arrive at
[ 85%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/sctptransport.cpp.o
it stop e disconnect the terminal

### What did you expect to happen

finish installation

### How to reproduce

every time i try installation

mcmara, 2023-06-11:
Maybe a memory issue? I tried on a RPI4 8GB and, despite it slows down a lot at 85%, it will move forward. If you have only 1 GB of memory, I'd suggest to try to add the -j1 flag to make (sudo make -j1 install). It will be much slower but less memory intensive.

KwadFan, 2023-06-15:
I am sorry that I didnt answer earlier. Quite busy and preparing for vacation...

Yes, it is unfortunatly a known issue, I have only a Pi 4 (2 and 4G model) and a Zero2.
Both worked just fine, couldnt test other combinations here. But will switch to ayufans binary releases in the future.
Cant tell how fast I get that done but it will come, promised.

Regards Kwad

needs-coffee, 2023-07-16:
I will also add to this that i run into a problem trying to build during the install process on a pizero2 despite setting the make flag to `-j1` but it looks like the current build script was already doing this for my device based on the pr #124 
```
[ 78%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/datachannel.cpp.o
```
is the step that is stalling for me, i am going to try and stop all other processes on the pi and increase the swap and see if the build completes. Using precompiled binaries for the official install would make a lot of sense here for use of resource constrained devices.

RoisinDubh79, 2023-08-23:
I was having issues with the build hanging at 85% on a Pi Zero 2W. 
In the end I moved everything over to a spare 2GB Pi4 I had and that let the build complete.

needs-coffee, 2023-08-23:
@RoisinDubh79 did you change the swap size and disable all other processes before attempting building?

RoisinDubh79, 2023-08-23:
No, I was doing this all through the KIAUH menus, and at that point wasn't aware there were flags I could set.

The pi zero 2W wasn't able to run klipper and the pi camera V2 at the same time (mcu shut down), so I didn't look into it any further.
It's running on a 2GB pi4 but the video is choppy and sluggish.

needs-coffee, 2023-08-23:
I think the end solution here will be using precompiled binaries for the installer instead of building from source as in reality there isn’t much reason to rebuild from source unless you are on a dev build 

I the camera working fine for mine with klipper but the frame rate is capped at 10, it’s not quite powerful enough to handle the zerocam I use and klipper at an uncapped frame rate without occasional crashes mid print

SartorialGrunt0, 2023-08-26:
Is there any progress on resolving this?

needs-coffee, 2023-08-26:
> Is there any progress on resolving this?

Until the installer uses precompiled binaries instead of building your options are A) using a device with more available memory, or B) increasing the swap size and disabling all background processes before building.

SartorialGrunt0, 2023-08-26:
> > Is there any progress on resolving this?
> 
> 
> 
> Until the installer uses precompiled binaries instead of building your options are A) using a device with more available memory, or B) increasing the swap size and disabling all background processes before building. 

Can you give some more detail on how to attempt B?

needs-coffee, 2023-08-26:
> Can you give some more detail on how to attempt B?
Sure  😃 
Change the swap size -> 
1. `sudo nano  /etc/dphys-swapfile`
2. edit the line with `CONF_SWAPSIZE=100` to something larger, i found 300 enough but you can go higher if having issues.
3. reboot to have the new swapsize take effect. you can confirm this with `htop` - swap and its usage will be at the top.

Close background services ->
```bash
sudo systemctl stop klipper.service
sudo systemctl stop crowsnest.service
sudo systemctl stop moonraker.service
sudo systemctl stop mainsail.service
sudo systemctl stop nginx.service
sudo systemctl stop klipper-mcu.service
```
depending on your setup you may not have all of these running ( or you may have more)

check htop for running services ->
if you have installed other programs on your klipper pi you may need to stop some other services or terminate processes with htop/kill (F9 over a process in htop, you will need to open htop with sudo to kill processes not owned by your user). 

build crowsnest -> 
```bash
cd ~
git clone https://github.com/mainsail-crew/crowsnest.git
cd ~/crowsnest
sudo make install
```

Note it may still fail, restart and retry if this happens. Mine has worked on the 1st or 2nd attempt.
The current setting of GPU memory on my pi zero 2 at 160 currently stops me from booting and i needed to lower this to allow booting. This is most easily done by editing the file after install BEFORE rebooting. `sudo nano /boot/config.txt` and change the last line with `gpu_mem=160` to a lower value e.g. 120.
if you reboot and find the pi fails to start after a few minutes you an edit this file by taking the pi sd card into your pc and editing the config.txt and changing the gpu_mem line in the same way as above.

SartorialGrunt0, 2023-08-28:
> > Can you give some more detail on how to attempt B?
> 
> 
> 
> Sure  😃 
> 
> Change the swap size -> 
> 
> 1. `sudo nano /etc/dphys-swapfile`
> 
> 2. edit the line with `CONF_SWAPSIZE=100` to something larger, i found 300 enough but you can go higher if having issues.
> 
> 3. reboot to have the new swapsize take effect. you can confirm this with `htop` - swap and its usage will be at the top.
> 
> 
> 
> Close background services ->
> 
> ```bash
> 
> sudo systemctl stop klipper.service
> 
> sudo systemctl stop crowsnest.service
> 
> sudo systemctl stop moonraker.service
> 
> sudo systemctl stop mainsail.service
> 
> sudo systemctl stop nginx.service
> 
> sudo systemctl stop klipper-mcu.service
> 
> ```
> 
> depending on your setup you may not have all of these running ( or you may have more)
> 
> 
> 
> check htop for running services ->
> 
> if you have installed other programs on your klipper pi you may need to stop some other services or terminate processes with htop/kill (F9 over a process in htop, you will need to open htop with sudo to kill processes not owned by your user). 
> 
> 
> 
> build crowsnest -> 
> 
> ```bash
> 
> cd ~
> 
> git clone https://github.com/mainsail-crew/crowsnest.git
> 
> cd ~/crowsnest
> 
> sudo make -j1 install
> 
> ```
> 
> 
> 
> Note it may still fail, restart and retry if this happens. Mine has worked on the 1st or 2nd attempt.
> 
> The current setting of GPU memory on my pi zero 2 at 160 currently stops me from booting and i needed to lower this to allow booting. This is most easily done by editing the file after install BEFORE rebooting. `sudo nano /boot/config.txt` and change the last line with `gpu_mem=160` to a lower value e.g. 120.
> 
> if you reboot and find the pi fails to start after a few minutes you an edit this file by taking the pi sd card into your pc and editing the config.txt and changing the gpu_mem line in the same way as above.

Ok so this did work! It took awhile for sure. I am however now getting MCU timer to close failures. Should I reduce the swap size back down? Or just reduce the camera FPS and resolution?

needs-coffee, 2023-08-28:
> Ok so this did work! It took awhile for sure. I am however now getting MCU timer to close failures. Should I reduce the swap size back down? Or just reduce the camera FPS and resolution?

I have run into this issue on a couple of prints.
For longer prints i currently disable the webcam streams altogether to not risk this as klipper does not currently have a realistic way of recovering from stopped prints. 
Reducing the swap again may make a small difference but I think it is more likely the CPU is overloaded. I reduced the swap and it made no difference, which makes sense as swap is only used if there is insufficient free RAM which does happen during building but not during normal klipper runtime (At least for me).

You could try reducing resolution and FPS as this is most likely the issue. The end result however is that the pi zero2 is probably not quite powerful enough to run klipper and camera streaming simultaneously at higher resolutions and a more powerful device may be required. I am likely to move to a Pi4 as i have one that used to run homeassistant and the DSI interface will be useful for klipperscreen.

mryel00, 2023-08-28:
Just to give my 2 cents.
I already built on the pi zero 2 multiple times without any issues, but without any klipper mcu connected, so disconnecting everything can maybe help too. I'm only running a pi zero 2 as a test box, to build and test crowsnest, so I cannot say anything about a "real" use-case scenario for printing.

@needs-coffee Your suggestions to reduce the fps and resolution are absolutely right, as this is most likely a CPU overload or USB-bandwidth issue.

the-incredulous-bulk, 2023-10-12:
> > Can you give some more detail on how to attempt B?
> 
> Sure 😃 Change the swap size ->
> 
>     1. `sudo nano /etc/dphys-swapfile`
> 
>     2. edit the line with `CONF_SWAPSIZE=100` to something larger, i found 300 enough but you can go higher if having issues.
> 
>     3. reboot to have the new swapsize take effect. you can confirm this with `htop` - swap and its usage will be at the top.
> 
> 
> Close background services ->
> 
> ```shell
> sudo systemctl stop klipper.service
> sudo systemctl stop crowsnest.service
> sudo systemctl stop moonraker.service
> sudo systemctl stop mainsail.service
> sudo systemctl stop nginx.service
> sudo systemctl stop klipper-mcu.service
> ```
> 
> depending on your setup you may not have all of these running ( or you may have more)
> 
> check htop for running services -> if you have installed other programs on your klipper pi you may need to stop some other services or terminate processes with htop/kill (F9 over a process in htop, you will need to open htop with sudo to kill processes not owned by your user).
> 
> build crowsnest ->
> 
> ```shell
> cd ~
> git clone https://github.com/mainsail-crew/crowsnest.git
> cd ~/crowsnest
> sudo make -j1 install
> ```
> 
> Note it may still fail, restart and retry if this happens. Mine has worked on the 1st or 2nd attempt. The current setting of GPU memory on my pi zero 2 at 160 currently stops me from booting and i needed to lower this to allow booting. This is most easily done by editing the file after install BEFORE rebooting. `sudo nano /boot/config.txt` and change the last line with `gpu_mem=160` to a lower value e.g. 120. if you reboot and find the pi fails to start after a few minutes you an edit this file by taking the pi sd card into your pc and editing the config.txt and changing the gpu_mem line in the same way as above.

I registered an account to add to this thread and confirm that I was having issues building with my Pi 3 Model B, only 1GB memory, but this work-around sorted everything. Thanks very much!

basementbuilder17, 2024-02-24:
This worked perfectly for me, same situation on Pi3b. Thank you.

rpsdhillon, 2024-10-22:
This worked for me on my pi z2w aswell, many thanks

closed 2026-04-17 by mryel00