mainsail-crew/crowsnest #135: Camera Streamer UDP Ghost Stream

Xaver106, 2023-06-29:
### What happened

Camera Streamer currently has an isse, where a stream keeps running after it has been closed. This can overwhelm older Hardware quite quickly. In my case even causing me to no longer  be able to reach my mainsail instance.
https://github.com/ayufan/camera-streamer/issues/55

### What did you expect to happen

A mention about this on the Documentation.

### How to reproduce

Explainewd in the Linked Bug Report: https://github.com/ayufan/camera-streamer/issues/55

### Additional information

It is probably best to either not reccomend the camera Streamer as default or add a warning.

Xaver106, 2023-06-29:
Ustreamer seems to still depend on the legacy camera support. Not using the WebRTC Stream option and sticking to the MJPEG-streamer seems to not trigger the Ghost Stream

jpiccari, 2023-07-27:
Ghost streams are caused by lack of proper shutdown of the stream by the clients. Chrome, Edge, Safari, etc do not properly signal closing the peer connection and instead just stop receiving communication. For mobile devices this happens quite frequently due to locking/unlocking a phone. I have a fix that adds a keep-alive type mechanism to camera-streamer but it will also require updates to Mainsail as it has its own front-end code displaying the stream.

I haven't looking into the rtsp implementation used within Mainsail/camera-streamer but that might be another viable workaround.

meteyou, 2023-07-29:
@jpiccari Mainsail don't have a rtsp stream client right now, but feel free to contact me, if you need help to implement the keep-alive mechanism in the webrtc client in Mainsail.

jpiccari, 2023-08-05:
I can submit a PR with the changes I have in my local mainsail. It will be dependent on camera-streamer being updated (also need to submit a PR over there) and frankly, camera-streamer is just not very reliable (webrtc or otherwise) in my experience running on a Raspberry Pi 3B+ so even with the fix I think it is probably better for users to look to ustreamer + janus-gateway for webrtc streaming (although I haven't been able to set that up yet, anyone point me to some docs?).

jasonmacdonald, 2023-12-08:
I'm often running into these ghost streams; when I check the bandwidth used by my Pi 4, it's often up in the 20 MB/sec range in only a couple of days. Whereas a single stream should only be using around 1MB. I have to continue to shutdown crowsnest and restart it to clear them out. 

Is there a better workaround?

AndyHazz, 2025-12-03:
That was my anecdotal report on Reddit 😬 I've been following any updates and I could still reproduce the issue today on mainsail 2.14.0 ...

1. Use crowsnest WebRTC (camera-streamer) service, with URL stream /webcam/webrtc
2. Access mainsail and view webcam using wireguard on android, can see in wireguard UI data streaming constantly (around 500kb/s on my device)
3. Close all browser tabs, wait a while, data still streaming
4. Restart phone
5. Reenable wireguard, don't view mainsail or anything else, wireguard is immediately streaming data at the same rate as before.
6. Stop crowsnet service in mainsail - data transfer in wireguard instantly drops to normal levels showing the stream was coming from crowsnest

500kb/s doesn't sound like much but it mounts up pretty quickly when it's relentless and you're away from home for any length of time.

Now the good news! I've just updated to mainsail 2.15.0 released last week - repeated the same tests, and the wireguard data stream stops within a minute of viewing the webcam in mainsail.

closed 2025-12-03 by mryel00