mainsail-crew/crowsnest #140: Camera not working with klipperscreen

Voxel3DPrinting, 2023-07-15:
### What happened

The camera (raspicam V3) stopped working after klipperscreen install

**_Now reports;_**
 **libcamera-hello --list-cameras**
Available cameras
-----------------
0 : imx708_wide [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SBGGR10_CSI2P' : 1536x864 [30.00 fps - (0, 0)/0x0 crop]
                             2304x1296 [30.00 fps - (0, 0)/0x0 crop]
                             4608x2592 [30.00 fps - (0, 0)/0x0 crop]

**libcamera-jpeg -o test.jpg;**
Preview window unavailable
[0:02:31.871029917] [1444]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:02:31.907783532] [1445] ERROR V4L2 v4l2_device.cpp:352 'imx708_wide': Unable to set controls: Device or resource busy
[0:02:31.936512337] [1445]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:02:31.936836069] [1444]  INFO Camera camera.cpp:841 Pipeline handler in use by another process
ERROR:  failed to acquire camera /base/soc/i2c0mux/i2c@1/imx708@1a 

### What did you expect to happen

Install klipperscreen via KIAUH, the camera keeps functioning (worked flawlessly before this install)

### How to reproduce

Install klipperscreen plugin via kiauh

### Additional information

Install;
Raspberry pi 3 model b V1.2 1GB 
Raspberry pi camera V3 wide
BTT piTFT50
CONF; [crowsnest (2).txt](https://github.com/mainsail-crew/crowsnest/files/12061776/crowsnest.2.txt)
LOG; [crowsnest (12).log](https://github.com/mainsail-crew/crowsnest/files/12061740/crowsnest.12.log)

KwadFan, 2023-07-17:
@Voxel3DPrinting 
I am sorry, but that is a whole topic that drives many peoples crazy, I can't do anything here. Crowsnest just launches a streamer. Configuration issues, that are dependending on kernel and/or thirdparty libaries/binarys is out of my scope.

Also please read https://github.com/mainsail-crew/crowsnest#support ...
That said I will close this issue. Please feel free to join our Discord for help about that.
This issue tracker is only meant for topics related to my scripts, not for configuration issues. Thanks in advance.

Regards Kwad

closed 2023-07-17 by github-actions[bot]

NerdyGriffin, 2024-06-10:
To clarify the cause of this issue for anyone who finds this in the future:

KlipperScreen is not compatible with the new camera stack 'libcamera',
and the Camera Module V3 is not compatible with the legacy camera stack.
Therefore, KlipperScreen is not compatible with the Camera Module v3

mryel00, 2024-06-10:
> KlipperScreen is not compatible with the new camera stack 'libcamera'

That's wrong. Klipperscreen is compatible with the libcamera stack. The problem is that there are some `/boot/config.txt` settings causing problems and some screens depend on them.
The real problem is the kernel, as the libcamera stack and screens cause a lot of traffic. The Raspberry Pi foundation fixed some bugs regarding screens+libcamera cams in the kernel. So a Bookworm installation could work for **_some_** people.

Why only some people? Because each screen is different. Depending on the screen and the `/boot/config.txt` settings, you might be lucky and get all functions of the screen with a cam, but most people cannot rotate the screen or use touch, and some people cannot use them at all, even on Bookworm.