mainsail-crew/crowsnest #143: Crowsnest is broken after system packages upgrade.

vladimir-poleh, 2023-07-20:
### What happened

Board: Raspberry Pi 4
OS: Raspberry Pi OS (Debian GNU/Linux 11 (bullseye) arm64)

After updating system packages I have broken camera streaming with following error in the log:

```
[07/20/23 21:28:23] crowsnest: Starting camera-streamer with Device /base/soc/i2c0mux/i2c@1/ov5647@36 ...
[07/20/23 21:28:23] crowsnest: DEBUG: camera-streamer [cam 1]: Parameters: --http-port=8080 --camera-path=/base/soc/i2c0mux/i2c@1/ov5647@36 --camera-type=libcamera --camera-format=YUYV --camera-snapshot.height=1080 --camera-width=1296 --camera-height=972 --camera-fps=15 --camera-auto_reconnect=1 --camera-options=rotate=180
[07/20/23 21:28:23] crowsnest: DEBUG: camera-streamer [cam 1]: /home/pi/crowsnest/bin/camera-streamer/camera-streamer: error while loading shared libraries: libcamera.so.0.0.4: cannot open shared object file: No such file or directory
[07/20/23 21:28:23] crowsnest: ERROR: Start of camera-streamer [cam 1] failed!
```

Upgraded packages:

```
libgpac10:arm64 (1.0.1+dfsg1-4+deb11u2, 1.0.1+dfsg1-4+deb11u3)
libcamera-dev:arm64 (0~git20230302+923f5d70-1, 0~git20230707+2783c8d8-1)
python3-picamera2:arm64 (0.3.9-1, 0.3.12-1)
libcamera-apps-lite:arm64 (0~git20230301+54a781d-1, 1.2.1-1)
libssl-dev:arm64 (1.1.1n-0+deb11u5, 1.1.1n-0+deb11u5+rpt1)
libssl1.1:arm64 (1.1.1n-0+deb11u5, 1.1.1n-0+deb11u5+rpt1)
libcamera0:arm64 (0~git20230302+923f5d70-1, 0~git20230707+2783c8d8-1)
libgstreamer-plugins-base1.0-0:arm64 (1.18.4-2, 1.18.4-2+deb11u1)
python3-libcamera:arm64 (0~git20230302+923f5d70-1, 0~git20230707+2783c8d8-1)
openssl:arm64 (1.1.1n-0+deb11u5, 1.1.1n-0+deb11u5+rpt1)
```

File `libcamera.so.0.0.4` is missing in file system, updated .so file exists in `/usr/lib/aarch64-linux-gnu/libcamera.so.0.0.5`

### What did you expect to happen

Crowsnest is able to stream video from camera.

### How to reproduce

Upgrade `libcamera0` to version `0~git20230707+2783c8d8-1`.

iCarl, 2023-07-20:
I have this exact issue

KwadFan, 2023-07-21:
Hey there, I heard that internally yesterday. Please recompile camera-streamer.

Run 
```cd ~/crowsnest
make buildclean
make build
```
That should solve the Issue for now.

Regards Kwad

rnaharro, 2023-07-22:
> Hey there, I heard that internally yesterday. Please recompile camera-streamer.
> 
> Run
> 
> ```
> cd ~/crowsnest
> make buildclean
> make build
> ```
> 
> That should solve the Issue for now.
> 
> Regards Kwad

This solve the issue for me!
Thanks!

vladimir-poleh, 2023-07-22:
Recompiling resolves this problem.

closed 2023-07-22 by vladimir-poleh

drizztguen77, 2023-08-01:
Recompiling fixed it for me as well

travmo, 2023-08-09:
Can also confirm, KwadFan's steps resolved the issue for me as well

jinnn1989, 2023-08-26:
I have same issue, resinstalled Crowsnest but when i hit command for Afmode=2, nothing happens, I think this has broken autofocus. and even LenPosition is not working @KwadFan

kipp6912, 2023-10-14:
I cant even get it to compile camera-streamer. Did a clean install did the build clean and I always get this error.
make[2]: Leaving directory '/home/vmoczulski/crowsnest/bin/camera-streamer'
make[1]: *** [Makefile:62: all] Error 2
make[1]: Leaving directory '/home/vmoczulski/crowsnest/bin/camera-streamer'
make: *** [Makefile:23: build] Error 2

mryel00, 2023-10-14:
@testack18 @kipp6912 Please join the [Discord](https://discord.gg/mainsail) if you need help. If your issue isn't resolved after doing this or gettings errors while building camera-streamer, the issues aren't related to this problem.

sermayoral, 2025-02-18:
I've had the same error today. Just update all packages, and camara is not working. After trying

```
cd ~/crowsnest
make buildclean
make build
```

I got the following errors:

device/libcamera/options.cc: In function ‘int libcamera_device_dump_control_option(int (*)(device_option_t*, void*), void*, const libcamera::ControlId&, const libcamera::ControlInfo*, const libcamera::ControlValue*, bool)’:
device/libcamera/options.cc:183:10: error: enumeration value ‘ControlTypeUnsigned16’ not handled in switch [-Werror=switch]
  183 |   switch (control_id.type()) {
      |          ^
device/libcamera/options.cc:183:10: error: enumeration value ‘ControlTypeUnsigned32’ not handled in switch [-Werror=switch]
device/libcamera/options.cc:183:10: error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
device/libcamera/options.cc: In function ‘int libcamera_device_set_option(device_t*, const char*, const char*)’:
device/libcamera/options.cc:400:12: error: enumeration value ‘ControlTypeUnsigned16’ not handled in switch [-Werror=switch]
  400 |     switch (control_id->type()) {
      |            ^
cc -std=gnu17 -MMD -Werror -Wall -g -I/home/sergio/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o html/webrtc.html.o html/webrtc.html.c
device/libcamera/options.cc:400:12: error: enumeration value ‘ControlTypeUnsigned32’ not handled in switch [-Werror=switch]
device/libcamera/options.cc:400:12: error: enumeration value ‘ControlTypePoint’ not handled in switch [-Werror=switch]
cc1plus: all warnings being treated as errors
make[2]: *** [Makefile:106: device/libcamera/options.o] Error 1

Any clue? Thanks!!

mryel00, 2025-02-18:
It's a different error but running `make update` should fix it. Further help -> [Discord](https://disord.gg/mainsail)

sermayoral, 2025-02-18:
> It's a different error but running `make update` should fix it. Further help -> [Discord](https://disord.gg/mainsail)

It works! Thank you very much :-)