mainsail-crew/crowsnest #150: tools/pkglist.sh not present in repo

gjsmo, 2023-07-24:
### What happened

The docs say to change "install_script: tools/install.sh to install_script: tools/pkglist.sh", however `tools/pkglist.sh` is not present in the master, develop, or legacy/v3 branches. As a result, Moonraker throws an error on loading.

### What did you expect to happen

Docs should match repo.

### How to reproduce

N/A

mryel00, 2023-07-26:
On installation the [install.sh](https://github.com/mainsail-crew/crowsnest/blob/8b54eae98dcb5de73f5aab121805ef018cda5052/tools/install.sh#L58-L64) script is creating symlinks to `tools/libs/pkglist-rpi.sh` for raspbian and `tools/libs/pkglist-generic.sh` for other distributions. As this file has to be different for different distributions, regarding the camera-streamer dependencies, it cannot be inside the repos and has to be created during installation. The functions creating the symlinks can be found [here](https://github.com/mainsail-crew/crowsnest/blob/8b54eae98dcb5de73f5aab121805ef018cda5052/tools/libs/core.sh#L45-L51).
If there isn't a symlink for you, please run the commands for your distribution:

Rasbian:
```bash
rm -rf ~/crowsnest/tools/pkglist.sh
ln -sf ~/crowsnest/tools/libs/pkglist-rpi.sh ~/crowsnest/tools/pkglist.sh
```
Other distributions:
```bash
rm -rf ~/crowsnest/tools/pkglist.sh
ln -sf ~/crowsnest/tools/libs/pkglist-generic.sh ~/crowsnest/tools/pkglist.sh
```

Currently is only a bug with MainsailOS and that symlink known. The symlink will always link to `/home/pi/tools/libs/pkglist-rpi.sh` no matter what username got specified. But that shouldn't affect manual or kiauh installations and can be fixed with above commands.

closed 2023-07-26 by mryel00

gjsmo, 2023-07-27:
Interesting, I did run the script and no symlink was produced. Crowsnest is working properly and reports itself as version `v4.0.2-2-g8b54eae`. I'll make the symlink and chalk it up to gamma rays. ¯\\\_(ツ)_/¯

mryel00, 2023-07-27:
Weird. If you or someone else can reproduce it, then let us know how and I will reopen this issue. As of now I can't think of how that should fail🤔 
Especially if the commands work for you.

PockyBum522, 2023-08-08:
Just to add my 0.02, I installed Crowsnest via KIAUH last week, Tuesday August 1st, I think. Everything came up, crowsnest.conf was set up, everything looked good except I was getting pkglist.sh missing errors in mainsail about crowsnest. Making the symlink then rebooting (Had to reboot twice, actually, first time it said crowsnest is corrupt.) worked to fix that error, but no webcam stream yet.

I was then getting ustreamer not found messages in the logs. I ran 
cd ~/crowsnest 
sudo make install

(Which you would have thought KIAUH would have done.) And now things are working great.

I know this might just be an issue where KIAUH isn't running crowsnest install properly, but I figure more info might be helpful so I'm leaving this here.

mryel00, 2023-08-08:
@PockyBum522 On what OS did you install with kiauh?
I really don't know why it shouldn't work with kiauh, as it's really calling just `sudo make install`.
You can see that [here](https://github.com/th33xitus/kiauh/blob/65854c8da633a927e6f29a2c05d8263deb2df488/scripts/crowsnest.sh#L108)

PockyBum522, 2023-08-09:
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)

And yeah, I figured that's all it would be doing. If I get a reason to redo everything I'll check carefully when I install Crowsnest to see what happens when and get logs.