mainsail-crew/crowsnest #151: crowsnest: ERROR: No usable Devices Found. Stopping crowsnest.

seven-sam, 2023-07-24:
### What happened

crowsnet can't detect the camera csi

### What did you expect to happen

that the camera is working.

### How to reproduce

none it's the basic installation

### Additional information

I have a raspberry pi 4+raspberry pi camera csi with raspbian lite 32bits bullsey full upgrade with user: sevensam and not pi
I use mainsail install from kiauh, same for crowsnest
When I try to put the camera  on mainsail, it finds nothing.
I went to the logs and it says this: (basic settings installation)
INFO: Detect available Devices
[07/25/23 00:44:53] crowsnest: ERROR: No usable Devices Found. Stopping crowdsnest.
I tried port 8081 etc.. but if it can't find /dev/video0 it will be complicated I also tried to put direct from /dev/v4l/by-patch/***** (I tried everything) nothing works.

I ended up installing ustreamer from apt and I ran the following command line: sevensam@voron350:/dev $ ustreamer --device=/dev/video0 --host=192.168.1.20 --port=8083
and I can access the page to see the snaps or the camera stream and it works, I can even put it in mainsail (but it bugs)
then I cut CTRL+C and I only launched userstream and I started crowsnet and it worked only once, then nothing, impossible to recognize /dev/video0

blondsign, 2023-07-25:
got the same issue...

seven-sam, 2023-07-25:
ok i found

solutions for Raspberry pi Camera V2

disable camera in raspi-config (Interface - Legacy Camera : disable)
edit : /boot/config.txt
add under [all] : dtoverlay=imx219
reboot : sudo reboot
When restarting, type the command: libcamera-hello --list-cameras
result in my case :
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SBGGR10_CSI2P' : 640x480 [30.00 fps - (0, 0)/0x0 crop]
                             1640x1232 [30.00 fps - (0, 0)/0x0 crop]
                             1920x1080 [30.00 fps - (0, 0)/0x0 crop]
                             3280x2464 [30.00 fps - (0, 0)/0x0 crop]
           'SBGGR8' : 640x480 [30.00 fps - (0, 0)/0x0 crop]
                      1640x1232 [30.00 fps - (0, 0)/0x0 crop]
                      1920x1080 [30.00 fps - (0, 0)/0x0 crop]
                      3280x2464 [30.00 fps - (0, 0)/0x0 crop]
                      
take the line between brackets ( ) in my case : /base/soc/i2c0mux/i2c@1/imx219@10

edit crowsnest.conf change mode and device :

mode: camera-streamer
device: /base/soc/i2c0mux/i2c@1/imx219@10

and add your camera on mainsail 

PS : I found it all by myself here and there on the internet, but I just found it on the mainsail site https://crowsnest.mainsail.xyz/faq/how-to-setup-a-raspicam , but it's weird that I had trouble finding this info

Well that's solved for me.

mryel00, 2023-07-26:
@blondsign as @seven-sam already correct stated you can find the setup informations inside the [crowsnest documentation](https://crowsnest.mainsail.xyz/faq/how-to-setup-a-raspicam). The needed `/boot/config.txt` changes can be found under in the [tourbleshooting(https://crowsnest.mainsail.xyz/faq/how-to-setup-a-raspicam#troubleshooting) section. If you can't get it work, please join the [Mainsail Discord](https://discord.gg/mainsail) and open a post inside the support-forum.

closed 2023-07-26 by mryel00