mainsail-crew/crowsnest #152: Can't start at systemd

lyj0309, 2023-07-26:
### What happened

When i start the pc, the crowsnest.service  is failed ,and i can't see it in web

```
systemctl status crowsnest.service 
● crowsnest.service - crowsnest - Multi Webcam/Streamer Control Deamon
   Loaded: loaded (/etc/systemd/system/crowsnest.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2023-07-26 09:21:44 HKT; 6min ago
     Docs: https://github.com/mainsail-crew/crowsnest
  Process: 5917 ExecStart=/usr/local/bin/crowsnest $CROWSNEST_ARGS (code=exited, status=1/FAILURE)
 Main PID: 5917 (code=exited, status=1/FAILURE)

Jul 26 09:21:44 mkspi systemd[1]: crowsnest.service: Service RestartSec=5s expired, scheduling restart.
Jul 26 09:21:44 mkspi systemd[1]: crowsnest.service: Scheduled restart job, restart counter is at 10.
Jul 26 09:21:44 mkspi systemd[1]: Stopped crowsnest - Multi Webcam/Streamer Control Deamon.
Jul 26 09:21:44 mkspi systemd[1]: crowsnest.service: Start request repeated too quickly.
Jul 26 09:21:44 mkspi systemd[1]: crowsnest.service: Failed with result 'exit-code'.
Jul 26 09:21:44 mkspi systemd[1]: Failed to start crowsnest - Multi Webcam/Streamer Control Deamon.
```

### What did you expect to happen

see it in web.

### How to reproduce

```
cat /home/mks/printer_data/systemd/crowsnest.env
CROWSNEST_ARGS="-c /home/mks/printer_data/config/crowsnest.conf"
```
```
root@mkspi:/home/mks# cat /etc/systemd/system/crowsnest.service
#### crowsnest - A webcam Service for multiple Cams and Stream Services.
####
#### Written by Stephan Wendel aka KwadFan <me@stephanwe.de>
#### Copyright 2021 - 2022
#### https://github.com/mainsail-crew/crowsnest
####
#### This File is distributed under GPLv3
####

[Unit]
Description=crowsnest - Multi Webcam/Streamer Control Deamon
Documentation=https://github.com/mainsail-crew/crowsnest
After=udev.service network-online.target nss-lookup.target
Wants=udev.service network-online.target
Before=nginx.service
StartLimitBurst=10
StartLimitIntervalSec=180

[Install]
WantedBy=multi-user.target

[Service]
Type=simple
User=mks
RemainAfterExit=Yes
WorkingDirectory=/home/mks/crowsnest
EnvironmentFile=/home/mks/printer_data/systemd/crowsnest.env
ExecStart= /usr/local/bin/crowsnest $CROWSNEST_ARGS
Restart=on-failure
RestartSec=5
```

### Additional information

I can start it ok with manually on terminal , but can't start with systemd

start it with manually
[... 70 lines cut ...]
[07/26/23 09:35:33] crowsnest:          0: Disabled
[07/26/23 09:35:33] crowsnest:          1: 50 Hz
[07/26/23 09:35:33] crowsnest:          2: 60 Hz
[07/26/23 09:35:33] crowsnest:          white_balance_temperature 0x0098091a (int) : min=2800 max=6500 step=1 default=3980 value=3980 flags=inactive
[07/26/23 09:35:33] crowsnest:          sharpness 0x0098091b (int) : min=1 max=7 step=1 default=2 value=2
[07/26/23 09:35:33] crowsnest:          backlight_compensation 0x0098091c (int) : min=0 max=128 step=0 default=0 value=0
[07/26/23 09:35:33] crowsnest: 
[07/26/23 09:35:33] crowsnest:          Camera Controls
[07/26/23 09:35:33] crowsnest: 
[07/26/23 09:35:33] crowsnest:          auto_exposure 0x009a0901 (menu) : min=0 max=3 default=3 value=1
[07/26/23 09:35:33] crowsnest:          1: Manual Mode
[07/26/23 09:35:33] crowsnest:          3: Aperture Priority Mode
[07/26/23 09:35:33] crowsnest:          exposure_time_absolute 0x009a0902 (int) : min=10 max=626 step=1 default=156 value=626
[07/26/23 09:35:33] crowsnest: INFO: No usable CSI Devices found.
[07/26/23 09:35:33] crowsnest: V4L2 Control: Device: [cam 1]
[07/26/23 09:35:33] crowsnest: V4L2 Control: Options: --set-ctrl=rotate=180
grep: unrecognized option '--set-ctrl'
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
[07/26/23 09:35:34] crowsnest: V4L2 Control: Parameter '--set-ctrl=rotate=180' not available for '/dev/video4'. Skipped.
[07/26/23 09:35:34] crowsnest: Try to start configured Cams / Services...
[07/26/23 09:35:35] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[07/26/23 09:35:36] crowsnest: Starting ustreamer with Device /dev/video4 ...
[07/26/23 09:35:37] crowsnest: ... Done!
```

mryel00, 2023-07-27:
Your service and environment file looks like the default version to me. So there shouldn't be any problems with your specific files.
Please send the output of `journalctl -r -g crowsnest`.

Small Note: `--set-ctrl=rotate=180` shouldn't be like that inside the `v4l2ctl` section. Change it to `v4l2ctl: rotate=180`

lyj0309, 2023-08-02:
well, it's works when i update it to newest version

YaphetS1, 2023-11-06:
Same issue at the latest commit in master...

YaphetS1, 2023-11-06:
how to solve it?

mryel00, 2023-11-06:
@YaphetS1 Sry if that now sound rude, but what do you expect us to do? You gave no information, not even the information I asked OP for. For OP an update solved it, so that would be the answer for this specific case but this issue can have multiple different reasons.
Therefore: no information => no help possible
Also if you need help on this, you should join the [Discord](https://discord.gg/mainsail) as I already wrote in this issue.

YaphetS1, 2023-11-06:
The problem was in custom working directory, so the systemd output is (code=exited, status=200/CHDIR). 
That output was to look at /etc/systemd/system/crowsnest.service. When I changed "WorkingDirectory" to crowsnest directory, systemctl turned out to be satisfactory

YaphetS1, 2023-11-06:
It's strange. Why script doesn't update it by 'pwd'?

mryel00, 2023-11-22:
@YaphetS1 Sry, I completely forgot to answer you here. We decided against this, as it's less error prone like this.
But we added #205 recently. With this you can fix your WorkingDirectory easily. The intended usecase of this is for developing and to make it easier to let someone test something without the need of setting up multitple GitHub remote paths.