mainsail-crew/crowsnest #164: Two C270 Cameras Not Working: WATCHDOG: Lost Device

rullywowr, 2023-08-10:
### What happened

New installation of Fluidd/Moonraker/Klipper/Crowsnest on Raspberry Pi4 1GB.

Cannot get two C270 cameras working. Log file shows `WATCHDOG: Lost Device: 'usb-046d_0825_3227F860-video-index0'` on second camera immediately on boot.

Tried reinstalling. Switched USB ports. Rebooted many times. Tried delaying the systemctl service.Tried using MPEG on one camera and camera-streamer on other. Single camera works fine. Camera is operational.

Watching `tail -F crowsnest.log` shows the camera is lost immediately after ustreamer is started every time.

### What did you expect to happen

Expected to be able to use two cameras like before switching over to crowsnest.

### How to reproduce

Install Fluidd/Mainsail using KIAUH script. Setup crowsnest.cfg file.

### Additional information

[crowsnest.log](https://github.com/mainsail-crew/crowsnest/files/12308115/crowsnest.log)
[crowsnest.conf.txt](https://github.com/mainsail-crew/crowsnest/files/12308117/crowsnest.conf.txt)

rullywowr, 2023-08-10:
Also will add that xhci_hcd seems be resetting the USB device:

[Wed Aug  9 21:43:14 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 21:43:15 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[Wed Aug  9 21:43:18 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 21:43:22 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[Wed Aug  9 21:43:38 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[Wed Aug  9 21:49:35 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[Wed Aug  9 21:53:13 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 21:53:14 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[Wed Aug  9 21:54:52 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 21:54:52 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[Wed Aug  9 21:59:06 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 21:59:06 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[Wed Aug  9 21:59:11 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 22:13:36 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 22:13:36 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[Wed Aug  9 22:13:41 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 22:16:08 2023] usb 1-1.1: reset high-speed USB device number 3 using xhci_hcd
[Wed Aug  9 22:16:08 2023] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd

mryel00, 2023-08-10:
This seems like a hardware problem to me. Please join the [discord](https://discord.gg/mainsail) and open a post inside the support-forum, for further troubleshooting

closed 2023-08-10 by github-actions[bot]