mainsail-crew/crowsnest #166: libcamera.so.0.0.4: cannot open shared object file: No such file or directory

Kaelum, 2023-08-12:
### What happened

The latest update broke my Nozzle CAM.

### What did you expect to happen

It should work.

### How to reproduce

Updated from a previous version.

Kaelum, 2023-08-12:
Doing the following, fixed the issue after a restart:

```
cd ~/crowsnest
make buildclean
make build
```

mryel00, 2023-08-15:
Sry for the late reply, I somehow forgot to answer this issue.
You only need to rebuild camera-streamer, as this is a problem with building it manually, so nothing we can do about it directly atm.
The commands would be:
```bash
cd ~/crowsnest/bin/camera-streamer
make clean
make
```
If the `make` command fails for someone, try to use `make -j2` or `make -j1`.

closed 2023-08-15 by mryel00

othbert, 2023-09-10:
Also needed to do this on a fresh mainsailos image today. Thanks for the report and the fix!