mainsail-crew/crowsnest #182: BROKEN

figuerres, 2023-09-14:
### What happened

updating my mainsail os / crowsnest has broke my pi camera setup

### What did you expect to happen

for it to work

### How to reproduce

upgrade from the web ui on a pi 4 with a rpi camera

### Additional information

i am now down a rabbit hole of conflicting and confusing information on how to fix this mess.....
i now have no idea if this will ever work again..... and i do not want to re-do my whole install......

this is why some days i hate linux updates..... 

and discord is not a great place ... grrr.... 

suckage!!

closed 2023-09-14 by github-actions[bot]

meteyou, 2023-09-14:
sry, but we only do support on our discord server. to fix your issue, we need more informations. pls join our discord server and a helper will help you with your issue.

mryel00, 2023-09-14:
Discord might not be a place for everyone or the best place to give support, as you won't find the answers with google or another search engine, but it's a lot easier for use to communicate there and not having multiple issues open with multiple people writing that they got the same problem without reading through issues.

> i am now down a rabbit hole of conflicting and confusing information on how to fix this mess.....

You will find conflicting informations, as there are multiple reasons, why an update might have broke it. As I don't have any informations, I can just give some links to you.

As I don't know what MainsailOS version it could be that you got a version prior to 1.x or prior to 1.2. For prior to 1.x follow [this](https://crowsnest.mainsail.xyz/faq/use-legacy-branch-on-buster). If you are prior to 1.2, follow [this](https://crowsnest.mainsail.xyz/faq/upgrade-from-v3-to-v4) or the first link I sent.

> this is why some days i hate linux updates.....

As you wrote this specific line, it could be even that you are on 1.2.x, made system packages update and then the solution could be #166 

So as mete said, if you need further help please join the discord server.