mainsail-crew/crowsnest #184: Unable to install v4 on Debian 11 - Bullseye

bellyhold, 2023-09-26:
### What happened

The following packages have unmet dependencies:
 libcamera-apps-lite : Depends: libboost-program-options1.67.0 but it is not installable
E: Unable to correct problems, you have held broken packages.
Install dependencies ... [FAILED]

Nothing shows when running `sudo apt-mark showhold`. 

### What did you expect to happen

Install to succeed. 

### How to reproduce

Switch from legacy/v3 to master and run make install.

bellyhold, 2023-09-26:
Fixed with a full re-install of OS

closed 2023-09-26 by bellyhold

mryel00, 2023-09-26:
@bellyhold Just fyi and for everyone that is coming around to this post. Your installation was an in-place upgraded version of Buster and not a "real" Debian installation.
We won't give support for such a thing as even the Raspberry Pi Foundation is not recommending it, as you can read in their [documentation](https://www.raspberrypi.com/documentation/computers/os.html#upgrading-from-previous-operating-system-versions).
So your solution was already correct with a full reinstall of your OS.

wccrawford, 2023-10-06:
For anyone else experiencing this, I found that one of my apt sources files still referenced buster, instead of bullseye.  Updating that file and then upgrading the outdated packages let it get past this point.

Redoing the crowsnest install from scratch let me get the rest of the way through it.

meteyou, 2023-10-06:
@wccrawford this also sounds like an upgraded RPI installation from buster to bullseye. this is NOT recommended. pls read the post from @mryel00 .

wccrawford, 2023-10-06:
> @wccrawford this also sounds like an upgraded RPI installation from buster to bullseye. this is NOT recommended. pls read the post from @mryel00 .

Yeah, I saw it.  At the time, I thought it was a good alternative solution for existing systems, because I didn't want to go through all this yet again.  Since then, I have rebooted and found out why it's not a good solution.  So now I'm having to set up everything again anyhow.

iplayfast, 2023-10-12:
https://www.tomshardware.com/how-to/upgrade-raspberry-pi-os-to-bullseye-from-buster worked fine for me. Just verify that ALL the apt sources are changed from buster to bullseye

mryel00, 2023-10-12:
@iplayfast It can work, but most likely something is broken. Even if you update every single apt sources and everything still works after an update, you will end up with a broken libcamera setup, that you have to install for yourself and you will missing the device tree files for the raspicams. So no raspicams for you.
So yeah it might work for some users, but it won't for a lot others. An in-place upgrade can also end up in a complete broken state of your system.
With bookworm now released you should also not just upgrade your bullseye to bookworm. The network manager got switched and we will don't know what else might make problems with an in-place upgrade. Raspberry Pi Foundation will most likely not even give a guide this time.
And as a last point, your linked guide isn't different from the guide of the Raspberry Pi Foundation. I just didn't link it as I don't want people to follow it blindly, as it might break systems and will end up in broken libcamera.

wccrawford, 2023-10-12:
Maybe it *can* work, but it can definitely go wrong.  I'm sure I got all the references this last time (I grepped), but it left my system in a state that would *not* connect to a network.  Both wifi and ethernet were broken.

So in the end, I regret my "I can do it anyhow!" attitude, and join the devs here in recommending that people do *not* upgrade the OS in place.  It's not even the first time I've had problems going between versions of Linux.