mainsail-crew/crowsnest #188: Raspberry Pi OS Bookwork - Crowsnest Install Fails

4rianton, 2023-10-13:
### What happened

I am setting up a new machine and installing the essential programs using KIUAH on my newly flashed RPi Bookworm setup. 

I was successfully able to install Klipper, Moonraker, Mainsail, and Fluidd but crowsnest install fails. 

Here is the log: 

```
###### Cloning 'crowsnest' repository ...
Cloning into '/home/pi/crowsnest'...
remote: Enumerating objects: 1451, done.
remote: Counting objects: 100% (1451/1451), done.
remote: Compressing objects: 100% (550/550), done.
remote: Total 1451 (delta 884), reused 1431 (delta 877), pack-reused 0
Receiving objects: 100% (1451/1451), 561.39 KiB | 4.46 MiB/s, done.
Resolving deltas: 100% (884/884), done.

###### Checking for the following dependencies:
● git
● make
[✓ OK] Dependencies already met!
Installer will prompt you for sudo password!

###### Launching crowsnest installer ...
crowsnest - A webcam daemon for multiple cams and stream services.

	Ahoy!
	Thank you for installing crowsnest ;)
	This will take a while ...
	Please reboot after the installation has finished.

Running apt-get update first ...

Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Reading package lists...
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Running apt-get update first ... [OK]
Installing dependencies ...

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
crudini is already the newest version (0.9.4-1).
bsdutils is already the newest version (1:2.38.1-5).
findutils is already the newest version (4.9.0-4).
v4l-utils is already the newest version (1.22.1-5).
curl is already the newest version (7.88.1-10+rpi1).
build-essential is already the newest version (12.9).
libevent-dev is already the newest version (2.1.12-stable-8).
libjpeg-dev is already the newest version (1:2.1.5-2).
libbsd-dev is already the newest version (0.11.7-2).
cmake is already the newest version (3.25.1-1).
libavformat-dev is already the newest version (8:5.1.3-1+rpt4).
libavutil-dev is already the newest version (8:5.1.3-1+rpt4).
libavcodec-dev is already the newest version (8:5.1.3-1+rpt4).
libcamera-dev is already the newest version (0.1.0+pi5~0-1).
libcamera-apps-lite is already the newest version (1.2.1-1+bookworm+1).
liblivemedia-dev is already the newest version (2020.01.19-1+build1).
pkg-config is already the newest version (1.8.1-1).
xxd is already the newest version (2:9.0.1378-2).
libssl-dev is already the newest version (3.0.10-1~deb12u1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
Install dependencies ... [OK]
Reading configuration ...
No user configuration file found ...
Using default configuration ... [OK]
Creating file structure ...
[... 629 lines cut ...]
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/websocketserver.cpp.o
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/wstransport.cpp.o
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/wshandshake.cpp.o
[100%] Linking CXX static library libdatachannel-static.a
make[5]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
[100%] Built target datachannel-static
make[4]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[3]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[2]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
rm device/buffer_lock.o device/v4l2/buffer.o output/http_hls.o device/camera/camera_output.o device/v4l2/buffer_list.o device/buffer_queue.o device/buffer.o device/camera/camera_decoder.o device/links.o cmd/list-devices/main.o device/v4l2/debug.o device/camera/camera_input.o device/v4l2/device_options.o output/http_ffmpeg.o device/camera/camera.o device/camera/camera_rescaller.o device/libcamera/fake_camera.o cmd/camera-streamer/main.o device/buffer_list.o device/camera/camera_debug.o util/http/http.o device/camera/camera_pipeline.o device/dummy/buffer_list.o output/http_jpeg.o cmd/camera-streamer/opts.o output/output.o device/v4l2/device_list.o device/device_list.o device/v4l2/device_media.o device/dummy/device.o device/dummy/dummy.o util/http/http_methods.o device/dummy/buffer.o cmd/camera-streamer/http.o device/camera/camera_isp.o device/v4l2/v4l2.o device/device.o device/v4l2/device.o output/http_h264.o
make[1]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer'
make: *** [Makefile:62: all] Error 2
make: *** [Makefile:17: install] Error 2
>>>>>> Something went wrong! Please try again...
```

### What did you expect to happen

that crowsnest installs without issue. 

I have tried: reinstalling, deleting the installation, rebooting the pi, and the usual troubleshooting. 

### How to reproduce

have Raspberry Pi OS Full Bookworm installation, install crowsnest, installation fails.

mryel00, 2023-10-13:
The problem is about camera-streamer not building. Crowsnest is still using an old branch of camera-streamer atm. That branch don't get any updates. Also the newest version of camera-streamer doesn't build atm under bookworm too, as you can read in this issue https://github.com/ayufan/camera-streamer/issues/107.
So the current options we offer would be to stay on bullseye or try to use the legacy branch like described [here](https://crowsnest.mainsail.xyz/faq/use-legacy-branch-on-buster) (not tested under bookworm).

ntressler7, 2023-10-14:
I was experiencing the same issue that @4rianton described. I tried everything to get Crowsnest v4 to work with no luck. I ultimately [installed v3](https://crowsnest.mainsail.xyz/faq/use-legacy-branch-on-buster) and can confirm that it works with Raspberry Pi OS 12 - Bookworm.

adrianstop, 2023-10-15:
I had the same issue as @4rianton, but unlike @ntressler7 I couldn't get the legacy/v3 branch to work with my setup either. Would love for an update to ensure compatibility with Raspi OS Bookworm :)

KwadFan, 2023-10-16:
Well,

This is influenced by my personal experience and thoughts about that.
I know users, especcially new users, grabbing always the latest Software and expect **not** to face any issue.

So, here are my thoughts about that. You can't expect to have a flawless experience if an underlaying OS changes that drastically, like Raspberry Pi OS did.

The Major changes that are affecting "us" as a 3D Printer Community is
A) Breaking Changes in Libcamera, which camera-streamer heavily relies on.
B) Choices that Devs to leave X11 and replace it with Wayland. This might/will have a huge impact on KlipperScreen,
which by itself always was an 'enemy' against crowsnest because it, in some combinations, lead to more issues than I want to handle/want users to handle.

That said, I highly encurage you to use Bullseye as long as possible.
We (and I as Core Maintainer of MainsailOS and Crowsnest) will not rush here to get compatibilty, only to have it.
We ( means I ) will do my best to provide support for it, in the long run, because also MainsailOS has to switch at some point. **But** this causes more trouble and frustration about choices made by others, than it is fun.

The switch from buster to bullseye was kinda disaster or disastrous for some crowds ot there, that I am not willing to do this step another time.

Therefore I have to apologize that it isnt working that great currently, but we do this all in our free time to enjoy our selfs or learning something new. This isnt a paid job and even it seems crowsnest is the only valid option for our community, @mryel00 and I do our best to put all that developing thingys between a Dayjob, Family and Friends.
 
 I for myself will stay as long as possible on bullseye based OSes because, I dont like the desitions are made by Raspberry Foundation. Bookworm kills every SBC that is older than a 3B+. On other Models is either the cpu not powerful enough or its lacking of RAM. 512M is to less for bookworm, promised!

To get to an end and not ranting further here, we will provide functionality in Bookworm, but earliest will be to end of this year because there are to many moving parts that has to be addressed first. I cant provide Bookworm support if that means breaking changes for all other users that are not on bookworm. Also I need time for it which I am lacking of currently due massive changes in my life.
As said, you will get support for bookworm but not in rush. I hope you understand my POV and desitions.

Thanks for you patience, regards Kwad

ayufan, 2023-10-17:
The bookworm is supported by `camera-streamer` if installed via prebuilt: https://github.com/ayufan/camera-streamer/releases/tag/v0.2.7.

mryel00, 2023-10-24:
> how can i make this work to be recognized by crowsnest?

Just install crowsnest like normal, after that is finished (after failing to build camera-streamer), you can just install camera-streamer with the prebuilt package and then create a symlink to `~/crowsnest/bin/camera-streamer/camera-streamer`.
So I think the correct command would be `ln -s /usr/bin/camera-streamer ~/crowsnest/bin/camera-streamer/camera-streamer` to link the installation.
Another option would be to use PR #137 or the duplicate #191. Using the PRs won't make it possible to update automatically, but crowsnest would be able to build camera-streamer (at least a high chance that it will work). Also I mentioned some reasons inside both PRs, why we won't just update the branch.
One last thing I should mention is that the copy-paste command of camera-streamer is currently broke. So there will be some problems, if you just want to copy and paste it.

> the build script seems to do statically without checking if it is already installed/available.

The installation should be only done once or maybe again when you have some weird errors. So there is no reason to check for already available installations. A check like this will only introduce more error than it's of any use.

Surion79, 2023-10-24:
> Just install crowsnest like normal, after that is finished (after failing to build camera-streamer), you can just install camera-streamer with the prebuilt package and then create a symlink to [...]

I looked at the PR and while checking the commit, it seems that the multiple references to the camera-streamer repo is not unified. some point to the -research account and some not :) Guess the last one wins all

pmaersk, 2023-10-25:
I just tried "sudo make install" on Raspberry newest updated "bookworm" and as described it failed building the camera-streamer. However the fix was easy.

The file ~/crowsnest/bin/camera-streamer/util/ffmpeg/remuxer.c has in line 97 the following:

  AVInputFormat *input_format = av_find_input_format(remuxer->input_format);

However since av_find_input_format() is defined as returning a pointer to a "const"

/usr/include/aarch64-linux-gnu/libavformat/avformat.h:const AVInputFormat *av_find_input_format(const char *short_name);

and since the makefile compiles with treat warnings as errors, you need to prepend a "const" to the declaration in that line 97 as shown below:
 
  const AVInputFormat *input_format = av_find_input_format(remuxer->input_format);

I know it is a problem for camera_streamer, but I thought some here may like to know how to fix it.

Regards
Peter Maersk-Moller

gjsmo, 2023-10-31:
For anyone else having this same issue, a better solution I have found is to point the installer to the `main` repo of camera-streamer. Instead of modifying the source code, this will pull the latest version, which compiles without issue. To do so, make a file tools/.config with the following content:

```
CROWSNEST_CAMERA_STREAMER_REPO_BRANCH="main"
```

Surion79, 2023-10-31:
The camer-streamer repo pointed to is from a secondary account of the creator (-research), also in at least one of the PRs it was mentioned, that changing the branch to a newer version would require additional camera-streamer settings, which are not in scope of crowsnest. (Some sort of IP stream access list)But such issues would also create issues here and also take a lot of free time of the crew.Additionally the build process of even the ‚newest’ camerastreamer has a ton of security related warnings (deprecation since SSLv3), so the other option would be to go with crowsnest v3 as stated from a different comment in this issue. Updating the base OS without checking compatibility is not even close to best practice.For those people i think MainsailOS is the correct way to go.

gjsmo, 2023-11-01:
@Surion79 The repo I am using is the official [ayufan/camera-streamer](https://github.com/ayufan/camera-streamer), however if you do try to go to ayufan-research/camera-streamer you'll find that it's a redirect to the first. As far as options for camera-streamer, I'm sure there's something different but it works just fine on multiple installations.

Regarding the OS version, this isn't necessarily a big deal for people who have a running printer, but it's a very big deal for people installing for the first time. Stock Raspberry Pi OS + KIAUH is a very common installation method, and as of right now that gets you Debian 12 and no working Crowsnest. This is a very bad user experience, especially if the advice is to either install an older version manually or just wait.

mryel00, 2023-11-01:
> The repo I am using is the official [ayufan/camera-streamer](https://github.com/ayufan/camera-streamer), however if you do try to go to ayufan-research/camera-streamer you'll find that it's a redirect to the first.

That's right and this will be fixed with the next update. This was just some oversite but shouldn't change anything in the result.

> but it's a very big deal for people installing for the first time. Stock Raspberry Pi OS + KIAUH is a very common installation method, and as of right now that gets you Debian 12 and no working [...]

We know that it's a bad user experience and it's currently not optimal, especially for non techies. But you have to see it from our viewpoint. We don't want to push anything that could introduce bugs for new users and the same goes for old users. So we have to test every change in the install workflow extensively, as we got 3 different methods of installing:

1. The normal way of cloning+`sudo make install`
2. KIAUH should be nearly the same as 1, but should be tested too, to be sure that there are no weird interactions
3. CustomPiOS module is a bit more complex and can make problems pretty fast

> As far as options for camera-streamer, I'm sure there's something different but it works just fine on multiple installations.

Just an overall change to main branch can introduce problems with parameters as I already mentioned in #137. If we would change it, there might be conflicts with the old version, if we want to provide the "correct" and "full" functionality for both versions. So a quick change of the branch isn't an option for us.

This is opensource, so everyone can contribute to it (at least in a way that suits our plans/ideas) or make their own fork. We are providing this for free and work on this in our free time. We know that this all could be already tested a lot sooner but we had other plans at first that didn't go as we wanted because of some personal stuff.

So @gjsmo as I already told you in #137, I'm already working on it and please just have patience. I know  that I don't have to answer on everything coming here, but I feel like some people don't know what we have to consider and test just to make a small patch like this.
For completion I will share my statement about the small patch I'm currently working on here too:
> There will be soon an update to main branch, but Bookworm exclusive at first, as Crowsnest fails to build on Bookworm with master branch (https://github.com/mainsail-crew/crowsnest/issues/188). As we (more like me) currently think that the Bookworm Raspberry Pi image is more like in a testing phase for the software inside the klipper universe.
So we will only update it for Bookworm that those people with Bookworm can use Crowsnest again. The Bookworm people will be like a testing group for this. We won't release it for Bullseye as we want to do enough testing on it first that we can assure that there aren't any issues like breaking changes or maybe missing features.

P.S. I'm basically doing the last few tests right now

mryel00, 2023-11-03:
#195 is now merged into master. So installing on Bookworm should now be possible with kiauh or manually

closed 2023-11-03 by mryel00

nogoer, 2025-06-25:
this appears to have resurfaced. I cant install crowsnest via kiauh or manual due to complile errors for missing RTSPServerSupportingHTTPStreaming.hh

.....g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/buffer.o device/libcamera/buffer.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/buffer_list.o device/libcamera/buffer_list.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/device.o device/libcamera/device.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/libcamera.o device/libcamera/libcamera.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/options.o device/libcamera/options.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o output/rtsp/rtsp.o output/rtsp/rtsp.cc
output/rtsp/rtsp.cc:28:10: fatal error: RTSPServerSupportingHTTPStreaming.hh: No such file or directory
   28 | #include <RTSPServerSupportingHTTPStreaming.hh>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:106: output/rtsp/rtsp.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer'
make: *** [Makefile:67: all] Error 2
make: *** [Makefile:19: install] Error 2

PR #195 (2023-11-01): fix: add bookworm support
Fixes #188 
This will force the main branch of camera-streamer on Bookworm installations.
Therefore it will ignore the branch specified inside `tools/.config` on Bookworm installations.