mainsail-crew/crowsnest #190: Multiple camera not detecting

NV4RE, 2023-10-22:
### What happened

When adding multiple cameras (both on the same model), only one detects

crowsnest only detect one camera

```
...

[10/23/23 02:18:42] crowsnest: INFO: Detect available Devices
[10/23/23 02:18:42] crowsnest: INFO: Found 1 total available Device(s)
[10/23/23 02:18:43] crowsnest: INFO: Found 1 available v4l2 (UVC) camera(s)
...
```

Hardware
```
BTT manta m8p (v1.0)
BTT CB1
Camera: ashu h800
```

software
```
beacon
 v1.0.0-7 > v1.0.0-8
crowsnest
v4.0.4-6-g767c53aa
klipper
v0.11.0-299-gb1f597c5
Klipper-Adaptive-Meshing-Purging
v1.1.2-4-g2389a994
KlipperScreen
v0.3.6-50-gc14fd4d2
mainsail
v2.8.0
mainsail-config
v1.0.0-16-gc64d3af9
moonraker
v0.8.0-186-g2641fc54
sonar
v0.1.2-0-g7962a122
timelapse
v0.0.1-142-g57bec0de
System
OS-Packages
```

lsusb
```sh
lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 011: ID 1d50:614e OpenMoko, Inc. stm32g0b1xx
Bus 002 Device 007: ID 1a86:e5e3 QinHeng Electronics USB2IIC_CTP_CONTROL
Bus 002 Device 010: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 002 Device 006: ID 0806:0806 SJ-180517-N 1080P Webcam
Bus 002 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 002 Device 003: ID 0806:0806 SJ-180517-N 1080P Webcam
Bus 002 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

v4l2-ctl --list-devices
```
[... 171 lines cut ...]
[10/23/23 02:18:43] crowsnest:		Camera Controls
[10/23/23 02:18:43] crowsnest:		
[10/23/23 02:18:43] crowsnest:		auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
[10/23/23 02:18:43] crowsnest:		1: Manual Mode
[10/23/23 02:18:43] crowsnest:		3: Aperture Priority Mode
[10/23/23 02:18:43] crowsnest:		exposure_time_absolute 0x009a0902 (int)    : min=78 max=10000 step=1 default=312 value=350
[10/23/23 02:18:43] crowsnest: Try to start configured Cams / Services...
[10/23/23 02:18:46] crowsnest: INFO: Configuration of Section [cam top] looks good. Continue ...
[10/23/23 02:18:48] crowsnest: INFO: Configuration of Section [cam rear] looks good. Continue ...
[10/23/23 02:18:50] crowsnest: V4L2 Control: Device: [cam top]
[10/23/23 02:18:50] crowsnest: V4L2 Control: Options: saturation=75,hue=0,auto_exposure=1,exposure_time_absolute=350,brightness=2,contrast=1,white_balance_automatic=0,white_balance_temperature=4200
[10/23/23 02:18:51] crowsnest: V4L2 Control: Device: [cam rear]
[10/23/23 02:18:51] crowsnest: V4L2 Control: Options: saturation=75,hue=0,auto_exposure=1,exposure_time_absolute=350,brightness=2,contrast=1,white_balance_automatic=0,white_balance_temperature=4200
[10/23/23 02:18:55] crowsnest: Starting ustreamer with Device /dev/video0 ...
[10/23/23 02:18:55] crowsnest: Starting ustreamer with Device /dev/video2 ...
[10/23/23 02:18:59] crowsnest:  ... Done!
```

### What did you expect to happen

Detect both camera

### How to reproduce

Just use multiple camera

NV4RE, 2023-10-22:
Possible related to https://github.com/mainsail-crew/crowsnest/issues/109

mryel00, 2023-10-22:
@NV4RE It's not related to the issue you mentioned. The issue you mentioned is about connection problems with a cb1. Your issue is just about crowsnest.log not showing the second **_identical_** camera. Crowsnest currently only shows the entry of `/dev/v4l/by-id/`. The problem about two identical cams is, that it only shows one of them there. So please follow the [documentation](https://crowsnest.mainsail.xyz/faq/how-to-use-identical-usb-cam-models-twice) on using two identical cameras.
So this isn't a bug, but more like a "missing" feature that doesn't have that much priority as the amount of users with two identical cams are pretty low. If you use two unidentical USB cams, you will see that both should get printed inside the log.

mryel00, 2023-10-23:
I updated this issue to represent a feature request. Fyi this feature is internally already under development.

mryel00, 2024-04-02:
@NV4RE I need some information for this issue to tackle it correctly.
I need the output of `http://<ip.of.your.pi>:<moonraker_port>/machine/peripherals/video`. `<moonraker_port` is by default `7125` and can be seen inside your `moonraker.conf`. If that url doesn't work, make sure you are on the latest moonraker version.

NV4RE, 2024-04-03:
> @NV4RE I need some information for this issue to tackle it correctly. I need the output of following url `http://<ip.of.your.pi>:<moonraker_port>/machine/peripherals/video`. `<moonraker_port` is by default `7125` and can be seen inside your `moonraker.conf`. If that url doesn't work, make sure you are on the latest moonraker version.

Hi, thanks for the follow-up. I apologize that I can't be of more help in testing the fix right now. I actually removed all the webcams from the machine a while back, so I can't send you the output or confirm whether the issue is resolved.

However, I wanted to provide some additional information that might be helpful. Since the last time I interacted with this issue, I tried using multiple webcams with the machine, but from different models and brands. Unfortunately, the problem persisted regardless of the webcams used. This leads me to believe that the root cause might be related to the bandwidth limitations of the BTT Manta M8P's USB chip. It's possible it doesn't have enough capacity to handle multiple webcams efficiently.

NV4RE, 2024-04-06:
> @NV4RE I need some information for this issue to tackle it correctly. I need the output of following url `http://<ip.of.your.pi>:<moonraker_port>/machine/peripherals/video`. `<moonraker_port` is by [...]

Hello, I've reinstalled the webcam, but the problem persists. Here's the output of `/machine/peripherals/video`

```json
{
  "result": {
    "v4l2_devices": [
      {
        "device_name": "video0",
        "device_path": "/dev/video0",
        "camera_name": "1080P Webcam: 1080P Webcam",
        "driver_name": "uvcvideo",
        "hardware_bus": "usb-5200000.usb-1.1.2",
        "capabilities": ["VIDEO_CAPTURE", "EXT_PIX_FORMAT", "STREAMING"],
        "version": "5.16.17",
        "path_by_hardware": "/dev/v4l/by-path/platform-5200000.usb-usb-0:1.1.2:1.0-video-index0",
        "path_by_id": null,
        "alt_name": "1080P Webcam: 1080P Webcam",
        "usb_location": "2:5",
        "modes": [
          {
            "format": "MJPG",
            "description": "Motion-JPEG",
            "flags": ["COMPRESSED"],
            "resolutions": [
              "640x480",
              "1920x1080",
              "1280x720",
              "320x240",
              "1024x768",
              "1280x1024",
              "160x120",
              "800x600"
            ]
          },
          {
            "format": "YUYV",
            "description": "YUYV 4:2:2",
            "flags": [],
            "resolutions": [
              "640x480",
              "1920x1080",
              "1280x720",
              "320x240",
              "1024x768",
              "1280x1024",
              "160x120",
              "800x600"
            ]
          }
        ]
      },
      {
        "device_name": "video2",
        "device_path": "/dev/video2",
        "camera_name": "1080P Webcam: 1080P Webcam",
        "driver_name": "uvcvideo",
        "hardware_bus": "usb-5200000.usb-1.1.4",
        "capabilities": ["VIDEO_CAPTURE", "EXT_PIX_FORMAT", "STREAMING"],
        "version": "5.16.17",
        "path_by_hardware": "/dev/v4l/by-path/platform-5200000.usb-usb-0:1.1.4:1.0-video-index0",
        "path_by_id": "/dev/v4l/by-id/usb-SJ-180517-N_1080P_Webcam-video-index0",
        "alt_name": "1080P Webcam: 1080P Webcam",
        "usb_location": "2:16",
        "modes": [
          {
            "format": "MJPG",
            "description": "Motion-JPEG",
            "flags": ["COMPRESSED"],
[... 199 lines cut ...]
[04/06/24 23:42:49] crowsnest:		1: 50 Hz
[04/06/24 23:42:49] crowsnest:		2: 60 Hz
[04/06/24 23:42:49] crowsnest:		white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4200
[04/06/24 23:42:49] crowsnest:		sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=1
[04/06/24 23:42:49] crowsnest:		backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=0 value=3
[04/06/24 23:42:49] crowsnest:		
[04/06/24 23:42:49] crowsnest:		Camera Controls
[04/06/24 23:42:49] crowsnest:		
[04/06/24 23:42:49] crowsnest:		auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
[04/06/24 23:42:49] crowsnest:		1: Manual Mode
[04/06/24 23:42:49] crowsnest:		3: Aperture Priority Mode
[04/06/24 23:42:49] crowsnest:		exposure_time_absolute 0x009a0902 (int)    : min=78 max=10000 step=1 default=312 value=350
[04/06/24 23:42:49] crowsnest: Try to start configured Cams / Services...
[04/06/24 23:42:52] crowsnest: INFO: Configuration of Section [cam rear] looks good. Continue ...
[04/06/24 23:42:55] crowsnest: INFO: Configuration of Section [cam test] looks good. Continue ...
[04/06/24 23:42:56] crowsnest: V4L2 Control: Device: [cam rear]
[04/06/24 23:42:56] crowsnest: V4L2 Control: Options: pixelformat=YUYV,saturation=75,hue=-1800,auto_exposure=1,exposure_time_absolute=350,brightness=-6,contrast=1,white_balance_automatic=0,white_balance_temperature=4200,backlight_compensation=3,sharpness=1
[04/06/24 23:42:56] crowsnest: V4L2 Control: Parameter 'pixelformat=YUYV' not available for '/dev/v4l/by-path/platform-5200000.usb-usb-0:1.1.2:1.0-video-index0'. Skipped.
[04/06/24 23:42:57] crowsnest: V4L2 Control: Device: [cam test]
[04/06/24 23:42:57] crowsnest: V4L2 Control: Options: pixelformat=YUYV,saturation=75,hue=-1800,auto_exposure=1,exposure_time_absolute=350,brightness=-6,contrast=1,white_balance_automatic=0,white_balance_temperature=4200,backlight_compensation=3,sharpness=1
[04/06/24 23:42:57] crowsnest: V4L2 Control: Parameter 'pixelformat=YUYV' not available for '/dev/v4l/by-path/platform-5200000.usb-usb-0:1.1.4:1.0-video-index0'. Skipped.
[04/06/24 23:43:02] crowsnest: Starting ustreamer with Device /dev/v4l/by-path/platform-5200000.usb-usb-0:1.1.4:1.0-video-index0 ...
[04/06/24 23:43:02] crowsnest: Starting ustreamer with Device /dev/v4l/by-path/platform-5200000.usb-usb-0:1.1.2:1.0-video-index0 ...
[04/06/24 23:43:05] crowsnest:  ... Done!
```

mryel00, 2024-04-06:
Thanks for the output.
I still didn't implement anything for this. I needed the output to even have a starting point.

Drizzt321, 2024-04-08:
I'm facing something similar. I tried using the `by-path`, as per https://crowsnest.mainsail.xyz/faq/how-to-use-identical-usb-cam-models-twice, but after editing (via Mainsail, verifying over SSH the changes were saved), and reboot, the `crowsnest.conf` is reset to using `/dev/videoN` for my cameras. NOT the `by-path`.  It's super weird, it just automatically resets it.

Although I notice above the bandwidth issue...which I think will be an issue for me as well. Thinking I might need to use an external RPi, but any way I can have crowsnest reference an external source of the video, rather than direct attach USB cameras? Or should I just redo everything, put Mainsail/etc on an RPi or other similar, and use CB1+Manta M8P with the base Klipper/moonraker?

mryel00, 2024-04-08:
@Drizzt321 Please stay on the Discord as you I'm the person helping here and on the server. Your issue is different. Here is no place for support, it's just there to identify an issue with the code and not with your setup.

jeffalanmcclain, 2025-01-23:
Hello, have similar problem.  It is sort of random on which of the two IDENTICAL cameras are detected.  I can unplug ONE or THE OTHER and get each one working...but crowsnest always just detects ONE...  Here is my output from the moonraker reques:
```{"result": {"v4l2_devices": [{"device_name": "video0", "device_path": "/dev/video0", "camera_name": "TSTC USB20 WEB CAMERA: TSTC USB", "driver_name": "uvcvideo", "hardware_bus": "usb-fe9c0000.xhci-1.1", "capabilities": ["VIDEO_CAPTURE", "EXT_PIX_FORMAT", "STREAMING"], "version": "6.6.62", "path_by_hardware": "/dev/v4l/by-path/platform-fe9c0000.xhci-usb-0:1.1:1.0-video-index0", "path_by_id": "/dev/v4l/by-id/usb-TSTC_USB20_WEB_CAMERA_TSTC_USB20_WEB_CAMERA_01.00.00-video-index0", "alt_name": "TSTC USB20 WEB CAMERA: TSTC USB", "usb_location": "1:10", "modes": [{"format": "MJPG", "description": "Motion-JPEG", "flags": ["COMPRESSED"], "resolutions": ["1920x1080", "1280x960", "1280x720", "1024x768", "800x600", "640x480", "320x240"]}, {"format": "YUYV", "description": "YUYV 4:2:2", "flags": [], "resolutions": ["1920x1080", "1280x960", "1280x720", "1024x768", "800x600", "640x480", "320x240"]}]}, {"device_name": "video2", "device_path": "/dev/video2", "camera_name": "TSTC USB20 WEB CAMERA: TSTC USB", "driver_name": "uvcvideo", "hardware_bus": "usb-fe9c0000.xhci-1.3", "capabilities": ["VIDEO_CAPTURE", "EXT_PIX_FORMAT", "STREAMING"], "version": "6.6.62", "path_by_hardware": "/dev/v4l/by-path/platform-fe9c0000.xhci-usb-0:1.3:1.0-video-index0", "path_by_id": null, "alt_name": "TSTC USB20 WEB CAMERA: TSTC USB", "usb_location": "1:12", "modes": [{"format": "MJPG", "description": "Motion-JPEG", "flags": ["COMPRESSED"], "resolutions": ["1920x1080", "1280x960", "1280x720", "1024x768", "800x600", "640x480", "320x240"]}, {"format": "YUYV", "description": "YUYV 4:2:2", "flags": [], "resolutions": ["1920x1080", "1280x960", "1280x720", "1024x768", "800x600", "640x480", "320x240"]}]}, {"device_name": "video14", "device_path": "/dev/video14", "camera_name": "bcm2835-isp", "driver_name": "bcm2835-isp", "hardware_bus": "platform:bcm2835-isp", "capabilities": ["VIDEO_CAPTURE", "EXT_PIX_FORMAT", "STREAMING"], "version": "6.6.62", "path_by_hardware": "/dev/v4l/by-path/platform-bcm2835-isp-video-index1", "path_by_id": null, "alt_name": "bcm2835-isp-capture0", "usb_location": null, "modes": []}, {"device_name": "video15", "device_path": "/dev/video15", "camera_name": "bcm2835-isp", "driver_name": "bcm2835-isp", "hardware_bus": "platform:bcm2835-isp", "capabilities": ["VIDEO_CAPTURE", "EXT_PIX_FORMAT", "STREAMING"], "version": "6.6.62", "path_by_hardware": null, "path_by_id": null, "alt_name": "bcm2835-isp-capture1", "usb_location": null, "modes": []}, {"device_name": "video21", "device_path": "/dev/video21", "camera_name": "bcm2835-isp", "driver_name": "bcm2835-isp", "hardware_bus": "platform:bcm2835-isp", "capabilities": ["VIDEO_CAPTURE", "EXT_PIX_FORMAT", "STREAMING"], "version": "6.6.62", "path_by_hardware": null, "path_by_id": null, "alt_name": "bcm2835-isp-capture0", "usb_location": null, "modes": []}, {"device_name": "video22", "device_path": "/dev/video22", "camera_name": "bcm2835-isp", "driver_name": "bcm2835-isp", "hardware_bus": "platform:bcm2835-isp", "capabilities": ["VIDEO_CAPTURE", "EXT_PIX_FORMAT", "STREAMING"], "version": "6.6.62", "path_by_hardware": "/dev/v4l/by-path/platform-bcm2835-isp-video-index2", "path_by_id": null, "alt_name": "bcm2835-isp-capture1", "usb_location": null, "modes": []}], "libcamera_devices": [{"libcamera_id": "/base/scb/xhci@7e9c0000-1.1:1.0-1bcf:2cd1", "model": "TSTC USB20 WEB CAMERA: TSTC USB", "modes": [{"format": "MJPEG", "resolutions": ["320x240", "640x480", "800x600", "1024x768", "1280x720", "1280x960", "1920x1080"]}, {"format": "YUYV", "resolutions": ["320x240", "640x480", "800x600", "1024x768", "1280x720", "1280x960", "1920x1080"]}]}, {"libcamera_id": "/base/scb/xhci@7e9c0000-1.3:1.0-1bcf:2cd1", "model": "TSTC USB20 WEB CAMERA: TSTC USB", "modes": [{"format": "MJPEG", "resolutions": ["320x240", "640x480", "800x600", "1024x768", "1280x720", "1280x960", "1920x1080"]}, {"format": "YUYV", "resolutions": ["320x240", "640x480", "800x600", "1024x768", "1280x720", "1280x960", "1920x1080"]}]}]}}```

and my logfile:
```crowsnest - A webcam Service for multiple Cams and Stream Services.
[01/23/25 16:00:35] crowsnest: Version: v4.1.10-1-gdd390b6
[01/23/25 16:00:35] crowsnest: Prepare Startup ...
[01/23/25 16:00:35] crowsnest: INFO: Host information:
[01/23/25 16:00:35] crowsnest: Host Info: Distribution: Debian GNU/Linux 12 (bookworm)
[01/23/25 16:00:35] crowsnest: Host Info: Kernel: Linux 6.6.62+rpt-rpi-v8 aarch64
[01/23/25 16:00:35] crowsnest: Host Info: Model:  Raspberry Pi Compute Module 4 Rev 1.1
[01/23/25 16:00:35] crowsnest: Host Info: Available CPU Cores: 4
[01/23/25 16:00:35] crowsnest: Host Info: Available Memory: 3831820 kB
[01/23/25 16:00:35] crowsnest: Host Info: Diskspace (avail. / total): 14G / 29G
[01/23/25 16:00:35] crowsnest: INFO: Checking Dependencies
[01/23/25 16:00:35] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[01/23/25 16:00:35] crowsnest: Dependency: 'find' found in /usr/bin/find.
[01/23/25 16:00:35] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[01/23/25 16:00:35] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[01/23/25 16:00:35] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[... cut ...]
[01/23/25 16:00:35] crowsnest:		custom_flags: --camera-format=YUYV
[01/23/25 16:00:35] crowsnest:		
[01/23/25 16:00:35] crowsnest:		
[01/23/25 16:00:35] crowsnest: INFO: Detect available Devices
[01/23/25 16:00:35] crowsnest: INFO: Found 1 total available Device(s)
[01/23/25 16:00:35] crowsnest: INFO: Found 1 available v4l2 (UVC) camera(s)
[01/23/25 16:00:35] crowsnest: /dev/v4l/by-id/usb-TSTC_USB20_WEB_CAMERA_TSTC_USB20_WEB_CAMERA_01.00.00-video-index0 -> /dev/video0
[01/23/25 16:00:35] crowsnest: Supported Formats:```

mryel00, 2025-01-23:
Thanks for the output of the moonraker API endpoint, but for fixing it you should follow the [docs](https://crowsnest.mainsail.xyz/faq/how-to-use-identical-usb-cam-models-twice). I only needed at least some API endpoint output that I have some point of reference for the implementation. I only needed it once.
The correct logging will be implented with https://github.com/mainsail-crew/crowsnest/pull/270 but I didn't do it yet.

jeffalanmcclain, 2025-01-23:
Hi...thanks.  I followed the /dev/by-path/... info and that also does not work.  I get ONE view, while the other just says "connecting" and crowsnest log still shows only one device.  if I swap USB ports they are plugged into, then I can get the other one to show first, while the other one just sits and says "connecting..." but never does.  I've tried ONE on ustreamer and one camera-streamer, tried both on camera-streamer, tried both on ustreamer, rebooted multiple times, tried one as YUYV and one as JPEG, both at JPEG, both as YUYV...nothing works to get both working.  I can *easily* get two working at same time if I plug in a different camera (Logitech C920).  And actually, two logitech C920 at same time works fine, even without using the /dev/by-path/

mryel00, 2025-01-23:
For support, please join our [Discord](https://discord.gg/mainsail) or use the Github Discussions of the Mainsail-Crew.

mitaipyo, 2025-04-26:
Identical USB cameras streaming is possible in Crowsnest. This took a lot of tries but got it working like this. 
First.  SSH into your device and apt update, apt upgrade all. Make sure your device has the correct time. 
Connect USB cams. Under /dev/ you should see both USB cams as video0 and video1 ( it is easier to get there with WinSCP) 
Now, in crowsnest.cfg add your cam 1 section.  I did not use the id-by-path method. Here is mine: 

[cam 1]
mode: ustreamer                         
enable_rtsp: false                      
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG stream/snapshot port
device: /dev/video1                    #video1 !
resolution: 1920x1080                     #NO SPACE before/after the x!
max_fps: 30                            
#custom_flags:                          
#v4l2ctl:                              

Save and restart. I use Fluidd and in settings/cameras, I added a camera keeping all the pre-filled options. 
Name Ender
Stream type MJPEG Adaptive
FPS Target 30
FPS Target when not in focus 5
Camera Url Stream    /webcam/?action=stream
Camera Url Snapshot   /webcam/?action=snapshot

Save and restart.    Camera 1 should be up now. 

For camera 2, here is my crowsnest.cfg settings.  Add this: 
[cam 2]
mode: ustreamer                         
                                        
enable_rtsp: false                      
rtsp_port: 8551                      # Set different ports for each device!
port: 8081                          # HTTP/MJPG stream/snapshot port
device: /dev/video0                  #YES, /dev/video0 (zero)
resolution: 1920x1080                #NO SPACE before/after the x!
max_fps: 30                            

Save and restart.  Now in Fluidd's settings/cameras I added the second camera like below. Note the number 2 in camera URL. 
Stream type MJPEG Adaptive
FPS Target 30
FPS Target when not in focus 5
Camera Url Stream    /webcam2/?action=stream
Camera Url Snapshot   /webcam2/?action=snapshot

Both cameras should be up.    I see them in Fluidd and in Klipperscreen.   

I spent a lot of time trying to get this to work and for some reason under /dev/ a new device video2 appeared and made the screen go blank in klipperscreen using my sonicpad debian.  I had to reflash the debian firmware on the sonicpad and start over to fix it. Hope it helps.

mryel00, 2025-04-26:
As a note, that's wrong for most cams. Most cams use two `/dev` paths, so `/dev/video0` and `/dev/video1` are the same device most of the time.
Also your resolution is written wrong, as you use some non Latin character instead of a latin `x`