mainsail-crew/crowsnest #192: Never works after reboot

lin-ycv, 2023-10-26:
### What happened

After a reboot, crowsnest always fails to start ustreamer, I'd have to then "edit" `crowsnest.conf` to get it working again.
Edit as in I have to click Edit then press `save and restart`, I don't actually have to change the file in any way.

### What did you expect to happen

I expect it to work on boot.

### How to reproduce

reboot the printer, ustreamer fails to start.

### Additional information

Log after reboot
```
[10/25/23 18:16:01] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[10/25/23 18:16:02] crowsnest: Version: v4.0.4-6-g767c53a
[10/25/23 18:16:02] crowsnest: Prepare Startup ...
[10/25/23 18:16:02] crowsnest: INFO: Host information:
[10/25/23 18:16:02] crowsnest: Host Info: Distribution: Ubuntu 20.04.6 LTS
[10/25/23 18:16:02] crowsnest: Host Info: Kernel: Linux 4.9.191 aarch64
[10/25/23 18:16:02] crowsnest: Host Info: Available CPU Cores: 4
[10/25/23 18:16:02] crowsnest: Host Info: Available Memory: 999676 kB
[10/25/23 18:16:02] crowsnest: Host Info: Diskspace (avail. / total): 7.9G / 15G
[10/25/23 18:16:02] crowsnest: INFO: Checking Dependencies
[10/25/23 18:16:02] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[10/25/23 18:16:02] crowsnest: Dependency: 'find' found in /usr/bin/find.
[10/25/23 18:16:02] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[10/25/23 18:16:02] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[10/25/23 18:16:02] crowsnest: Version Control: ustreamer new version available: 83a77ea (v5.45).
[10/25/23 18:16:02] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[10/25/23 18:16:02] crowsnest:		[crowsnest]
[10/25/23 18:16:02] crowsnest:		log_path: /home/pi/printer_data/logs/crowsnest.log
[10/25/23 18:16:02] crowsnest:		log_level: verbose
[10/25/23 18:16:02] crowsnest:		delete_log: true
[10/25/23 18:16:02] crowsnest:		no_proxy: false
[10/25/23 18:16:02] crowsnest:		
[10/25/23 18:16:02] crowsnest:		[cam 1]
[10/25/23 18:16:02] crowsnest:		mode: ustreamer
[10/25/23 18:16:02] crowsnest:		
[10/25/23 18:16:02] crowsnest:		enable_rtsp: false
[10/25/23 18:16:02] crowsnest:		rtsp_port: 8554
[10/25/23 18:16:02] crowsnest:		port: 8080
[10/25/23 18:16:02] crowsnest:		device: /dev/video0
[10/25/23 18:16:02] crowsnest:		resolution: 640x480
[10/25/23 18:16:02] crowsnest:		max_fps: 30
[10/25/23 18:16:02] crowsnest: INFO: Detect available Devices
[10/25/23 18:16:02] crowsnest: INFO: Found 1 total available Device(s)
[10/25/23 18:16:02] crowsnest: INFO: Found 1 available v4l2 (UVC) camera(s)
[10/25/23 18:16:02] crowsnest: /dev/v4l/by-id/usb-YT-221117-J_USB2.0_Camera-video-index0 -> /dev/video0
[10/25/23 18:16:02] crowsnest: Supported Formats:
[10/25/23 18:16:02] crowsnest:		[0]: 'MJPG' (Motion-JPEG, compressed)
[10/25/23 18:16:02] crowsnest:		Size: Discrete 1920x1080
[10/25/23 18:16:02] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[10/25/23 18:16:02] crowsnest:		Size: Discrete 640x480
[10/25/23 18:16:02] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[10/25/23 18:16:02] crowsnest:		Size: Discrete 352x288
[10/25/23 18:16:02] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[10/25/23 18:16:02] crowsnest:		Size: Discrete 424x240
[10/25/23 18:16:02] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[10/25/23 18:16:02] crowsnest:		Size: Discrete 640x360
[10/25/23 18:16:02] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[10/25/23 18:16:02] crowsnest:		Size: Discrete 800x480
[10/25/23 18:16:02] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[10/25/23 18:16:02] crowsnest:		Size: Discrete 800x600
[10/25/23 18:16:02] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[10/25/23 18:16:02] crowsnest:		Size: Discrete 1024x576
[10/25/23 18:16:02] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[... 134 lines cut ...]
[10/26/23 21:03:21] crowsnest:		brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
[10/26/23 21:03:21] crowsnest:		contrast 0x00980901 (int)    : min=0 max=95 step=1 default=4 value=4
[10/26/23 21:03:21] crowsnest:		saturation 0x00980902 (int)    : min=0 max=100 step=1 default=62 value=62
[10/26/23 21:03:21] crowsnest:		hue 0x00980903 (int)    : min=-2000 max=2000 step=100 default=0 value=0
[10/26/23 21:03:21] crowsnest:		white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
[10/26/23 21:03:21] crowsnest:		gamma 0x00980910 (int)    : min=100 max=300 step=1 default=125 value=125
[10/26/23 21:03:21] crowsnest:		power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
[10/26/23 21:03:21] crowsnest:		0: Disabled
[10/26/23 21:03:21] crowsnest:		1: 50 Hz
[10/26/23 21:03:21] crowsnest:		2: 60 Hz
[10/26/23 21:03:21] crowsnest:		white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
[10/26/23 21:03:21] crowsnest:		sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=4 value=4
[10/26/23 21:03:21] crowsnest:		backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=0 value=0
[10/26/23 21:03:21] crowsnest:		exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
[10/26/23 21:03:21] crowsnest:		1: Manual Mode
[10/26/23 21:03:21] crowsnest:		3: Aperture Priority Mode
[10/26/23 21:03:21] crowsnest:		exposure_absolute 0x009a0902 (int)    : min=78 max=10000 step=1 default=312 value=312 flags=inactive
[10/26/23 21:03:21] crowsnest:		focus_absolute 0x009a090a (int)    : min=0 max=1023 step=1 default=0 value=0 flags=inactive
[10/26/23 21:03:21] crowsnest:		focus_auto 0x009a090c (bool)   : default=1 value=1
[10/26/23 21:03:21] crowsnest: Try to start configured Cams / Services...
[10/26/23 21:03:23] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[10/26/23 21:03:24] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[10/26/23 21:03:26] crowsnest: Starting ustreamer with Device /dev/video0 ...
[10/26/23 21:03:28] crowsnest:  ... Done!
```

lin-ycv, 2023-10-26:
Also on a side note, the logs shows that my webcam supports 1920x1080, but if I use that resolution, ustreamer also fails to start.

mryel00, 2023-10-26:
Please join our [Discord](https://discord.gg/mainsail) for debugging this issue.
The only issue I know that could lead to this, is maybe that the camera is starting before the network service.

closed 2023-10-26 by github-actions[bot]

lin-ycv, 2023-10-27:
not sure why I didn't see it before, but the solution is here:
https://github.com/mainsail-crew/crowsnest/issues/117#issuecomment-1575017371