mainsail-crew/crowsnest #218: Feature request: Using prebuild camera-streamer in install crowsnest.

Neko-vecter, 2023-12-06:
### Is your feature request related to a problem? Please describe

The feature request relate to [mainsail-crew/crowsnest issue 188](https://github.com/mainsail-crew/crowsnest/issues/188) and [ayufan/camera-streamer pull request](https://github.com/ayufan/camera-streamer/pull/119)

### Describe the solution you'd like

using the prebuild package when install camera-streamer. This can let install crowsnest much quick on raspberry pi. Also, much stable to install (since using the package that already build). 

[link to camera-streamer release](https://github.com/ayufan/camera-streamer/releases)

### Describe alternatives you've considered

-

### Additional information

The issue #188 it is caused by missing files. The file `version.h` is missing in camera-streamer that causes the problem.

mryel00, 2023-12-06:
Hey, we are already planning this for the next big update.
We will need a few things to consider to get on binaries that's why we didn't switch yet. You can read a few things in #137 and #191. There is maybe more to it that I didn't wrote there, I cannot remember everything I wrote😅 

> The issue #188 it is caused by missing files. The file version.h is missing in camera-streamer that causes the problem.

Sry, but that's wrong. #188 was about a build problem of the `master` branch of camera-streamer on bookworm. That's why we needed to switch to `main` branch of camera-streamer. We did this only for bookworm for the reasons I mentioned in the two PRs above.

For the build problems you mentioned in https://github.com/ayufan/camera-streamer/pull/119, those weren't happening with installing camera-streamer through crowsnest, did they? I installed multiple times on different SBCs on bullseye & bookworm and didn't encounter any errors the last few days.

Neko-vecter, 2023-12-06:
Since we are planning to use the pre-build package in next big update. I think this issue (feature request) can mark as done.

Have a nice day~

-Neko.vecter

mryel00, 2023-12-06:
> I think this issue (feature request) can mark as done.

I will reopen it, that other people won't post the same FR and that everyone can see when it's done 😄

ss8688, 2024-02-15:
Is that been fnished?

mryel00, 2024-02-15:
@ss8688 as this issue is still opened. No. I didn't yet have enough time to work further on that update. In 2 weeks I should have finally more time for to work on that.