mainsail-crew/crowsnest #219: do_memory_split not found in raspi-config

jedahan, 2023-12-11:
### What happened

raspi-config removed do_memory_split (not sure exactly when, but likely when bookworm was released).

During the install, it failed with the following error
```
Build 'camera-streamer' ... [SUCCESS]
/usr/bin/raspi-config: 3069: do_memory_split: not found
Trying to set minimum GPU Memory to 160 MB ... [SKIPPED]
	==> Tool 'raspi-config' not found ...
```

I am worried the install is now only partially working.

### What did you expect to happen

To only run `raspi-config do_memory_split` when on older versions of raspberry pi OS

### How to reproduce

On a fresh version of the latest raspberry pi os:

```
git clone https://github.com/mainsail-crew/crowsnest
cd crowsnest
sudo make install BASE_USER=$USER
```

jedahan, 2023-12-11:
Also the `Tool 'raspi-config' not found` is a little bit of a misleading error - raspi-config was found, and ran, but errors out.

jedahan, 2023-12-11:
https://github.com/mainsail-crew/crowsnest/blob/f7ac6aa298143f0019eae9b2b2be8039196ebeda/tools/libs/set_gpumem.sh#L37

Adding another check for lsb_release or something else might be best. Or just making it a non-fatal error if it isn't already.

mryel00, 2023-12-11:
You are right and I always forget to fix this line....
The error is non-lethal as you can see on the yellow `[SKIPPED]` message. So no worry, everything should work without problems.

PR #223 (2024-01-07): fix: skip do_memory_split on bookworm
`do_memory_split` got removed on bookworm. Fixes #219
commit 4feaab72 (2024-01-07): fix: skip do_memory_split on bookworm (#223)

`do_memory_split` got removed on bookworm. Fixes #219
closed 2024-01-08 by meteyou