mainsail-crew/crowsnest #228: Pas de flux vidéo alors que tout parais correct dans crowsnest

JYCrois, 2024-01-11:
### What happened

je viens de finir l'installation des versions officiel sur un Flsun SpeederPad suivant le tuto de Guilouz, tout fonctionne correctement je n'ai qu'un seul problème, ma caméra ne fonctionne pas, elle fonctionne très bien avant d'installer les versions officielles mais maintenant sous crowsnest rien à faire, pourtant je n'ai aucune erreur tout est à jour (voir PJ), une idée ? si vraiment il n'y a pas de solution pour crowsnest est-il possible de réinstaller MPG-Streamer et comment (je n'ai pas vu l'option dans KIAUH) ? cordialement en vous remerciant pour votre travail,

### What did you expect to happen

pouvoir avoir ma caméra fonctionnel avec crowsnest
[image]
[image]
[image]
[image]

### How to reproduce

.

### Additional information

.

mryel00, 2024-01-11:
Sry, but I don't speak french. So please edit your issue and translate it in english.

mryel00, 2024-01-11:
Ok, thank you, now I can read it, but it seems like you want support, so please join our [Discord](https://discord.gg/mainsail) and open a post inside the support-forum or open a post in our [GitHub Discussions](https://github.com/orgs/mainsail-crew/discussions/categories/help). In both cases you should post some logs.

> is it possible to reinstall MPG-Streamer

We don't provide any installation instructions for this, you have to do your own research on that. But this software is outdated by years. Also it should only be uninstalled when you have uninstalled it with KIAUH. Crowsnest only disables the service.

Overall FLSUN Speederpads work with Crowsnest, as you can see in e.g. this issue #117

closed 2024-01-11 by github-actions[bot]