mainsail-crew/crowsnest #241: Unable to add webcam in Mainsail - RPI 5 and Working ov5647 module

LadPack, 2024-02-09:
### What happened

When attempting to add in a ov5647 camera on mainsail, there is consistently a no-signal when adding via the mainsail UI.  Tested the camera module via CLI and was able to receive pictures.

### What did you expect to happen

Normally when I have done this in the past (including with this camera module on a Pi4) it would pick up the picam with no intervention on my end. 

### How to reproduce

Fresh Pi5 install, via KIAUH, plug in camera.

### Additional information

[02/08/24 19:18:27] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[02/08/24 19:18:27] crowsnest: Version: v4.1.3-1-gc0c9c41
[02/08/24 19:18:27] crowsnest: Prepare Startup ...
[02/08/24 19:18:27] crowsnest: INFO: Host information:
[02/08/24 19:18:27] crowsnest: Host Info: Distribution: Debian GNU/Linux 12 (bookworm)
[02/08/24 19:18:27] crowsnest: Host Info: Kernel: Linux 6.1.0-rpi7-rpi-2712 aarch64
[02/08/24 19:18:27] crowsnest: Host Info: Model:  Raspberry Pi 5 Model B Rev 1.0
[02/08/24 19:18:27] crowsnest: Host Info: Available CPU Cores: 4
[02/08/24 19:18:27] crowsnest: Host Info: Available Memory: 4143600 kB
[02/08/24 19:18:27] crowsnest: Host Info: Diskspace (avail. / total): 18G / 29G
[02/08/24 19:18:27] crowsnest: INFO: Checking Dependencies
[02/08/24 19:18:27] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[02/08/24 19:18:27] crowsnest: Dependency: 'find' found in /usr/bin/find.
[02/08/24 19:18:27] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[02/08/24 19:18:27] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[02/08/24 19:18:27] crowsnest: Version Control: ustreamer new version available: b578e98 (v5.49).
[02/08/24 19:18:27] crowsnest: INFO: Print Configfile: '/home/david/printer_data/config/crowsnest.conf'
[02/08/24 19:18:27] crowsnest:		[crowsnest]
[02/08/24 19:18:27] crowsnest:		log_path: /home/david/printer_data/logs/crowsnest.log
[02/08/24 19:18:27] crowsnest:		log_level: verbose
[02/08/24 19:18:27] crowsnest:		delete_log: true
[02/08/24 19:18:27] crowsnest:		no_proxy: false
[02/08/24 19:18:27] crowsnest:		
[02/08/24 19:18:27] crowsnest:		[cam 1]
[02/08/24 19:18:27] crowsnest:		mode: ustreamer
[02/08/24 19:18:27] crowsnest:		
[02/08/24 19:18:27] crowsnest:		enable_rtsp: false
[02/08/24 19:18:27] crowsnest:		rtsp_port: 8554
[02/08/24 19:18:27] crowsnest:		port: 8080
[02/08/24 19:18:27] crowsnest:		device: /dev/video0
[02/08/24 19:18:27] crowsnest:		resolution: 640x480
[02/08/24 19:18:27] crowsnest:		max_fps: 15
[02/08/24 19:18:27] crowsnest: INFO: Detect available Devices
[02/08/24 19:18:27] crowsnest: INFO: Found 1 total available Device(s)
[02/08/24 19:18:27] crowsnest: ================================================================
[02/08/24 19:18:27] crowsnest:  WARN: 'libcamera' devices are currently not supported on Pi 5! 
[02/08/24 19:18:27] crowsnest: ================================================================
[02/08/24 19:18:27] crowsnest: Detected 'libcamera' device -> /base/axi/pcie@120000/rp1/i2c@80000/ov5647@36
[02/08/24 19:18:27] crowsnest: Try to start configured Cams / Services...
[02/08/24 19:18:27] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[02/08/24 19:18:28] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[02/08/24 19:18:28] crowsnest: Starting ustreamer with Device /dev/video0 ...
[02/08/24 19:18:30] crowsnest:  ... Done!

mryel00, 2024-02-09:
> Tested the camera module via CLI and was able to receive pictures.

Please elaborate further on this. What exact command did you run? This won't change anything below, I'm just curious.

As you can read on your log snippet, we don't support libcamera devices on the Pi5. Therefore no CSI cam support atm.

If you used the legacy stack to get an image, you have to know that we cannot make it work consistently with the legacy stack on Bookworm. Also that legacy stack will be removed in future versions of Raspberry Pi OS, so it shouldn't be used.

We will add some support in the future but for now we cannot support it.

mryel00, 2024-02-09:
I changed it to a feature request, as we currently don't support it as clearly stated in the log.

LadPack, 2024-02-09:
I used libcamera-hello to get a picture file.

Ah got it, so any and all of the pi cameras are no longer supported as of the pi5? Is that more of a Pi5 thing or just because it is new? Is there a workaround potentially to have the pi serve up some sort of ip camera to look at?

I tried camera-streamer which didn't work either for this purpose.

mryel00, 2024-02-09:
It's an overall Pi5 issue atm. We are using camera-streamer as our backend. As that doesn't work, we don't support it.
The problem with the Pi5 is the missing JPEG, MJPG and H264 hardware encoders, that's why camera-streamer doesn't support it.

Some kind of workaround/alternative would be [spyglass](https://github.com/roamingthings/spyglass) with some adjustments. I might go into detail about it later today. Basically you need to change one of the includes that it's using the software encoder instead of the hardware encoder.

We will add spyglass in an upcoming update, but, like I wrote earlier, that's currently not possible for us.

LadPack, 2024-02-09:
So interim, if I have the option to use a pi4, then its probably better to use it?  Just for ease of use

mryel00, 2024-02-09:
Exactly a Pi4 would work as expected. It's really just the Pi5 with it's hardware limitations (sounds wrong as it's the newer device but it's actually like that).

To the Spyglass workaround:
**_Only use this for picams/CSI cams on a Pi5!_** This doesn't have WebRTC support!
You need to install some packages for spyglass to work.
```bash
sudo apt update
sudo apt install python3-libcamera python3-kms++ python3-picamera2
```
Then clone the repository and follow the instructions on how to install as a service.
To get the best performance, you should change line 11 inside `~/spyglass/spyglass/cli.py` from
```py
from picamera2.encoders import MJPEGEncoder
```
to
```py
from picamera2.encoders import JpegEncoder as MJPEGEncoder
```
Make sure to comment out the picam/csi cam in crowsnest or stop/disable crowsnest completely.
You should also adjust the `spyglass.conf` (found at the same place as the crowsnest.conf) to the correct streaming and snapshot url.

Like written above someone should only use this for picams/CSI cams. Spyglass only supports one cam at the same time.

**_I won't give support on this workaround in this issue or in this repository. If you need some help, join our [Discord](https://discord.gg/mainsail) or use the spyglass issue tracker. This should only be treated as a workaround and not as a final solution._**
### We do not recommend to use a Pi5!

chalb0, 2024-07-02:
### This is a +1 from me with the same issue now.
I only have a pi5 available which is why I'm trying to use it. Would this feature request more accurately be described as using an H.265 codec method for streaming for Pi5's?

FYI Spyglass isn't a workaround for me, as last time I tried it, this wouldn't stream to Obico which is the main reason I use a webcam.

My logs are the same...

### LOGS
> [07/02/24 23:06:25] crowsnest: INFO: Found 1 total available Device(s)
> [07/02/24 23:06:25] crowsnest: ================================================================
> [07/02/24 23:06:25] crowsnest:  WARN: 'libcamera' devices are currently not supported on Pi 5!
> [07/02/24 23:06:25] crowsnest: ================================================================
> [07/02/24 23:06:25] crowsnest: Detected 'libcamera' device -> /base/axi/pcie@120000/rp1/i2c@88000/imx519@1a
> [07/02/24 23:06:25] crowsnest: Try to start configured Cams / Services...
> [07/02/24 23:06:26] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
> [07/02/24 23:06:26] crowsnest: WARN: Mode 'camera-streamer' is not supported on your device!
> [07/02/24 23:06:26] crowsnest: WARN: Falling back to Mode 'ustreamer'
> [07/02/24 23:06:26] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
> [07/02/24 23:06:26] crowsnest: Starting ustreamer with Device /base/axi/pcie@120000/rp1/i2c@88000/imx519@1a ...
> [07/02/24 23:06:28] crowsnest:  ... Done!
> [07/02/24 23:06:28] crowsnest: WATCHDOG: Lost Device: '/base/axi/pcie@120000/rp1/i2c@88000/imx519@1a'

mryel00, 2024-07-03:
If you read the log, you would understand, why you have the same problem.
It's still not supported.

There is no reason why Spyglass shouldn't work. Crowsnest (ustreamer/camera-streamer) do nothing special compared to Spyglass.

> Would this feature request more accurately be described as using an H.265 codec method for streaming for Pi5's?

There is no intention from us to use H265 encoding, and we are not aware of any backend that is able to do this easily with the Pi5 encoder, but we didn't really search for it. H265 isn't available in some browsers by default, therefore that's no solution we will consider atm. 

Our solution will be Spyglass, as we will add it with the next big version update, that has no current ETA, for multiple reasons.

rykerg9, 2024-07-06:
> There is no intention from us to use H265 encoding, and we are not aware of any backend that is able to do this easily with the Pi5 encoder, **but we didn't really search for it**.
Wow, so helpful...
Glad I switched to something other than Crowsnest since Mainsail devs are _lightning fast at development and sooooo helpful_ /s

Why even have a feature request tag if you don't have any intentions of adding features people want?

mryel00, 2024-07-07:
> Glad I switched to something other than Crowsnest

Good that you found something that already supports H265 encoding and that works for you, but the rest of your message is just intentionally upsetting right? Please read everything, if you want to understand the background of this whole thing.
> Why even have a feature request tag if you don't have any intentions of adding features people want?

You didn't read all the messages, did you? You just picked one thing out of this whole thread. The FR is support for pi cams on the Pi5. The current workaround that I shared, will be the future implementation, but there are other things that have to be done first. There is only no intention for H265 support (See the last paragraph). If you got a backend that supports it, you can share it if you want, and I can take a look at it. Otherwise there will be only Spyglass with MJPG and WebRTC (only H264 planned atm) support.
Spyglass doesn't have WebRTC support atm, that's one of the reasons, why this FR still take some time.
> Mainsail devs are lightning fast at development

The Mainsail Crew is just some loosely group of people that are helping meteyou with development of different projects around Mainsail for the Klipper project. Mete, and we others, don't get money and do this in our free time. It's sad to see that people think we are obligated to get everything as fast as some company, with multiple devs behind each project.
This project is pretty slow developing, but that's due to the time constraints I and other people have.
Also this is open source, so just make a fork if someone is too slow for you, but I bet you won't take your time to develop such a thing for free for random complaining people on the internet.
> sooooo helpful

When were we not helpful? We give support on different plattforms (one of the things consuming my time, so you can help out, if I should invest more time into Crowsnest 😉). I even provided a workaround that I can personally support, instead of sharing other peoples projects and sending all the support to them, and therefore "stealing" their time.

Are we not helpful, because I wrote, that we won't add H265? Why did I choose to not search for some H265 backend? You should maybe see, if your browser even got support activated, with a site like [this](https://html5test.com/). For me on the latest Chrome version on Windows (I guess this is one of the most common setups), it isn't activated. So I think, you didn't know what H265 is, while you wrote that message. I won't support a thing, just for the sake of supporting it and then creating even more support posts from people that are already overwhelmed with this project (most prominent problem is activating a picam....)

mryel00, 2024-07-07:
I edited now the title, that people understand what's the feature behind this FR

meteyou, 2024-07-07:
@rykerg9 this is all open source. feel free to fork this project or start your own project, develop, maintain and support it all in your free time without get paid for anything and then let other people insult you for doing something free for everyone.

so this is all free... just use it, or let it... but pls don't be a bastard...

rykerg9, 2024-07-07:
@meteyou 
>but pls don't be a dick...

Never expected name calling as well, wow. Very professional. *slow clap*

also @mryel00 I'm not reading the literal book that you replied with since I have better things to do than get pinged and bothered by the people here, like actually printing things and not getting butt hurt that someone called you out on your piss poor "support".

Thanks anyways though <3

mryel00, 2024-07-07:
@rykerg9 Then a short version of that book:
We will add Pi5 support, but won't add H265 specifically, as it isn't supported by default in most browsers.
Also please share your alternative for other people, that your comment wasn't just a complaint, but something useful for this discussion.
> someone called you out on your piss poor "support"

I hope you mean the Pi5 support and not overall the support we are doing.

chalb0, 2024-07-07:
Just to clarify, my “+1” was to confirm I knew I encountered the same issue and wanted to understand a little more about the situation to decide if I source a pi4, or continue to invest time with spyglass or wait, exactly what you have given me @mryel00, thank you.

As it turns out, I’ve managed to get spyglass working and using Obico with a little more effort (I could already get it streaming fine to mainsail) I have done this with a 32bit OS as I’ve found comments that 64bit didn’t work (which I too have yet to get working).

For anyone else interested… my setup is (what I had available from other projects):
Arducam 16MP AF (IMX519).
Raspberry Pi5.
Raspberry Pi OS 32bit Bookworm
Spyglass (not installed crowsnest on a new setup).

Clean install of OS 32bit
Setup Klipper and others with KIAUH.
Followed install setups for Arducam on their wiki.
Followed setup guide for spyglass with Obico on Spyglass GitHub (there’s a few steps not recorded to do with using the right path on the Obico config).

mryel00, 2024-07-07:
> setup guide for spyglass with Obico on Spyglass GitHub

Can you link me would you mean with that?

chalb0, 2024-07-07:
More detailed workaround steps. I've not got it working on x64 with Spyglass and Obico.

### Install the Camera
Install Klipper, Moonraker, Mainsail, Obico for Moonraker via KUAIH:
https://github.com/dw-0/kiauh
Link Obico to your account as part of this step.

### Install the Camera
Using whichever camera guide you have. My Arducam 16mp AF was here:
https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/16MP-IMX519/
Make sure you set the right interface you're using (CAM0 needed to be explicitly listed in /boot/firmware/config.txt for me).

### Spyglass setup guide: 
https://github.com/roamingthings/spyglass
CLI on Pi5:
```
cd ~
git clone https://github.com/roamingthings/spyglass
cd ~/spyglass
make install
```

### Modify spyglass.conf
This can be to your preferences, or may need to set specific Resolution compatible with your camera.

### Setup the Webcam in Mainsail UI
URL Steam: `/webcam/stream`
URL Snapshot: `/webcam/snapshot`
Service: `UV4L-MJPEG`
At this point, you should be able to see the webcam in your UI. If not, revisit camera setup.

### Install Janus
This doesn't seem to be included in Spyglass and is needed for Obico to stream.
CLI on Pi5:
```
sudo apt install janus
```

### Modify moonraker-obico.cfg
under the webcam section make sure these two entries are there.
moonraker-obico.cgf:
```
[webcam]
stream_url = "http://127.0.0.1:8080/stream"
snapshot_url = "http://127.0.0.1:8080/snapshot"
```

### Troubleshooting:
1. Check Obico Notifications (via web interface https://app.obico.io/printer_events/ )
2. Check the moonraker obico logs `cat /home/pi/printer_data/logs/moonraker-obico.log` (note change "pi" for your username)

mryel00, 2024-08-23:
I read again my previous longer explanation and wanted to clarify it as a separate comment, might be interesting for @rykerg9 too:
The Pi5 has no HW video encoding. It's only a H265 HW decoder. It seems like some people get good results with specific settings  of ffmpeg that I don't know of. But I also don't know the quality of the image output, I just heard about some good results. So any information on that will be helpful if someone also wants WebRTC without high CPU usage on the Pi5.

Also I added WebRTC to Spyglass, but it's still experimental, as I have to do some testing on it. The CPU usage on the Pi5 is quite high and therefore it shouldn't be used with it, or at least only with caution: https://github.com/mryel00/spyglass/pull/84

tismofied, 2024-11-30:
> More detailed workaround steps. I've NOW got it working on x64 with Spyglass and Obico.
> 
> 
> 
> ### Install the Camera
> 
> Install Klipper, Moonraker, Mainsail, Obico for Moonraker via KUAIH:
> 
> https://github.com/dw-0/kiauh
> 
> Link Obico to your account as part of this step.
> 
> 
> 
> 
> 
> ### Install the Camera
> 
> Using whichever camera guide you have. My Arducam 16mp AF was here:
> 
> https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/16MP-IMX519/
> 
> Make sure you set the right interface you're using (CAM0 needed to be explicitly listed in /boot/firmware/config.txt for me).
> 
> 
> 
> 
> 
> ### Spyglass setup guide: 
> 
> https://github.com/roamingthings/spyglass
> 
> CLI on Pi5:
> 
> ```
> 
> cd ~
> 
> git clone https://github.com/roamingthings/spyglass
> 
> cd ~/spyglass
> 
> make install
> 
> ```
> 
> 
> 
> 
> 
> ### Modify spyglass.conf
> 
> This can be to your preferences, or may need to set specific Resolution compatible with your camera.
> 
> 
> 
> 
> 
> ### Setup the Webcam in Mainsail UI
> 
> URL Steam: `/webcam/stream`
> 
> URL Snapshot: `/webcam/snapshot`
> 
> Service: `UV4L-MJPEG`
> 
> At this point, you should be able to see the webcam in your UI. If not, revisit camera setup.
> 
> 
> 
> 
> 
> ### Install Janus
> 
> This doesn't seem to be included in Spyglass and is needed for Obico to stream.
> 
> CLI on Pi5:
> 
> ```
> 
> sudo apt install janus
> 
> ```
> 
> 
> 
> 
> 
> 
> 
> ### Modify moonraker-obico.cfg
> 
> under the webcam section make sure these two entries are there.
> 
> moonraker-obico.cgf:
> 
> ```
> 
> [webcam]
> 
> stream_url = "http://127.0.0.1:8080/stream"
> 
> snapshot_url = "http://127.0.0.1:8080/snapshot"
> 
> ```
> 
> 
> 
> 
> 
> ### Troubleshooting:
> 
> 1. Check Obico Notifications (via web interface https://app.obico.io/printer_events/ )
> 
> 2. Check the moonraker obico logs `cat /home/pi/printer_data/logs/moonraker-obico.log` (note change "pi" for your username)

Thank you for this guide. This should be pinned for guys with pi5s

mryel00, 2024-11-30:
> This should be pinned for guys with pi5s

I disagree on that. First of all, these are the issues of Crowsnest, not some tutorial hub for workarounds for workarounds. I sadly cannot even pin single comments.
Second this is a tutorial for Obico with Spyglass, so it's not even related to this issue/FR directly, as it's meant to track support of CSI cams on the Pi5.
Third, we got like 11 issues open right now, it's not that hard to find anything here.

Also please don't reference whole comments. Imagine if I would have referenced your whole comment too, then we would have pasted a really long tutorial three times into this conversation. I started to read through the mail and didn't even realize that this comment were just two sentences and I wondered who posted another tutorial here. You were not the first one and won't be the last one, but if someone references such a long comment, it makes readability in an issue go down dramastically.

black97devil, 2025-09-04:
thank you very much [chalb0]

you made my day , after 2 days of trying finally worked

closed 2026-04-17 by mryel00