mainsail-crew/crowsnest #242: install_script: tools/pkglist.sh is not listed in the tools directory

vinaymahendra, 2024-02-16:
### What happened

I am following your instructions on your website to add the following lines to moonraker.conf:

[update_manager crowsnest]
type: git_repo
path: ~/crowsnest
origin: https://github.com/mainsail-crew/crowsnest.git
install_script: tools/pkglist.sh

In my Fluid instance, I am now getting the error, "Moonraker warnings found.
[update_manager]: Failed to load extension crowsnest: Invalid path for option `install_script` in section [update_manager crowsnest]: Path `/home/mks/crowsnest/tools/pkglist.sh` does not exist"

So I check you repo and pkglist.sh is not listed in that directory.

### What did you expect to happen

Adding the lines of code to moonraker.conf would not cause an error to occure.

### How to reproduce

Add the lines to your moonraker.cfg and see that it produces the same error.

mryel00, 2024-02-16:
Duplicate of #150 and #186

mryel00, 2024-02-16:
Ofc you get an error if you just add it without installing crowsnest.
I think you are following our docs:
> After a successful installation, you should update your moonraker.conf as shown below to keep Crowsnest up to date. You will be asked during the installation if you want the script to do this for you.

**_AFTER A SUCCESFUL INSTALLATION_** means what it says. First instlal and then add it, but the scrip already asks you if the lines should be added.
If that file doesn't exist **_after a successful installation_**, join our [Discord](https://discord.gg/mainsail) or use our [GitHub discussions](https://github.com/orgs/mainsail-crew/discussions).

closed 2024-02-16 by mryel00