mainsail-crew/crowsnest #243: IMX519 autofocus in streamer mode

vigoral, 2024-02-22:
### Is your feature request related to a problem? Please describe

I have ras pi 4 and klipper with moonraker and fluid
Also I have camera module connected via CSI to ras pi (https://www.waveshare.com/wiki/IMX519-78_16MP_AF_Camera)
I did not find how to configure autofocus for this camera
My conf file:
`[crowsnest]
log_path: /home/ihvor/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: camera-streamer
enable_rtsp: false
rtsp_port: 8554
port: 8080
device: /base/soc/i2c0mux/i2c@1/imx519@1a
resolution: 1920x1080
max_fps: 60`

I saw that it is possible to use autofocus by libcamera but I did not find solution how to do it myself

### Describe the solution you'd like

add new param in 'cam' section and if it is using libcamera then use solution from link https://www.waveshare.com/wiki/IMX519-78_16MP_AF_Camera

closed 2024-02-22 by github-actions[bot]

mryel00, 2024-02-22:
References for self research:
https://crowsnest.mainsail.xyz/configuration/cam-section#v4l2ctl
https://libcamera.org/api-html/namespacelibcamera_1_1controls.html

mryel00, 2024-02-22:
I should also mention #240 and here is a small example:
```
v4l2ctl: afmode=0, lensposition=4.5
```