mainsail-crew/crowsnest #246: crowsnest: Dependency: 'ustreamer' not found. Exiting!

seniv, 2024-02-28:
### What happened

After a fresh installation of crowsnest via Kiauh on RPi 4 with PRi OS(x64) I saw an error `crowsnest: Dependency: 'ustreamer' not found. Exiting!` in the logs and the crowsnest didn't work.

### What did you expect to happen

crowsnest to work right after installation without errors

### How to reproduce

install fresh version of crowsnest

### Additional information

After a bit of research, I found out that crowsnest was looking for the file `~/crowsnest/bin/ustreamer/ustreamer` but there was only `ustreamer.bin`

I see that https://github.com/pikvm/ustreamer made a lot of new commits during the last few days, most likely something was changed.

after all, I fixed the problem by copying the file `ustreamer.bin` as `ustreamer` by running the command
```
cp ~/crowsnest/bin/ustreamer/ustreamer.bin ~/crowsnest/bin/ustreamer/ustreamer
```
But would be nice if dependencies were installed with specific commit hash or tag, instead of using the latest version

seniv, 2024-02-28:
I see that @mryel00 already created PR https://github.com/mainsail-crew/crowsnest/pull/216 for using specific versions of the dependencies, that should fix the issue and prevent it to happen in the future

mryel00, 2024-02-28:
@seniv we just released an update shortly before you opened this issue: https://github.com/mainsail-crew/crowsnest/pull/245

I started this commit pinning stuff, but we came across some issues as the implementation of the older parts, so I stopped developing on it for now and will most likely close it completely.

At the moment I'm rewriting the whole running code base in python and we will change to precompiled binaries with that update too. Therefore this issue won't happen in the next version and we won't need commit pinning in future.

closed 2024-02-28 by mryel00