mainsail-crew/crowsnest #252: MKS Pi board not detecting as Pi board

bathtubtoast, 2024-03-21:
### What happened

It says 'not Pi' during build in bullseye and bookworm.
In bullseye this causes the ustreamer to fail to build, which is unexpected.

### What did you expect to happen

that it would at least detect as a Pi/Pi clone and/or properly build ustreamer

### How to reproduce

using an MKS KLIPR board, or TwoTrees SK1 mainboard run compile.

### Additional information

The output of `cat /proc/device-tree/model` is "Makerbase mks-pi"

changing the grep in tools/libs/core.sh for the "is_raspberry_pi()" from "Raspberry" to "mks-pi" causes the ustreamer build to build correctly and function as expected.

I don't understand enough of how the core.sh is called to add a "is_makerbase_pi()" function that would then be called correctly and would like some assistance, preferable in 'more information' so that I could learn and submit the bugfix myself, since I don't know if its only a change to one of these: versioncontrol.sh, hwhandler.sh, init_stream.sh, dev-helper.sh, or build.sh, or all of them. But in the realistic part of the world, if its a 10 second fix for someone knowledgeable, I can run a git diff to learn as well, since I'm 100% behind fixes not wasting extra time as well.

mryel00, 2024-03-21:
It's not a Raspberry Pi, therefore it will not be detected as one. The message is pretty clear:
> This device is not a Raspberry Pi therefore camera-streeamer cannot be installed ...

If you would look further in the code, there is no check about ustreamer and pis. Every device get's the same ustreamer. We just clone the repository (currently a fork to easy control the version) and then build it. So please give us the error message that we can see if it's an actual issue.

bathtubtoast, 2024-03-21:
Ok, I will close this issue, and create a new one after I have time to reflash my printer to rebuild from a 'clean' environment, if I see the issue again.

closed 2024-03-21 by bathtubtoast

mryel00, 2024-03-21:
Please just reopen this issue, if you see the issue again.