mainsail-crew/crowsnest #257: New version seems to have broken for me - had to remove libcamera

zack6849, 2024-04-03:
### What happened

I updated crowsnest to v4.1.7-1-g424954d and it started to fail to load, I had not changed anything in my config

### What did you expect to happen

The software starts

### How to reproduce

Unsure.

### Additional information

Removing libcamera fixed the issue for me (`$  sudo apt remove libcamera*`)

```
[04/02/24 16:00:57] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[04/02/24 16:00:57] crowsnest: Version: v4.1.7-1-g424954d
[04/02/24 16:00:57] crowsnest: Prepare Startup ...
[04/02/24 16:00:57] crowsnest: INFO: Host information:
[04/02/24 16:00:57] crowsnest: Host Info: Distribution: Debian GNU/Linux 11 (bullseye)
[04/02/24 16:00:57] crowsnest: Host Info: Release: MainsailOS release 1.3.2 (bullseye)
[04/02/24 16:00:57] crowsnest: Host Info: Kernel: Linux 6.1.21-v8+ aarch64
[04/02/24 16:00:57] crowsnest: Host Info: Model:  Raspberry Pi 3 Model B Plus Rev 1.3
[04/02/24 16:00:57] crowsnest: Host Info: Available CPU Cores: 4
[04/02/24 16:00:57] crowsnest: Host Info: Available Memory: 847624 kB
[04/02/24 16:00:57] crowsnest: Host Info: Diskspace (avail. / total): 44G / 59G
[04/02/24 16:00:57] crowsnest: INFO: Checking Dependencies
[04/02/24 16:00:57] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[04/02/24 16:00:57] crowsnest: Dependency: 'find' found in /usr/bin/find.
[04/02/24 16:00:57] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[04/02/24 16:00:57] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[04/02/24 16:00:57] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[04/02/24 16:00:57] crowsnest: Version Control: ustreamer is up to date. (v6.10)
[04/02/24 16:00:57] crowsnest: Version Control: camera-streamer is up to date. ((f1627aa))
[04/02/24 16:00:57] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[04/02/24 16:00:57] crowsnest:		[crowsnest]
[04/02/24 16:00:57] crowsnest:		log_path: /home/pi/printer_data/logs/crowsnest.log
[04/02/24 16:00:57] crowsnest:		log_level: verbose
[04/02/24 16:00:57] crowsnest:		delete_log: false
[04/02/24 16:00:57] crowsnest:		no_proxy: false
[04/02/24 16:00:57] crowsnest:		
[04/02/24 16:00:57] crowsnest:		[cam 1]
[04/02/24 16:00:57] crowsnest:		mode: ustreamer
[04/02/24 16:00:57] crowsnest:		
[04/02/24 16:00:57] crowsnest:		enable_rtsp: false
[04/02/24 16:00:57] crowsnest:		rtsp_port: 8554
[04/02/24 16:00:57] crowsnest:		port: 8080
[04/02/24 16:00:57] crowsnest:		device: /dev/video0
[04/02/24 16:00:57] crowsnest:		resolution: 640x480
[04/02/24 16:00:57] crowsnest:		max_fps: 15
[04/02/24 16:00:57] crowsnest: INFO: Detect available Devices
[04/02/24 16:00:58] crowsnest: ERROR: Error 1 occured on line 65
[04/02/24 16:00:58] crowsnest: ==> Error occured in file: hwhandler.sh -> detect_libcamera
[04/02/24 16:00:58] crowsnest: ERROR: Stopping crowsnest.
[04/02/24 16:00:58] crowsnest: Goodbye...
[04/02/24 16:00:58] crowsnest: ERROR: Error 1 occured on line 65
[04/02/24 16:00:58] crowsnest: ==> Error occured in file: hwhandler.sh -> detect_libcamera
[04/02/24 16:00:58] crowsnest: ERROR: Stopping crowsnest.
[04/02/24 16:00:58] crowsnest: Goodbye...
[04/02/24 16:00:58] crowsnest: ERROR: Error 1 occured on line 93
[04/02/24 16:00:58] crowsnest: ==> Error occured in file: logging.sh -> print_cams
[04/02/24 16:00:58] crowsnest: ERROR: Stopping crowsnest.
[04/02/24 16:00:58] crowsnest: Goodbye...
```

zack6849, 2024-04-03:
Looks like this may have been fixed in https://github.com/mainsail-crew/crowsnest/pull/256

mryel00, 2024-04-03:
Yes it should have been fixed, but you have to tell me, if it's fixed for you. This is already released for 13 hours and your post is 8 hours old. So please update and tell me if it's fixed.

> Removing libcamera fixed the issue for me

Next time don't uninstall just all packages that you can find, especially if it was working last update, then the solution is never to uninstall dependencies. The problem that we found was about, using a CSI camera.

As you just purged all you libcamera packages I recommend to reinstall your system I won't help you to get all the packages that you need.
Here a list of the installed once on my Pi4:
```
libcamera-apps-lite/oldstable,now 1.2.1-1 arm64 [installed]
libcamera-dev/oldstable,now 0~git20230720+bde9b04f-1 arm64 [installed]
libcamera0/oldstable,now 0~git20230720+bde9b04f-1 arm64 [installed,automatic]
```

zack6849, 2024-04-03:
Yes, the update fixed it, I removed libcamera as I was not using it to my knowledge and it seemed the easiest way to prevent that code branch from executing without me modifying the code, I needed to install these on mainsail OS:   
`$ sudo apt install libcamera-dev python3-picamera2 libcamera-apps-lite libcamera0 python3-libcamera`

closed 2024-04-03 by zack6849

mryel00, 2024-04-03:
Oh I forgot about the python3 packages. They aren't really needed but it makes sense to have them.

> I removed libcamera as I was not using it to my knowledge

That's correct as long as you are not using a CSI camera.

> it seemed the easiest way to prevent that code branch from executing without me modifying the code

The easiest way to prevent it, would be a rollback to a previous working version:
```sh
cd ~/crowsnest
git reset --hard v4.1.6
```
That command resets it to `v4.1.6`. That can be changed to every tag available on your current branch.