mainsail-crew/crowsnest #258: No camera image after update

pro2call-nl, 2024-04-08:
### What happened

I updated the v4.1.3-1-gc0c9c41 to v4.1.8-1-g6ac4711. Now my camera is not working anymore. 

### What did you expect to happen

a functional camera

### How to reproduce

reboot raspberry pi

### Additional information

I'm using a raspberry pi 4 with the ov5647 camera. 

**LOG:**
> [04/08/24 07:53:40] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[04/08/24 07:53:40] crowsnest: Version: v4.1.8-1-g6ac4711
[04/08/24 07:53:40] crowsnest: Prepare Startup ...
[04/08/24 07:53:40] crowsnest: INFO: Host information:
[04/08/24 07:53:40] crowsnest: Host Info: Distribution: Debian GNU/Linux 12 (bookworm)
[04/08/24 07:53:40] crowsnest: Host Info: Kernel: Linux 6.6.20+rpt-rpi-v8 aarch64
[04/08/24 07:53:40] crowsnest: Host Info: Model:  Raspberry Pi 4 Model B Rev 1.5
[04/08/24 07:53:40] crowsnest: Host Info: Available CPU Cores: 4
[04/08/24 07:53:40] crowsnest: Host Info: Available Memory: 1891256 kB
[04/08/24 07:53:40] crowsnest: Host Info: Diskspace (avail. / total): 22G / 58G
[04/08/24 07:53:40] crowsnest: INFO: Checking Dependencies
[04/08/24 07:53:40] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[04/08/24 07:53:40] crowsnest: Dependency: 'find' found in /usr/bin/find.
[04/08/24 07:53:40] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[04/08/24 07:53:40] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[04/08/24 07:53:40] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[04/08/24 07:53:40] crowsnest: Version Control: ustreamer is up to date. (v5.46)
[04/08/24 07:53:40] crowsnest: Version Control: camera-streamer is up to date. ((54bdd9d))
[04/08/24 07:53:40] crowsnest: INFO: Print Configfile: '/home/printer/printer_data/config/crowsnest.conf'
[04/08/24 07:53:40] crowsnest:		[crowsnest]
[04/08/24 07:53:40] crowsnest:		log_path: /home/printer/printer_data/logs/crowsnest.log
[04/08/24 07:53:40] crowsnest:		log_level: debug
[04/08/24 07:53:40] crowsnest:		delete_log: false
[04/08/24 07:53:40] crowsnest:		no_proxy: false
[04/08/24 07:53:40] crowsnest:		
[04/08/24 07:53:40] crowsnest:		
[04/08/24 07:53:40] crowsnest:		[cam 1]
[04/08/24 07:53:40] crowsnest:		mode: camera-streamer
[04/08/24 07:53:40] crowsnest:		enable_rtsp: false
[04/08/24 07:53:40] crowsnest:		port: 8080
[04/08/24 07:53:40] crowsnest:		device: /base/soc/i2c0mux/i2c@1/ov5647@36
[04/08/24 07:53:40] crowsnest:		resolution: 1920x1080
[04/08/24 07:53:40] crowsnest:		max_fps: 30.62
[04/08/24 07:53:40] crowsnest: INFO: Detect available Devices
[04/08/24 07:53:41] crowsnest: INFO: Found 1 total available Device(s)
[04/08/24 07:53:41] crowsnest: Detected 'libcamera' device -> /base/soc/i2c0mux/i2c@1/ov5647@36
[04/08/24 07:53:41] crowsnest: 'libcamera' device(s) resolution(s) :
[04/08/24 07:53:41] crowsnest:		0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
[04/08/24 07:53:41] crowsnest:		Colorspace: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
[04/08/24 07:53:41] crowsnest:		1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
[04/08/24 07:53:41] crowsnest:		1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
[04/08/24 07:53:41] crowsnest:		2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
[04/08/24 07:53:41] crowsnest:		
[04/08/24 07:53:41] crowsnest: 'libcamera' device controls :
[04/08/24 07:53:41] crowsnest: 
[04/08/24 07:53:41] crowsnest: Try to start configured Cams / Services...
[04/08/24 07:53:42] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[04/08/24 07:53:43] crowsnest: Starting camera-streamer with Device /base/soc/i2c0mux/i2c@1/ov5647@36 ...
[04/08/24 07:53:43] crowsnest: DEBUG: camera-streamer [cam 1]: Parameters: --http-port=8080 --camera-path=/base/soc/i2c0mux/i2c@1/ov5647@36 --camera-type=libcamera --camera-format=YUYV --camera-snapshot.height=1080 --camera-width=1920 --camera-height=1080 --camera-fps=30.62 --camera-auto_reconnect=1
[04/08/24 07:53:43] crowsnest: DEBUG: camera-streamer [cam 1]: util/http/http.c: ?: HTTP listening on 127.0.0.1:8080.
[04/08/24 07:53:43] crowsnest: DEBUG: camera-streamer [cam 1]: /home/printer/crowsnest/bin/camera-streamer/camera-streamer Version:  (54bdd9d)
[04/08/24 07:53:43] crowsnest: DEBUG: camera-streamer [cam 1]: [0:09:49.861264412] [2970] [1;32m INFO [1;37mCamera [1;34mcamera_manager.cpp:284 [0mlibcamera v0.1.0+118-563cd78e
[04/08/24 07:53:44] crowsnest: DEBUG: camera-streamer [cam 1]: terminate called after throwing an instance of 'std::out_of_range'
[04/08/24 07:53:44] crowsnest: DEBUG: camera-streamer [cam 1]: what():  unordered_map::at
[04/08/24 07:53:44] crowsnest: DEBUG: camera-streamer [cam 1]: xargs: /home/printer/crowsnest/bin/camera-streamer/camera-streamer: terminated by signal 6
[04/08/24 07:53:44] crowsnest: ERROR: Start of camera-streamer [cam 1] failed!
[04/08/24 07:53:44] crowsnest:  ... Done!

brdi, 2024-04-08:
Tried restarting crowsnest manually by opening the crowsnest.cfg  and Save&Restarting it in Mainsail?

After updating to v4.1.8-1 I have/had the same problem.
But after restarting crowsnest manually it worked allright.
 
However the mainsail update screen tells me the Crowsnest " Repo is corrupt "
After a "hard recovery" the camera did not work anymore.

The crowsnest.log indicates an error when checking on ustreamer, and indeed the ustreamer files seem to be missing on the speederpad.

Uninstalling and reinstalling crowsnest using kiauh restored the previous situation with a working camera(after manual restart of crowsnest) and an update screen saying that crwosnests Repo is corrup.

pro2call-nl, 2024-04-08:
hi @brdi,

I followed your tip about reinstalling crowsnest and this worked.

Thank you.

closed 2024-04-08 by pro2call-nl

brdi, 2024-04-09:
Tnx, learning everyday.
Still weird that after reinstalling crowsnest, the update manager indicaties that crowsnest is corrupt.
However, everything is working allthough I've got to restart crowsnest manually before the camera relays a signal....

mryel00, 2024-04-09:
> However, everything is working allthough I've got to restart crowsnest manually before the camera relays a signal....

Have a look at #117, especially at my comment: https://github.com/mainsail-crew/crowsnest/issues/117#issuecomment-1806943132
Maybe you got a similar problem to the one in the issue? If that doesn't help, join our [Discord](https://discord.gg/mainsal) or [Github discussions](https://github.com/orgs/mainsail-crew/discussions)

brdi, 2024-04-11:
@mryel00
Thx.
I did, Problem solved.
When I restarted  even the corwsnest "problem" was solved.
Working like a charm now.

KahnSoft, 2024-08-23:
I have rpi5   and bulleye,  the result was all cameras are deadth brotcominterfce from Chip removed and an new interface creating by full idiot was given, u must include all them shitty libs and must compile that all, has need weeks.  Now 2 years later i do an update on RPI5  (used as Laser Microscope)  And what is happen now ALL CAMERAS AGAIN ERROR OFFLINE AFTER INSTALL UPDATES   WTF What for full idiots are have written this shitty interface. Its not Bull Ey its BullShit    and i write code since 50 years.

NOW We plan remove RPI complettly and replace em with an x86 Windows mini PC   RPI NOT USEABLE IN INDUSTRIAL VISION SYSTEMS EACH WEEK ALL CAMERAS ARE DEATH AND THE DEVLOOPER NEED WEEKS FOR REACTIVATE ALL AGAIN.

In resulte that produce hate to the fucking libcamera shit hole !

meteyou, 2024-08-23:
@KahnSoft how do you use bullseye on rpi5? Bullseye is not compatible with rpi5. The first official raspberry pi is with rpi5 support is bookworm.

mryel00, 2024-08-23:
@KahnSoft Please take your time to read this, as I took my time to answer this.
> mini PC RPI NOT USEABLE IN INDUSTRIAL VISION SYSTEMS EACH WEEK ALL CAMERAS ARE DEATH AND THE DEVLOOPER NEED WEEKS FOR REACTIVATE ALL AGAIN.

A few things about this:
1. We got an issue open with a workaround: https://github.com/mainsail-crew/crowsnest/issues/241#issuecomment-1936695415
2. We cannot do anything about the decisions of the Pi foundation, and this project is nothing from the Pi foundation, but some project that some dudes are working on in their free time
3. You are working in the industry and I'm doing this in my free time for no money. Pay me some salary and I will fix Crowsnest for the Pi5 a lot faster. Without money I'm doing it at my pace, if I find personal time for it

> i write code since 50 years.

Feel free to contribute, even if it's just with your experience to review code. #270 is open and need some review. Also I need some more proffesional reviewer and tester of https://github.com/mryel00/spyglass/pull/84. This will help in finishing the update faster. I'm no proffesional in writing code and I'm still in university, so I don't have any long experience like you or others.

dw-0, 2024-08-23:
> I have rpi5 and bulleye, the result was all cameras are deadth brotcominterfce from Chip removed and an new interface creating by full idiot was given, u must include all them shitty libs and must [...]

Although I lost a few brain cells reading this intellectual outpouring, i think your outrage is kind of hillarous.