mainsail-crew/crowsnest #267: failed webRTC stream when trying to connect remotely

Online122228, 2024-05-27:
### What happened

On the mainsail webpage of my printer the webcam only says failed. when trying https://mainsail.mydomain.example/webcam/webrtc it works and in my home network it works too

### What did you expect to happen

Seeing the webRTC stream remotely

### How to reproduce

have a camera setup using camera-streamer, go outside or just use your cellular data and try to open the mainsail page. You won`t see anything on the webcam

### Additional information

[mainsail nginx config of my main server which points at the pi](https://github.com/mainsail-crew/crowsnest/files/15457476/mainsailentryofmainserver.txt)
[mainsail nginx config of my pi](https://github.com/mainsail-crew/crowsnest/files/15457533/mainsailentryofpi.txt)

mryel00, 2024-05-27:
What does this issue has to do with Crowsnest?
If I understand you correctly, the cam works without problems, doesn't it? The only problem is with Mainsail, right?

Before you start to open an issue on Mainsail, please use our [Discord](https://discord.gg/mainsail) or [Github discussions](https://github.com/orgs/mainsail-crew/discussions). If I misunderstood you, explain the bug with Crowsnest, that you encounter.

mryel00, 2024-05-29:
Closing this because no further input and not an issue of Crowsnest.

closed 2024-05-29 by mryel00