mainsail-crew/crowsnest #268: Can not install/compile anymore, undefined reference to `JxlThreadParallelRunnerCreate@JXL_0'

KlausHans, 2024-05-29:
### What happened

I had a problem with crowsnest, that i couldn't solve (nothing in the logs, Systemd reported only `crowsnest.service: Start request repeated too quickly.`).
So i tried to reinstall it via Kiauh. After uninstalling and reboot i can't get it to compile completely. The problem:

```
...
[100%] Linking CXX static library libdatachannel-static.a
make[5]: Verzeichnis „/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build“ wird verlassen
[100%] Built target datachannel-static
make[4]: Verzeichnis „/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build“ wird verlassen
make[3]: Verzeichnis „/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build“ wird verlassen
make[2]: Verzeichnis „/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build“ wird verlassen
g++ -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -o camera-streamer device/buffer.o device/buffer_list.o device/buffer_lock.o device/buffer_queue.o device/device.o device/device_list.o device/links.o output/http_ffmpeg.o output/http_h264.o output/http_hls.o output/http_jpeg.o output/output.o device/camera/camera.o device/camera/camera_debug.o device/camera/camera_decoder.o device/camera/camera_input.o device/camera/camera_isp.o device/camera/camera_output.o device/camera/camera_pipeline.o device/camera/camera_rescaller.o device/dummy/buffer.o device/dummy/buffer_list.o device/dummy/device.o device/dummy/dummy.o device/libcamera/fake_camera.o device/v4l2/buffer.o device/v4l2/buffer_list.o device/v4l2/debug.o device/v4l2/device.o device/v4l2/device_list.o device/v4l2/device_media.o device/v4l2/device_options.o device/v4l2/v4l2.o util/ffmpeg/remuxer.o util/http/http.o util/http/http_methods.o util/opts/control.o util/opts/fourcc.o util/opts/log.o util/opts/opts.o device/libcamera/buffer.o device/libcamera/buffer_list.o device/libcamera/device.o device/libcamera/libcamera.o device/libcamera/options.o output/rtsp/rtsp.o output/webrtc/webrtc.o html/control.html.o html/index.html.o html/webrtc.html.o cmd/camera-streamer/http.o cmd/camera-streamer/main.o cmd/camera-streamer/opts.o cmd/camera-streamer/status.o third_party/libdatachannel/build/libdatachannel-static.a -lpthread -lstdc++ -lavcodec -lavformat -lavutil -lcamera -lcamera-base  -lliveMedia -lgroupsock -lBasicUsageEnvironment -lUsageEnvironment  -Lthird_party/libdatachannel/build -ldatachannel-static -Lthird_party/libdatachannel/build/deps/usrsctp/usrsctplib -lusrsctp -Lthird_party/libdatachannel/build/deps/libsrtp -lsrtp2 -Lthird_party/libdatachannel/build/deps/libjuice -ljuice-static -lcrypto -lssl
/usr/bin/ld: warning: libjxl_threads.so.0.7, needed by /usr/lib/gcc/aarch64-linux-gnu/12/../../../aarch64-linux-gnu/libavcodec.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../aarch64-linux-gnu/libavcodec.so: undefined reference to `JxlThreadParallelRunnerCreate@JXL_0'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../aarch64-linux-gnu/libavcodec.so: undefined reference to `JxlThreadParallelRunnerDestroy@JXL_0'
/usr/bin/ld: /usr/lib/gcc/aarch64-linux-gnu/12/../../../aarch64-linux-gnu/libavcodec.so: undefined reference to `JxlThreadParallelRunner@JXL_0'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:80: camera-streamer] Fehler 1
make[1]: Verzeichnis „/home/pi/crowsnest/bin/camera-streamer“ wird verlassen
make: *** [Makefile:67: all] Fehler 2
make: *** [Makefile:19: install] Fehler 2
>>>>>> Something went wrong! Please try again...
```

### What did you expect to happen

It compiles successful.

### How to reproduce

- uninstall crowsnest via Kiauh
- install crowsnest via Kiauh

### Additional information

I am using a Raspberry 4 with 4GB ram.

mryel00, 2024-05-29:
Give us some context about your Pi and the OS version. Also make sure that all packages are up to date.

KlausHans, 2024-05-29:
I tried to update my packages after your hint, despite i made that just yesterday. 6 packages had updates, but dpkg reported a fatal error: `files list file for package 'linux-kbuild-6.6.28+rpt' contains empty filename`.

I tried to fix it by removing that list (https://unix.stackexchange.com/questions/425355/x11-common-contains-empty-filename) but then i got the same error for another package. And after that for another. So i think my Pi-OS/Debian installation is beyond repair. 😬
I think the reported problem has nothing to do with crowsnest.

Thank you for your super fast response and hint with the packages.

closed 2024-05-29 by KlausHans