mainsail-crew/crowsnest #269: since libcamera v0.3.0+rpt20240617, rip cam is not working anymore

mpuff, 2024-06-26:
### What happened

since update of the libcamera on raspios bookworm, the camera stream is not working anymore

### What did you expect to happen

to get it work again

### How to reproduce

do a software update of the debian bookworm

### Additional information

[06/25/24 19:26:24] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[06/25/24 19:26:25] crowsnest: Version: v4.1.9-1-gd75a3ae
[06/25/24 19:26:25] crowsnest: Prepare Startup ...
[06/25/24 19:26:25] crowsnest: INFO: Host information:
[06/25/24 19:26:25] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 12 (bookworm)
[06/25/24 19:26:25] crowsnest: Host Info: Kernel: Linux 6.6.31+rpt-rpi-v8 aarch64
[06/25/24 19:26:25] crowsnest: Host Info: Model:  Raspberry Pi 4 Model B Rev 1.5
[06/25/24 19:26:25] crowsnest: Host Info: Available CPU Cores: 4
[06/25/24 19:26:25] crowsnest: Host Info: Available Memory: 1891252 kB
[06/25/24 19:26:25] crowsnest: Host Info: Diskspace (avail. / total): 13G / 28G
[06/25/24 19:26:25] crowsnest: INFO: Checking Dependencies
[06/25/24 19:26:25] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[06/25/24 19:26:25] crowsnest: Dependency: 'find' found in /usr/bin/find.
[06/25/24 19:26:25] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[06/25/24 19:26:25] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[06/25/24 19:26:25] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[06/25/24 19:26:26] crowsnest: Version Control: ustreamer is up to date. (v6.10)
[06/25/24 19:26:26] crowsnest: Version Control: camera-streamer new version available: (6deb9c5) ().
[06/25/24 19:26:26] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[06/25/24 19:26:26] crowsnest:		[crowsnest]
[06/25/24 19:26:26] crowsnest:		log_path: /home/pi/printer_data/logs/crowsnest.log
[06/25/24 19:26:26] crowsnest:		log_level: verbose
[06/25/24 19:26:26] crowsnest:		delete_log: true
[06/25/24 19:26:26] crowsnest:		no_proxy: false
[06/25/24 19:26:26] crowsnest:		
[06/25/24 19:26:26] crowsnest:		[cam 1]
[06/25/24 19:26:26] crowsnest:		mode: camera-streamer
[06/25/24 19:26:26] crowsnest:		
[06/25/24 19:26:26] crowsnest:		enable_rtsp: false
[06/25/24 19:26:26] crowsnest:		rtsp_port: 8554
[06/25/24 19:26:26] crowsnest:		port: 8080
[06/25/24 19:26:26] crowsnest:		device: /base/soc/i2c0mux/i2c@1/ov5647@36
[06/25/24 19:26:26] crowsnest:		resolution: 1280x720
[06/25/24 19:26:26] crowsnest:		max_fps: 30
[06/25/24 19:26:26] crowsnest:		
[06/25/24 19:26:26] crowsnest:		
[06/25/24 19:26:26] crowsnest: INFO: Detect available Devices
[06/25/24 19:26:26] crowsnest: INFO: Found 1 total available Device(s)
[06/25/24 19:26:27] crowsnest: Detected 'libcamera' device -> /base/soc/i2c0mux/i2c@1/ov5647@36
[06/25/24 19:26:27] crowsnest: 'libcamera' device(s) resolution(s) :
[06/25/24 19:26:27] crowsnest:		0 : ov5647 [2592x1944 10-bit GBRG] (/base/soc/i2c0mux/i2c@1/ov5647@36)
[06/25/24 19:26:27] crowsnest:		    Colorspace: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
[06/25/24 19:26:27] crowsnest:		                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
[06/25/24 19:26:27] crowsnest:		                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
[06/25/24 19:26:27] crowsnest:		                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
[06/25/24 19:26:27] crowsnest:		
[06/25/24 19:26:27] crowsnest: 'libcamera' device controls :
[06/25/24 19:26:27] crowsnest: 
[06/25/24 19:26:27] crowsnest: Try to start configured Cams / Services...
[06/25/24 19:26:28] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[06/25/24 19:26:30] crowsnest: Starting camera-streamer with Device /base/soc/i2c0mux/i2c@1/ov5647@36 ...
[06/25/24 19:26:30] crowsnest: ERROR: Start of camera-streamer [cam 1] failed!
[06/25/24 19:26:30] crowsnest:  ... Done!

mpuff, 2024-06-26:
when I do "libcamera-still -o img.jpg -v" in cli, I get an image, so the camera is definitely working.

closed 2024-06-26 by github-actions[bot]

mryel00, 2024-06-26:
There is also a duplicate of this. I can search for it later.

mryel00, 2024-06-27:
For reference: Duplicate of #166

mpuff, 2024-06-27:
thank you very much, it is working again :)