mainsail-crew/crowsnest #271: Cannot open device /base/soc/i2c0mux/i2c@1/imx519@1a

kzuahelp, 2024-07-17:
### What happened

Crowsnest not working with imx519 module (Arducam 16M) on Pi Zero W2.

### What did you expect to happen

Should work because `libcamera-still -t 5000 -n -o test.jpg` works

### How to reproduce

 There is no errors in `~/printer_data/logs/crowsnest.log`, but when you manually start
`/usr/local/bin/crowsnest -c /home/pi/printer_data/config/crowsnest.conf` there is error line

```
INFO: Detect available Devices
INFO: Found 1 total available Device(s)
Detected 'libcamera' device -> /base/soc/i2c0mux/i2c@1/imx519@1a
'libcamera' device(s) resolution(s) :
'libcamera' device controls :

Try to start configured Cams / Services...
INFO: Configuration of Section [cam DEFAULT] looks good. Continue ...
 ... Done!
Cannot open device /base/soc/i2c0mux/i2c@1/imx519@1a, exiting.
Starting camera-streamer with Device /base/soc/i2c0mux/i2c@1/imx519@1a ...
```

How to debug this?

closed 2024-07-17 by github-actions[bot]

mryel00, 2024-07-17:
Best to debug is to get on our Discord, as this is most likely no bug, but just some weird Arducam thing going on.

kzuahelp, 2024-07-17:
> Best to debug is to get on our Discord, as this is most likely no bug, but just some weird Arducam thing going on.

I tried to switch to /dev/video0, it starts without error, but 'Can't queue buffer' when trying to connect.

```
libcamera-still -t 5000 -n -o test.jpg
[0:33:25.066591391] [4105]  INFO Camera camera_manager.cpp:297 libcamera v0.0.0+4367-ad9428b4
[0:33:25.195462077] [4106]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx519@1a to Unicam device /dev/media3 and ISP device /dev/media1
[0:33:25.195615983] [4106]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:33:25.200486312] [4105]  INFO Camera camera.cpp:1033 configuring streams: (0) 2328x1748-YUV420
[0:33:25.201696160] [4106]  INFO RPI vc4.cpp:572 Sensor: /base/soc/i2c0mux/i2c@1/imx519@1a - Selected sensor format: 2328x1748-SRGGB10_1X10 - Selected unicam format: 2328x1748-pRAA
[0:33:30.522909523] [4105]  INFO Camera camera.cpp:1033 configuring streams: (0) 4656x3496-YUV420 (1) 4656x3496-SRGGB10_CSI2P
[0:33:30.537860300] [4106]  INFO RPI vc4.cpp:572 Sensor: /base/soc/i2c0mux/i2c@1/imx519@1a - Selected sensor format: 4656x3496-SRGGB10_1X10 - Selected unicam format: 4656x3496-pRAA
Still capture image received
```

```
▶ ./bin/camera-streamer/camera-streamer --camera-path=/base/soc/i2c0mux/i2c@1/imx519@1a --camera-fps=15 --http-port=8080 --camera-width=640 --camera-height=480
./bin/camera-streamer/camera-streamer Version:  (f1627aa)
device/v4l2/device.c: CAMERA: Can't open device: /base/soc/i2c0mux/i2c@1/imx519@1a
device/device.c: CAMERA: Can't open device: /base/soc/i2c0mux/i2c@1/imx519@1a
device/camera/camera_input.c: CAMERA: Listing available v4l2 devices:
unicam (platform:3f801000.csi):
	/dev/video0
	/dev/video1
	/dev/media3

bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media0

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media1
	/dev/media2
```

```
./bin/camera-streamer/camera-streamer --camera-path=/dev/video0 --camera-fps=15 --http-port=8080 --camera-width=640 --camera-height=480                      
./bin/camera-streamer/camera-streamer Version:  (f1627aa)
device/v4l2/device.c: CAMERA: Device path=/dev/video0 fd=14 opened
device/v4l2/device_options.c: CAMERA: Configuring option horizontalflip (00980914) = 0
device/v4l2/device_options.c: CAMERA: Configuring option verticalflip (00980915) = 0
device/buffer_list.c: CAMERA:capture: Using: 640x480/pRAA, buffers=3, bytesperline=800, sizeimage=0.4MiB
device/buffer_list.c: CAMERA:capture: Opened 3 buffers. Memory used: 1.1 MiB
device/v4l2/device.c: ISP: Device path=/dev/video13 fd=19 opened
device/buffer_list.c: ISP:output: Using: 640x480/pRAA, buffers=3, bytesperline=800, sizeimage=0.4MiB
device/buffer_list.c: ISP:output: Opened 3 buffers. Memory used: 0.0 MiB
device/v4l2/buffer_list.c: ISP:capture: Device path=/dev/video14 fd=20 opened
device/buffer_list.c: ISP:capture: Using: 640x480/YUYV, buffers=3, bytesperline=1280, sizeimage=0.6MiB
device/buffer_list.c: ISP:capture: Opened 3 buffers. Memory used: 1.8 MiB
device/v4l2/device.c: SNAPSHOT: Device path=/dev/video31 fd=24 opened
device/buffer_list.c: SNAPSHOT:output:mplane: Using: 640x480/YUYV, buffers=3, bytesperline=1280, sizeimage=0.6MiB
device/buffer_list.c: SNAPSHOT:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Using: 640x480/JPEG, buffers=3, bytesperline=0, sizeimage=4.0MiB
device/buffer_list.c: SNAPSHOT:capture:mplane: Opened 3 buffers. Memory used: 12.0 MiB
device/v4l2/device.c: VIDEO: Device path=/dev/video11 fd=28 opened
device/buffer_list.c: VIDEO:output:mplane: Using: 640x480/YUYV, buffers=3, bytesperline=1280, sizeimage=0.6MiB
device/buffer_list.c: VIDEO:output:mplane: Opened 3 buffers. Memory used: 0.0 MiB
[... 27 lines cut ...]
util/http/http.c: HTTP8080/0: Request 'GET' '/' 'action=stream'
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
util/http/http.c: HTTP8080/0: Client disconnected 127.0.0.1.
```

I'm confused in device naming depend on streaming solutions