mainsail-crew/crowsnest #277: AVIF returned instead of JPEG

crysxd, 2024-11-16:
### What happened

I'm the dev of OctoApp 👋 
A user reached out and reported that the app wasn't able to display their webcam, but if ustreamer was used the webcam was visible in the app. We did some digging and found out that in his case crowsnest returns AVIF images from `/stream` and `/snapshot` (sample attached). Because iOS doesn't natively support AVIF, the stream can't be shown in the app.

Opening the snapshot with a text editor reveals the AVIF header:
[image]

This seems to be linked to the webcam used, a Logitech C615, as I can't reproduce the behavior on my end using a C920 and the same  v4.1.9-1-gd75a3aeb version of crowsnest.

### What did you expect to happen

crowsnest should always return JPEG images to guarantee compatibility with all clients. AVIF should be opt-in.

### How to reproduce

Unclear, seems to be dependent on the C615 camera.

### Additional information

Here is a image obtained from `/snapshot`, which is in AVIF format. Most browsers can display it.
[image]

This is a cURL capture of the stream:
[stream.bin.zip](https://github.com/user-attachments/files/17783944/stream.bin.zip)

mryel00, 2024-11-16:
There are a few things that are bothering me about this issue atm.
This doesn't seem like a bug, but more like configuration issue. Why? Because you wrote that the issue occurs with `ustreamer` but the config is containing `mode: camera-streamer`.
Instead of a config, we would also need a log, to see supported formats. If you don't want the username to be "leaked", you should edit the log beforehand.
Furthermore `ustreamer` only supports `YUYV, YVYU, UYVY, RGB565, RGB24, BGR24, MJPEG, JPEG` and defaults to `YUYV`. So if nothing is specified it will use `YUYV`. Therefore it's either way no bug with Crowsnest but a `ustreamer` bug, if it isn't a config error.
I recommend that either you or the user join our Github Discussions or Discord that we can contiue there to debug this further.

closed 2024-11-16 by github-actions[bot]

crysxd, 2024-11-16:
@mryel00 Thanks for your quick reply. I think you misread my initial comment, the issue does *not* occur with ustreamer. It only happens with camerastreamer, sorry not being clear about this. I will ask the user to get in contact because I'm kinda just the middle man.

mryel00, 2024-11-16:
Oh yeah I misread it, you are totally right. Sry for that 😅
If the user gets in contact, it would be the best. If it's indeed an issue with the code I will reopen this to keep track of it. But atm I would still not know what the problem is, without logs, as the supported formats are listed like the following for `camera-streamer`: `DEFAULT, YUYV, YUV420, YUYV, NV12, NV21, MJPG, MJPEG, JPEG, H264, RG10, GB10P, RG10P, BG10P, RGB565, RGBP, RGB24, RGB, BGR`

crysxd, 2024-11-16:
Totally clear, more info is needed. As ustreamer is a viable solution for the user, I'm not sure if they will pursue this. I asked to them to reach out on discord and provide the required information. My primary goal was to have this documented, thus the issue :)