mainsail-crew/crowsnest #278: camera-streamer fails to compile on Bookworm

mryel00, 2024-11-30:
### What happened

Currently Crowsnest fails to build on Bookworm installations. This is due to an error with camera-streamer during compilation.

### What did you expect to happen

camera-streamer compilation without errors.

### How to reproduce

Try to install Crowsnest on a Bookworm installation.

### Additional information

### This error only occurs on Bookworm installations!!!
[Libcamera](https://github.com/raspberrypi/libcamera) added a new `ControlType`, `Point`, with v0.3.2+rpt20241119. camera-streamer has an implementation error, that triggers, as soon as a new `ControlType` gets added, as it doesn't implement a default behavior for a not implemented `ControlType`.
This will be fixed with the merge of either https://github.com/ayufan/camera-streamer/pull/160 into the camera-streamer main branch, or changing the repo, that gets cloned during installation, to a repo containing the fix.

crim491, 2024-12-01:
I recreated this issue on a Rpi 3B using a fresh raspbian minimal install while installing klipper. The image that crowsnest is being built on was from the foundation dated 2024-11-19-raspios-bookworm-arm64-lite. 

[image]

mryel00, 2024-12-01:
To be clear, this issue will happen on all Bookworm installations, as we are updating the packages, before installing Crowsnest.

The merge into camera-streamer might take a few days, so you can do the following workaround until that:
Set `CN_CAMERA_STREAMER_REPO` in `~/crowsnest/tools/libs/config.sh` to `https://github.com/mryel00/camera-streamer.git`. You should also make sure that the camera-streamer folder is deleted before you try the installation again: `rm -rf ~/crowsnest/bin/camera-streamer`

After the installation succeeded you should reset the file with `cd ~/crowsnest && git restore tools/libs/config.sh`

mryel00, 2024-12-01:
I will not change the compiler settings, just to circumvent this warning. This would be even more work than the actual fix, as the curernt codebase is really annoying...
If you would look at the PR, it's implementing a clean fix instead of circumventing "errors" by just ignoring them. The warning is there for a good reason, as we would get weird behavior as soon as someone would try to use `ControlTypePoint` in any way.

It might sound rude, but please stop sending more messages that won't fix anything. I'm the maintainer of this project, if you didn't see it yet. I know what this "error" means, I know how to fix it and I decided to only fix this correct within camera-streamer. So either I will change the repo to my own repo, or, what's more likely, I will wait for the camera-streamer dev to merge it. He already knows of this PR and will need some days before merging it.

So please just have patience, instead of sending informations that won't fix anything. The official workaround for now, is to use my branch of camera-streamer temporarily for one compilation.

mryel00, 2024-12-01:
> Its super annoying to encounter compile time problems like this while executing scripts on an unrelated but operational project..i.e. KIAUH

You should understand what KIAUH is doing. It's executing install scripts. So if you encounter any issues, it's nearly always connected to the project that you want to install (I guess you understood that much already). In case of Crowsnest it's even worse, as you are "just starting" another script that will just compile something else again. So Crowsnest is also completely operational and basically unrelated to camera-streamer, but you encounter a compile time error. That annoys me a lot more as it was again just introduced by some update again and hinders people to install it.

> I am just reporting what I saw. Maybe something was missed or overlooked.

I'm thankful for that, but sometimes it's getting annoying if people just write, "I have the same issue", "That solved it for me too", .... It was just the content of your messages both didn't bring anything new to the table, that was the core idea of why I wrote "stop sending more messages". You just brought up a screenshot at first of the error, that was still fine, even if I already linked the fix. Then you pointed out the only error in your screenshot later, that is basically fixed with the PR that I already linked. This just lets me assume that you maybe don't have that much knowledge, that you obviously seem to have.

I hate how I always type that much, but I want that people can understand my intention behind decisions or messages. So now you hopefully understand, why I got a bit annoyed by your second message.

crim491, 2024-12-01:
I only just realized I posted to the wrong project. My bad.

spikeygg, 2024-12-01:
> To be clear, this issue will happen on all Bookworm installations, as we are updating the packages, before installing Crowsnest.

This totally works! Thanks for putting this together. I found this [thread on Reddit](https://www.reddit.com/r/klippers/comments/16qefe0/is_it_possible_to_use_logitech_usb_webcam_in/) that mentioned recompiling the camera-streamer but when I tried it died -- then looking through the issues I found yours about bookworm. I just got mine to compile, too. Hopefully, this change gets merged at some point.

BTW, in order to get my WebRTC to function correctly (to work in Fluidd/Mainsail and to be streamable by other devices on my network) I had to include: `custom_flags: --camera-format=YUYV` in every camera config which I [found on the discord server posted by Mr Yel](https://discord.com/channels/758059413700345988/1298019517682679979/1298036082830475296).

mryel00, 2024-12-07:
Fixed temporarily by changing to my camera-streamer fork containing the fix.
I talked to Ayufan (dev of camera-streamer) and he might be able to merge it in about 1-2 weeks (no guarantee). After the merge I will swap back the repo to the official camera-streamer repo.

PR #280 (2024-12-07): fix: change cs branch temporarily to fix compiling error
fixes #278 
This is just a temporary change, until https://github.com/ayufan/camera-streamer/pull/160, or some other fix, will be merged into camera-streamer.
closed 2024-12-07 by mryel00