mainsail-crew/crowsnest #282: MJPEG is slow, suggest YUYV in docs/config

strayr, 2024-12-14:
### Is your feature request related to a problem? Please describe

Some webcams are painful slow, crowsnest is brilliant for pi cam and on raspberry pi hardware. On other platforms with generic USB webcams, ustreamer in mjpeg mode is just painful. This reddit thread describes the pain https://www.reddit.com/r/klippers/comments/18v8cwh/extremely_low_fps_on_crowsnest_webcam/

I was particularly confused when i moved a webcam from one thin client to another and suddenly it went from a massive framerate to near unusable.

### Describe the solution you'd like

Suggest users try

`custom_flags: --format=YUYV `

if they are experiencing very slow framerates with usb webcams. This could be in the docs, or as a comment in the config file, the latter is more likely to be helpful. Seriously, just something like this will save many users screaming. 

```
#custom_flags: --format=YUYV         # try this if you are experiencing very low framerates with a usb webcam using ustreamer.
```

### Describe alternatives you've considered

Buying a raspberry pi and a pi cam - really expensive, i have usb webcams and thin clients that otherwise work.

mryel00, 2024-12-14:
You are right, I saw that sometimes in our support forum too. I'm currently rewriting some parts of the docs for v5, so I will add this under common problems.

mryel00, 2025-01-05:
`custom_flags: --format=YUYV` can also help if the cam is not putting out any video.
This was discovered during support on Discord. There is no error or similar to identify it. See [this post](https://discord.com/channels/758059413700345988/1298019517682679979) on Discord.

tastyratz, 2025-09-15:
Understandable. My goal here was to document my findings and my use case for others to come across and hopefully this information can contribute towards better documentation in the future to help others similar to me.

This git report showed up when I was sleuthing through discord/ the web for answers so if nothing else that's a start.

mryel00, 2025-11-01:
It's written there:
> The big fix for me however was instead using --encoder=HW --allow-truncated-frames

So just add `--encoder=HW --allow-truncated-frames` to your `custom_flags` like this:
`custom_flags: --encoder=HW --allow-truncated-frames`
If you need more help, join the [Discord](https://discord.gg/mainsail) or open a [GitHub Disussion](https://github.com/orgs/mainsail-crew/discussions).

P.S. Please do not cite a whole message with quote reply. Cite specific things or ping users instead.