mainsail-crew/crowsnest #284: `v4l2ctl:` Parameters being skipped, even though the device supports all parameters

Twinki14, 2024-12-28:
### What happened

When setting `v4l2ctl:` to `--device=/dev/video0 --set-fmt-video=width=1920,height=1080,pixelformat=MJPG` or `width=1920,height=1080,pixelformat=MJPG` crowsnest skips over the parameters with a "not available for" in it's logs

```
[12/28/24 18:17:30] crowsnest: V4L2 Control: Device: [cam 1]
[12/28/24 18:17:30] crowsnest: V4L2 Control: Options: width=1920,height=1080,pixelformat=MJPG
[12/28/24 18:17:30] crowsnest: V4L2 Control: Parameter 'width=1920' not available for '/dev/video0'. Skipped.
[12/28/24 18:17:30] crowsnest: V4L2 Control: Parameter 'height=1080' not available for '/dev/video0'. Skipped.
[12/28/24 18:17:30] crowsnest: V4L2 Control: Parameter 'pixelformat=MJPG' not available for '/dev/video0'. Skipped.
[12/28/24 18:17:30] crowsnest: Starting ustreamer with Device /dev/video0 ...
```

The width, height, and pixel format are all supported according to `crowsnest.log`
```
[12/28/24 18:17:29] crowsnest: /dev/v4l/by-id/usb-046d_HD_Pro_Webcam_C920_3FC7976F-video-index0 -> /dev/v4l/by-id/usb-046d_HD_Pro_Webcam_C920_3FC7976F-video-index0
[12/28/24 18:17:29] crowsnest: Supported Formats:
[12/28/24 18:17:29] crowsnest:          [0]: 'YUYV' (YUYV 4:2:2)
...
[12/28/24 18:17:29] crowsnest:          [1]: 'MJPG' (Motion-JPEG, compressed)
...
[12/28/24 18:17:29] crowsnest:          Size: Discrete 1920x1080
[12/28/24 18:17:29] crowsnest:          Interval: Discrete 0.033s (30.000 fps)
[12/28/24 18:17:29] crowsnest:          Interval: Discrete 0.042s (24.000 fps)
[12/28/24 18:17:29] crowsnest:          Interval: Discrete 0.050s (20.000 fps)
[12/28/24 18:17:29] crowsnest:          Interval: Discrete 0.067s (15.000 fps)
[12/28/24 18:17:29] crowsnest:          Interval: Discrete 0.100s (10.000 fps)
[12/28/24 18:17:29] crowsnest:          Interval: Discrete 0.133s (7.500 fps)
[12/28/24 18:17:29] crowsnest:          Interval: Discrete 0.200s (5.000 fps)
```

Yet for some reason the parameters are skipped and my logitech c920 remains stuck in YUYV pixel format, nothing in my `v4l2ctl:` appears to work.

When using `v4l2-ctl --device=/dev/video0 --set-fmt-video=width=1920,height=1080,pixelformat=MJPG` directly while crowsnest is stopped, v4l2-ctl accepts it and my C920 is configured as expected

### What did you expect to happen

Logitech C920 switching to the expect flags in the `v4l2ctl:` config field

### How to reproduce

- Install a Logitech C920 and configure it so that crowsnest picks it up
- Try setting `v4l2ctl:` to something basic/simple, like changing the pixel format

mryel00, 2024-12-28:
You are just using it wrong. In our docs you can find an example: https://crowsnest.mainsail.xyz/configuration/cam-section#v4l2ctl
It's not there to use it like the command. You basically define the device and resolution with other parameters too, so no idea, why you are even trying to change resolution like that. Furthermore the pixelformat gets set by `ustreamer`/`camera-streamer`, you have to use their parameters with `custom_flags` if you want to change it to something different than `MJPG`

closed 2024-12-28 by mryel00

Twinki14, 2024-12-28:
> You are just using it wrong. In our docs you can find an example: https://crowsnest.mainsail.xyz/configuration/cam-section#v4l2ctl It's not there to use it like the command. You basically define the [...]

I'm not trying to change it something other than MJPG, I'm trying to change it to something other than YUYV.

I do have `custom_flags: --format=JPEG` in my config as well but that isn't reflected in v4l2ctl

Twinki14, 2024-12-28:
> You are just using it wrong. In our docs you can find an example: https://crowsnest.mainsail.xyz/configuration/cam-section#v4l2ctl It's not there to use it like the command. You basically define the device and resolution with other parameters too, so no idea, why you are even trying to change resolution like that. Furthermore the pixelformat gets set by `ustreamer`/`camera-streamer`, you have to use their parameters with `custom_flags` if you want to change it to something different than `MJPG`. `MJPG` is getting set if it's getting detected.

How should I be using it then?

ustreamer is in JPEG format, yet v4l2ctl is stuck in YUYV and I'd like to change it to MPJEG, using `v4l2ctl: pixelformat=MPJPEG` is skipped, and my c920 is still stuck on YUYV

```
v4l2-ctl --all
Driver Info:
        Driver name      : uvcvideo
        Card type        : HD Pro Webcam C920
        Bus info         : usb-0000:02:00.0-3
        Driver version   : 6.5.13
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'YUYV' (YUYV 4:2:2)
        Field             : None
        Bytes per Line    : 3840
        Size Image        : 4147200
        Colorspace        : sRGB
        Transfer Function : Rec. 709
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Default (maps to Limited Range)
        Flags             : 
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 1920, Height 1080
        Default     : Left 0, Top 0, Width 1920, Height 1080
        Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 5.000 (5/1)
        Read buffers     : 0

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=255
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2000 max=6500 step=1 default=4000 value=2362 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
                                1: Manual Mode
                                3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=250 value=498 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=250 step=5 default=0 value=30 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=100 max=500 step=1 default=100 value=100
```

```
[cam 1]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video0                     # See Log for available ...
resolution: 1920x1080                     # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
custom_flags: --format=JPEG
v4l2ctl: pixelformat=MJPG
```

mryel00, 2024-12-28:
Please use the appropriate channels for help. Either our [discussions](https://github.com/orgs/mainsail-crew/discussions) or our [Discord](https://discord.gg/mainsail).
It's not a bug and countless people are using a C920 without such an issue. Just remove any `custom_flags` and/or `v4l2ctl` parameter you got set, as it's automatically set by Crowsnest. Furthermore you should use `MJPG` and not `JPEG`, as those are two different formats.
The debug log will show you the parameters we set, then you can see that `-m MJPEG` is already set by us.

mryel00, 2024-12-28:
> ustreamer is in JPEG format, yet v4l2ctl is stuck in YUYV and I'd like to change it to MPJEG, using v4l2ctl: pixelformat=MJPEG is skipped, and my c920 is still stuck on YUYV

It doesn't matter in what `v4l2ctl` is stuck, as you cannot use the cam with two different parameters. It's possible that `v4l2ctl` is reporting you wrong values. That's also the case for e.g. the zoom on some Logitech cams. `v4l2ctl: pixelformat=MPJPEG` will not work as ustreamer is changing it again, but it's in theory correct. As `JPEG` is not supported, it's most likely set back to `YUYV`.