mainsail-crew/crowsnest #286: Support for HTTPS/SSL

WolfeCub, 2025-01-17:
### Is your feature request related to a problem? Please describe

I currently have fluidd and moonraker setup properly with SSL and they work fine. However the browser is upset about loading mixed content. 

### Describe the solution you'd like

Ability to specify SSL cert pub and private key and have the webcam streamed over https.

mryel00, 2025-01-17:
Crowsnest is just a wrapper to make the setup and use of ustreamer and camera-streamer easier. Therefore we cannot add such support in Crowsnest. Ustreamer or camera-streamer need to support it and then you can specify it with the `custom_flags`.

closed 2025-01-17 by mryel00

meteyou, 2025-01-17:
you can also use nginx as a reverse proxy incl. ssl (should already be set up in your case)

WolfeCub, 2025-01-20:
@mryel00 It could setup a proxy through nginx automatically to save the user having to do it manually.

Is that not something that would be considered?

mryel00, 2025-01-20:
Sry, but that's not in consideration for two reasons:
Might be "breaking" current common installation scenarios, as it would also setup a nginx reverse proxy, like KIAUH is doing for Mainsail and Fluidd installations. So adding an installer in Crowsnest, would need to consider those installations that we don't add a second http reverse proxy on top.

I don't want to add further "tasks" to the Crowsnest installer.
How should we e.g. handle removing Crowsnest? Just removing the nginx config? Uninstall nginx? We cannot just uninstall nginx.
How would we reference the ssl certificate? Should we release something in the docs that says, "btw we have setup a nginx reverse proxy, your cert should be at this path and named like that", or ask during installation for the certificate? People then might reinstall Crowsnest, just to get the certificate at the correct place.
Why not just writing a guide to setup the whole nginx reverse proxy thing instead? It's not that much tbh, so we wouldn't need it in the installer, for those who really want it. But writing a good guide takes time and I bet there are already plenty enough out there.

The user group of a https proxy would be pretty small. People installing Crowsnest without Mainsail/Fluidd is also pretty small, therefore the user group of Crowsnest without a basic reverse proxy, is also pretty small. Adding it to the installer sounds like more headache in support, than it would benefit the project.

If you found a good guide, feel free to share it here or write one up and send it. But for the above reasons, I don't plan to add it now or any time in the future.