mainsail-crew/crowsnest #293: Latest crowsnest update failed had to do a hard reset not the camera no longer works.

jonnyc6, 2025-02-25:
### What happened

Hi Everything was working fine, system said there was an update so I updated it failed and said i had to click to hard reset, which I did now Crowsnest is having an issue with ustreamer log file attached.  Been playing for a while and can't work out what's wrong!

[crowsnest.log](https://github.com/user-attachments/files/18972289/crowsnest.log)

### What did you expect to happen

Update and work

### How to reproduce

do the update

jonnyc6, 2025-02-25:
Means the webcam isn't working although it is there and ls -l /dev/v4l/by-id lists the device fine.

mryel00, 2025-02-25:
You reset your installation, therefore you need to reinstall

closed 2025-02-25 by mryel00