mainsail-crew/crowsnest #295: Version that works on "Tina-Linux" which runs on openwrt hardware on the Creality K2 Plus

QuackenAudio, 2025-03-30:
### Is your feature request related to a problem? Please describe

I'd like to see a version that works on the creality K2 Plus
the k2 plus doesnt use standard linux, its using a sort of stripped down version called Tina Linux
bash isnt the same directory as normal, files will need to be edited to work with #!/bin/sh instead

its an oddball, I know this isnt likely
but I will gladly test anything you can make that  will install on the k2

### Describe the solution you'd like

I would like to see it with an install script we can run instead of having to install with mainsail thru a raspberry pi writer.

### Describe alternatives you've considered

I have tried to do it myself, but I'm not that linux smart and I am not a programmer

### Additional information

will gladly try to supply files and folder structure if needed (as best as I can, you may need to tell me how to show you this info if you need it)

mryel00, 2025-03-31:
Sry, but I will not consider adding support for such devices. There are other projects out there that are trying to achieve support on such devices, like this one https://github.com/Guilouz/Creality-Helper-Script-Wiki

> I would like to see it with an install script we can run instead of having to install with mainsail thru a raspberry pi writer

You mean you would like something like `make install`? https://crowsnest.mainsail.xyz/setup/installation
You also most likely mean MainsailOS and not Mainsail, but Crowsnest is not exclusive to either of these, as you can read here: https://github.com/mainsail-crew/crowsnest?tab=readme-ov-file#compatibility

closed 2025-03-31 by mryel00

mryel00, 2025-03-31:
Like I said, I do not consider adding support. I'm not exactly familiar with that Linux, I thought it might be similar to the OS used by e.g. the K1, therefore the creality helper script suggestion.
We only support the file structure used by a "normal" klipper installation. I don't want to add support for every file structure some company decides to use just to save some money.
If we go by "importance" the K1 structure would be the first one we should officially add, as those are currently used by a lot more users after that the K2/Tina Linux. Then we would need to maintain those in the future too, without having such machines and that just because Creality decided to use a stripped down Linux.
This project is open source, feel free to create something similar as the creality helper script and link it here. If you are not a programmer you can try your luck with AI or try to learn everything needed on your own, but I won't invest my personal free time to support these devices.