mainsail-crew/crowsnest #296: Enable rotation of camera feed

jarruda, 2025-04-13:
### Is your feature request related to a problem? Please describe

Right now webcams are limited to hflip and vflip, and don't support 90 or 270 degree rotations if the backend doesn't support it (e.g. camera-streamer and most rpi cameras.)  This limits users to mounting cameras in landscape orientations only.

### Describe the solution you'd like

The browser is capable of performing a rotation via css, however.  You can verify this by adding `transform: rotation(270deg)` to the style of the `video` tag when using the WebRTC service. The request is to expose an option for this in crowsnest to allow users to take advantage of this.

mryel00, 2025-04-13:
I think this PR was meant for Mainsail?
Crowsnest has no influence on the stream, as well as no influence on the service. Crowsnest can start the backend for you.

If this was meant for Mainsail, let me know and I will transfer this FR.

jarruda, 2025-04-13:
Ah, yes, sorry.