mainsail-crew/crowsnest #298: camera-streamer not installed on Debian 13

Abrynos, 2025-05-19:
### What happened

Cloned crowsnest on a current debian testing (aka. "trixie") installation and ran `sudo make install`.

Install script refuses to build camera-streamer.

Installation "succeeds" but starting the service fails because dependency camera-streamer is not found. 

### What did you expect to happen

Installation and service-start succeeds.

### How to reproduce

- Get a good old Raspberry Pi 3B+
- Install corresponding debian testing from [here](https://raspi.debian.net/daily-images/)
- Add user you want to use for mainsail/crowsnest
- Follow installation instructions [here](https://crowsnest.mainsail.xyz/setup/installation)

### Additional information

I've installed camera-streamer `.deb` directly from the upstream repositories release and manually ran the binary in `/usr/bin` with hand-crafted arguments (after looking through the crowsnest scripts, I knew what they are supposed to look like); This works perfectly fine. 

I've also tried `ln -s /usr/bin/camera-streamer /home/$USER/crowsnest/bin/camera-streamer/camera-streamer`, but crowsnest exits way before that and I didn't have the time to tinker too much with your scripts.

mryel00, 2025-05-19:
Camera-streamer only gets build if we have libcamera available. Without libcamera the WebRTC feature won't work. Afaik the plain Debian doesn't have libcamera installed and might not be able to install it, otherwise our script should detect it and build camera-streamer.
Without the WebRTC functionality camera-streamer doesn't make sense to install, as it's just more unstable compared to ustreamer.

If you can run camera-streamer with WebRTC and CSI camera support on a plain Debian and actually use them, then let me know and we might be able to add support for it in the future.

mryel00, 2025-05-22:
I also read through your comment again and have some opinion on 2 of these points:
> No Desktop environment (including BS like libreoffice, which I personally am never going to use on a raspberry pi)

RPiOS has a lite version of their images, without desktop environment and UI applications. If I talk about RPiOS I normally talk about their lite images. I would never recommend a Desktop environment on any Pi, besides the Pi5 maybe, but that isn't that good for printers or cameras tbh.

> Timely updates; As an example, rpi os used armhf32 for the longest time instead of actual arm64 (Debian guys supported it); 

I don't know what your point with that is. What is your problem with a 32bit kernel on a device using less than 4GB of RAM? It's kinda useless imo to use a 64bit kernel just because it's the "newer" standard, if you cannot even make proper use of it.

> AFAIK raspberry pi os is still based on stable release of debian rather than rolling release

Debian itself is not a rolling release OS. An image is called stable for a good reason. You can ofc use the testing sources to get to a rolling release with pretty stable packages, but that's not stable sometimes.
The bigger problem behind all this is, that Pi images are not really made for in place upgrades. This is due to their specific software and drivers they are using for their pi specific components, like CSI cameras and DSI screens. Doing in place upgrades had always a risk of breaking your image or some kind of software stack, so they always warned the user to not do it, or only with caution. For Bookworm they didn't even provide a guide anymore.
Two examples of software stacks breaking would be the CSI stack from Buster to Bullseye, as they deprecated their proprietary stack to use libcamera instead. The second example would be the desktop images from Bullseye to Bookworm, as they switched from xserver to wayland. Doing an inplace upgrade in these two scenarios would break those two stacks or not install the newer versions of the packages, everyone else was relying on. Manually installing them would break stuff too.

Remember this software is just meant to be a simple wrapper for camera streaming services that we use on our 3D printers. A 3D printer is not a security critical component and all the software we use for it is not made with security in mind. Therefore using "older" packages on the pi, is not a big security hole, imho.

There is enough to criticize the raspberry pi foundation for (e.g. your third point), but imo these two/three points I cited above are not actually problems speaking against the pi images.

closed 2025-05-27 by mryel00

omgsideburns, 2025-07-21:
This is marked closed, but as my search led me here, I'd like to leave a fix for the issue I ran into. 

I migrated from the Raspberry Pi Desktop (Deb 12) to Debian 13.  Not a fresh install, just forced an upgrade by changing repos.

After updating Crowsnest via KIAUH I ran into an issue where it was giving me 

`[...] crowsnest: Dependency: 'camera-streamer' not found. Exiting!`

I dug through the git and traced the dependency checks until I ran into where it checks for a rpi os by looking for "/etc/rpi-issue"

Found the offending file, a remnant from it's Raspberry Pi Desktop OS past, renamed the file to pi-issue.backup and the problem was fixed.

meteyou, 2025-07-21:
@omgsideburns Thank you for your post and for sharing your solution to the camera-streamer issue. It's great that you were able to find and fix the root cause!

However, we'd like to clarify a few things regarding your approach, as they have significant implications for the stability and support of Crowsnest:

1. Upgrading from Raspberry Pi Desktop to Debian 13: The type of upgrade you performed (a forced upgrade by changing repositories) is explicitly not recommended by Raspberry Pi themselves for production use. Here's a quote on this:
    > First, we will never officially recommend or support doing this - our recommended approach will always be to start with a clean trixie image and to install whatever programs and data you need from the previous bookworm image. While we do what we can to make sure that it is possible to upgrade a bookworm image, we can only test clean images; we cannot test every combination of software and configuration that a user might have applied, and any such changes can cause the update to fail in a fashion we could not predict which may leave you with a broken and unrecoverable system. You should not attempt to upgrade any system on which you are relying, and you should not attempt to upgrade any system without taking a full backup first.

    Link to this post: https://forums.raspberrypi.com/viewtopic.php?t=389477

2. Debian 13 and Raspberry Pi OS: Debian 13 (Trixie) is not officially released yet. Consequently, there isn't an official Raspberry Pi OS image based on Debian 13 available either. Therefore, stable development of Crowsnest that guarantees reliable functionality under this configuration is currently not possible.

3. Your solution of renaming the `/etc/rpi-issue` file to bypass the dependency check unfortunately circumvents internal checks in Crowsnest that are designed to determine if it's running on a Raspberry Pi with Raspberry Pi OS. This means that your system, due to this "fix," deviates even further from a supported configuration regarding the environment expected by Crowsnest.

For these reasons, and to ensure clarity and prevent other users from applying a "fix" that could have further undesirable side effects, we have hidden your post. We want to ensure that only solutions that are long-term stable and maintainable are shared here.