mainsail-crew/crowsnest #303: Latest Bookwom camera-streamer build errors with RTSP

mryel00, 2025-06-25:
### What happened

The latest Bookworm packages break the building of camera-streamer again. This time it affects RTSP, as you can read here: https://github.com/ayufan/camera-streamer/issues/177

Therefore we will disable RTSP in new camera-streamer builds by adjusting https://github.com/mryel00/camera-streamer
Furthermore we will remove the `enable_rtsp` and `rtsp_port` options from the example config. We will not add it back into the example config and it will only be listed in our [docs](https://crowsnest.mainsail.xyz/configuration/cam-section#enable_rtsp) even if the issue is fixed. For the reason why see additional information.

### What did you expect to happen

RTSP should be fixed in camera-streamer or get removed entirely.

### How to reproduce

Update to latest Bookworm system packages and try to install Crowsnest or camera-streamer.

### Additional information

RTSP should not get confused with WebRTC.
As we see a lot of people just mindlessly enabling RTSP without even knowing what it is, we will remove the option from the example config.

This issue only applies to Raspberry Pis prior to the Pi5, as the Pi5 is not able to build camera-streamer with WebRTC and RTSP support.

We don't know how many people are actually using RTSP, so if you are actually using it, please add a 😕 reaction. No need to comment on this issue.
To get a working RTSP, there are currently only two supported options from us:
- Install MainsailOS v2.0.0 and don't update your system packages
- Install some Bullseye PiOS based image and install Crowsnest on it

We currently cannot easily provide any other options for RTSP.

slepkaviba, 2025-07-09:
As a small comment - RTSP is the only way to have Mainsail + Obico running on Zero W2.

Without it it goes from awesome to "no print can start"... I do not know how popular is Zero W2 but it is a big thing for those users.

slepkaviba, 2025-07-09:
So I enable RTSP in Crowsnest. 

As Obico is "restreaming" from client - using Janus and ffmpeg - it is quite CPU intense task on Z2W - but it has 'copy' mode, that with RTSP uses basically no CPU. I tried to do copy now on h264 stream - and it consumes enough of CPU so that print cannot start.

I guess "alternative" is to use Obico in super low resolution/FPS. But as I use it for timelapse too, that will be gone.