mainsail-crew/crowsnest #308: X86 install fails on camera-streamer

HertogJanVI, 2025-08-08:
### What happened

Tried to update according to the v3->v4 guide.

Installer seems to see i run x86:
`Checking for camera-streamer dependencies ...

Checking if device is a Raspberry Pi ...

Checking if device is a Raspberry Pi ... [FAILED]
This device is not a Raspberry Pi therefore camera-streamer cannot be installed ...
Doing some tests ... [OK]
Installing without camera-streamer ...`

Then fails on installing camera streamer:
`done
make[1]: Map '/home/jan/crowsnest/bin/ustreamer' wordt verlaten
Build 'ustreamer' ... [SUCCESS]
Build 'camera-streamer' using 4 Cores ... 
make: *** Geen doelen opgegeven en geen Makefile gevonden.  Gestopt.
make: *** [Makefile:19: install] Fout 2`

### What did you expect to happen

Install without camera-streamer as that runs on raspberry sbc’s only.

### How to reproduce

Follow the update guide on x86.

HertogJanVI, 2025-08-09:
Sorry for the formatting. The terminal text contains symbols that are interpreted as markdown..

mryel00, 2025-08-09:
As you can read in the [Readme](https://github.com/mainsail-crew/crowsnest/blob/master/README.md?plain=1#L68), we do not support x86 PiOS.
Please tell me why you chose it, as you are now the second person we encounter with such an OS and it doesn't make any sense for us, why anyone would use PiOS on x86, as there is no benefit over e.g. Ubuntu.

mryel00, 2025-08-09:
As Bullseye is end of life, I would recommend to update to a newer OS by installing e.g. Ubuntu or Debian. If you don't want to install everything from scratch, follow [this guide](https://crowsnest.mainsail.xyz/faq/use-legacy-branch-on-buster) or [rewrite the check](https://github.com/mainsail-crew/crowsnest/blob/master/libs/core.sh#L98-L105) locally (I won't help with that)

closed 2025-08-09 by mryel00

ShadowSharmon, 2025-12-28:
Yes, agreed on your points. My issue should not be recoded to for sure as I'm not on supported hardware either, it just happened to be what I had <surfaceRT with a tegra 3, LOL, and someone packaged and wrote a guide for it so...>.

the one thing maybe to consider is that the make install has a decent check to see if you're really on a Pi <to see if it should try and compile camera-streamer> but then doesn't use that check output when checking the camera-streamer dependency... and since crowsnest can work without camera-streamer on a Pi, maybe it really isn't a dependency at all? just pondering if that would have more people succeeding upgrade...

Mine is back up and running with ustreamer <until I can buy a btt pad7 or something>. Thanks again for all your work on this!

mryel00, 2025-12-28:
> the one thing maybe to consider is that the make install has a decent check to see if you're really on a Pi but then doesn't use that check output when checking the camera-streamer dependency...

If we would do that check during runtime, a restart of the service would take even longer and other problems would arise too, so for runtime we had to find a simple solution. Tbh it's kinda good that we have a separation, as we can find some weird edge cases like this and without this, you would fail on the installation, and not during runtime. Both is pretty bad and should not happen on a device that could run it.

> since crowsnest can work without camera-streamer on a Pi, maybe it really isn't a dependency at all?

The word dependency is kinda misleading as it's not really a dependency at all. It's more like a streamer check and it failing if it cannot find one streamer, is like the worst thing that can happen but it's necessary with the current implementation.
That's why the python rewrite (pynest) is fixing this and just let you choose streamers that are actually there. Rewriting the current bash code to not fail, doesn't make any sense just to fix some small edge case that we officially don't support. In the past 7 months there were like 10-15 people with such an OS asking for help. It was just a coincident in the past that it worked ^^

The current implementation using the `/etc/rpi-issue` file to detect if it's a Rpi is the easiest and fastest option you can do during runtime in bash. Pynest will handle nearly everything during installation and checks during runtime if the streamer binary is there or not. There is just one exception and that's the Pi5 with MainsailOS. We do not want to distribute a different OS for the Pi5, so we have to check during runtime if it's a Pi5 or not.

I hope you can understand the current problem and our future plans on solving it, but this issue stays closed as we do not support it and PiOS for x86 is EOL anyway.

Development of Pynest is close to being finished but time is short that I can spend on developing and testing it right now. Mostly testing and reviewing the code is missing. So if you know Python and got some time to spare, you can test and review #270. Help is always appreciated 😄